ros2/ros2 #442: Tracking bug for runtime package for rosidl

clalancette, 2018-01-02:
Currently, all packages that generate messages have a dependency on https://github.com/ros2/rosidl_typesupport/blob/master/rosidl_default_runtime/package.xml .  In turn, the `rosidl_default_runtime` package depends on `rosidl_generator_cpp`, which it really shouldn't.  What we should do instead is make another package (let's call it `rosidl_cpp_runtime` for now), and have the headers that the generated message code depends on live there (basically all of the headers in https://github.com/ros2/rosidl/blob/master/rosidl_generator_cpp/include/rosidl_generator_cpp).  Once we have that, then `rosidl_default_runtime` can depend on `rosidl_cpp_runtime`, breaking the dependency on the generator.

mikaelarguedas, 2018-03-08:
@clalancette is it something that's still on your radar?

clalancette, 2018-03-08:
I should be able to get back to this pretty soon.

ahcorde, 2020-02-25:
Hello, 

I have started to split `rosidl_generator_c` and `rosidl_generator_cpp` in two packages: 

 - `rosidl_generator_c` and `rosidl_runtime_c`
 - `rosidl_generator_cpp` and `rosidl_runtime_cpp`

the idea is to include all the exec dependencies in the runtime package and the generator utils in the `rosidl_generator` package

I made the following division:

 - rosidl_generator_X will contain
   - bin 
   - cmake
   - resources

 - rosidl_runtime_X will contain
   - rosidl_runtime_X (python)
   - src 
   - include
   - test

@clalancette and @wjwwood do you have some thoughts about this?

dirk-thomas, 2020-02-25:
> I made the following division:

* Why should `src` be in the runtime package? It is only compiled once and doesn't need to be present at the runtime of message packages.

* I don't think `test` can be in the runtime package. E.g. in `rosidl_generator_c` the test covers generated message code.

wjwwood, 2020-02-25:
Doesn’t the src folder contain code for the shared library? In that case I do think it needs to be in the runtime package. I think only the c generator has this.

dirk-thomas, 2020-02-25:
> Doesn’t the src folder contain code for the shared library? In that case I do think it needs to be in the runtime package.

That sounds right.

clalancette, 2020-08-13:
@ahcorde Is this done now?

ahcorde, 2020-08-13:
@clalancette It's done

closed 2020-08-13 by ahcorde