ros2/ros2 #444: Build instructions error - OpenCV

Paramojo, 2018-01-04:
This actually started out when ROS bombed on doing the first test from the instructions for the Windows binary install but complaining about Fast-RTPS DLL, but suspect OpenCV might have been the root cause, so thought I'd have a go and build from source to see whats going on better.

I am now doing a second build with the --build-tests option to be doubly sure and then the additional tests.

## Bug report

**Required Info:**

- Operating System:
  -  Windows 10 Pro, ver 1709, build 16299.125
- Installation type:
  -  Source - Building on Visual Studio 2017x64 Native
- Version or commit hash:
  -  ?
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  -  ?

#### Steps to reproduce issue

``` Followed instructions here https://github.com/ros2/ros2/wiki/Windows-Development-Setup
 >Got to here https://github.com/ros2/ros2/wiki/Windows-Development-Setup#install-a-binary-distribution-of-opencv
did as told and did the rest and crashed out the minute it needed to use OpenCV with a Version error
Messed around for a while and finally settled on pointing environment variables to the latest WinPack version downloaded from the OpenCV site here, the main Build folder for the OPEN_CV_DIR and x64/vc15 for Path
https://opencv.org/releases.html

And bobs your Auntie no splashing out.
```

#### Expected behavior
It building as per instructions
#### Actual behavior
It not Building as per instructions
#### Additional information

dirk-thomas, 2018-02-01:
Since you have filled the issue we have updated the instructions to use a newer version of OpenCV for Visual Studio 2017: https://github.com/ros2/ros2/wiki/Windows-Development-Setup/_compare/a7e498c674cfd298b7b39bfb920e298ef9a69eaa...90f9e52734ef0e25d8a9319556c1412032d59926

Can you please try it again with the updated dependency and report back if this fixed your problem.

dirk-thomas, 2018-03-21:
I will close this due to no response. Please feel free to comment with the requested feedback and the issue can be reopened if necessary.

closed 2018-03-21 by dirk-thomas