ros2/ros2 #445: Would you please put our client on ros wiki?

minggangw, 2018-01-04:
Dear Sir/Madam,

We (@halton, @kenny-y and @minggangw) are developing [rclnodejs](https://github.com/RobotWebTools/rclnodejs), a Node.js client for ROS2, to bring the robot to the web world. Currently we have released four editions, the latest one is beta2 on [npmjs.com](https://www.npmjs.com/package/rclnodejs).

We have the same [features](https://github.com/RobotWebTools/rclnodejs/releases) and functionalities compared with [rclpy](https://github.com/ros2/rclpy), CI services which cover Linux/Windows/macOS platforms, over four hundreds unit [tests](https://github.com/RobotWebTools/rclnodejs/tree/develop/test) run by mocha and complete [documents](https://github.com/RobotWebTools/rclnodejs/tree/develop/docs) and examples.

For ROS1, there is a single [wiki page](http://wiki.ros.org/Client%20Libraries) to put all the clients on a list, but I did not find something like that in wiki of ROS2. I think our effort and project can strengthen the ROS eco-system, promote the popularization of ROS2 and make the ROS world available to hundreds of thousands of web programmers. Through the npm toolchain, a developer can deploy his/her apps for ROS easily.

So would you create a page on the current ROS2 wiki, which will demonstrates different kinds of client libraries (we will have more definitely). This will unites our endeavor!

Yours faithfully!

dirk-thomas, 2018-01-04:
Please feel free to create a wiki page for this and/or edit existing pages. After doing so it would be good to post a link to the new page / to the changes here for broader visibility.

alexandroid000, 2018-01-10:
Seems like this is where that info should go: https://github.com/ros2/ros2/wiki/ROS-2-Client-Libraries

It would also be good to have links to the Java, C# and Swift clients mentioned there, I'm not sure where those are.

esteve, 2018-01-10:
@fkromer is curating a list of projects that use ROS2 (https://github.com/fkromer/awesome-ros2), which also includes the community-supported client libraries. Perhaps the list could be integrated into the ROS2 wiki somehow or have a link to it.

alexandroid000, 2018-01-11:
I made some changes to [the client library wiki page](https://github.com/ros2/ros2/wiki/ROS-2-Client-Libraries), added a few links and moved the client libraries up. Feel free to change links, I just picked what popped up first on google.

The page by @fkromer is good, and I guess this "awesome" format is becoming kind of standardized, but yeah it would be good to avoid duplicate work.

fkromer, 2018-01-11:
@esteve There is a link on the [wiki landing page](https://github.com/ros2/ros2/wiki#where-to-find-more-information). Let me know if you want to host [awesome-ros2
](https://github.com/fkromer/awesome-ros2) from the [ros2 github organization](https://github.com/ros2). @alexandroid000 For me awesome lists and wikis have different purposes: list = quick reference for (mainly) libs, wikis = in depth information. I want to avoid maintenance effort. However if you come up with a good solution how to keep the wiki and the awesome list in sync let me know.

dirk-thomas, 2018-01-11:
The original request seems to be fulfilled with the additions to the wiki page. Therefore I am going to close this ticket.

closed 2018-01-11 by dirk-thomas

minggangw, 2018-01-16:
Hi there, thanks a lot for doing this, I believe together we will boost the ROS2 eco-system! Thanks!