ros2/ros2 #446: Cloning wiki, adding/modifying content locally and pushing changes

fkromer, 2018-01-05:
How can one get permission to be able to push additions/changes to the wiki (cloned via `git clone https://github.com/ros2/ros2.wiki.git`)?

wjwwood, 2018-01-05:
@fkromer You have to have write permissions on the repository. Why do you need to do it with a checkout and not through the web interface?

fkromer, 2018-01-05:
@wjwwood I would like to add .png diagrams (cannot be hyper linked). As far as I know it is not possible to add files to the wiki repository over the web interface.

wjwwood, 2018-01-05:
You can add images to the wiki which are hosted elsewhere. Why can it not be hyperlinked?

wjwwood, 2018-01-05:
I suppose I could make you a contributor on this repository to let you do that, but I need to check with the team before I do that. Just trying to figure out if there is a work around for now.

fkromer, 2018-01-05:
If it is not straightforward to grant permissions it's fine to hyper link images from else where as well. I just wanted to avoid that additional burden w.r.t. work flow.

wjwwood, 2018-01-06:
Unfortunately it doesn't allow us to control access at that level of granularity. I think we would prefer to keep the people with write access in the organization low, but we could add you, at least temporarily, to just this repository. @ros2/team what do you think?

It's too bad that GitHub neither let's you push to the wiki's git repo (despite being allowed to create/modify things via the web interface) nor does it let us grant you that permission separately from write access to the repository itself.  @nuclearsandwich is that all correct or are we missing something?

nuclearsandwich, 2018-01-06:
> It's too bad that GitHub neither let's you push to the wiki's git repo (despite being allowed to create/modify things via the web interface) nor does it let us grant you that permission separately [...]

That jives with my understanding as well. To add images to wikis, I usually upload them to an issue comment, copy the url then never submit the comment 😈

Karsten1987, 2018-01-06:
I've used imgur in the past to upload the pictures and then hyperlink them into the tutorials. Worked pretty well for me.

wjwwood, 2018-01-08:
It sounds like there is a decent workaround (it's what we do) to use imgur or ~~abuse~~ use GitHub's comment image hosting and linking to that.

I'll close this for now, if it becomes an extreme burden, let us know and I can reopen this.

closed 2018-01-08 by wjwwood