ros2/ros2 #449: Use eProsima Fast-RTPS on Multi-hop network

faciler, 2018-01-18:
I want to use eProsima Fast-RTPS  on Multi-hop network. By olsr to transpond data, for RTPS message on Multi-hop need transpond multicast data, so I download a OLSR Multicast Forwarding Plugin,  but if I `sudo ./olsrd`,    run eprosima code ,then got this error:
terminate called after throwing an instance of 'std::system_error'
what(): set_option: Address already in use

Is anyone know how to solve it?

sloretz, 2018-01-18:
Hi @faciler, would you mind filling out the following sections in the issue template? I'm not sure what the steps are to reproduce the issue.

Template
```
<!--
For general questions, please post on discourse: https://discourse.ros.org/c/ng-ros
For Bug report or feature requests, please fill out the relevant category below
-->

## Bug report

**Required Info:**

- Operating System:
  - <!-- OS and version (e.g. Windows 10, Ubuntu 16.04...) -->
- Installation type:
  - <!-- binaries or from source  -->
- Version or commit hash:
  - <!-- Output of git rev-parse HEAD, release version, or repos file  -->
- DDS implementation:
  - <!-- rmw_implementation used (e.g. Fast-RTPS, RTI Connext, etc -->
- Client library (if applicable):
  - <!-- e.g. rclcpp, rclpy, or N/A -->

#### Steps to reproduce issue
<!-- Detailed instructions on how to reliably reproduce this issue http://sscce.org/
code that can be copy-pasted is preferred  -->

#### Expected behavior

#### Actual behavior

#### Additional information

faciler, 2018-01-19:
##feature requests
- Operating System:
  - 
- Installation type:
- Version or commit hash:
- DDS implementation:
- 
#### Steps to reproduce issue

I implement a linear multi-hop network, the topology is as simple as
Node_A <--> Node_B <--> Node_C
where Node_A and Node_C are blocked by iptable rules.

I want to use eProsima Fast-RTPS on Multi-hop network, By olsr to transpond data, RTPS required transpond multicast data(e.g rti_connext_dds,opendds).
I have successfully implement it on rti_connext_dds, but I need eprosima Fast-rtps.
First I run olsr:
`cd olsr` 
`sudo ./olsrd`

Then run rtps HelloWorldExample demo
yhj@yhj:~/eprosima/eProsima_FastRTPS-1.5.0-Linux/examples/C++/HelloWorldExample$ ./HelloWorldExample publisher
Starting 
terminate called after throwing an instance of 'std::system_error'
  what():  set_option: Address already in use
Aborted (core dumped)

#### Expected behavior

#### Actual behavior

#### Additional information

dirk-thomas, 2018-01-19:
I would recommend to fill the issue [upstream](https://github.com/eProsima/Fast-RTPS) instead.

closed 2018-01-19 by faciler