ros2/ros2 #450: Static Remapping

sloretz, 2018-01-19:
## Feature request
This ticket is meant to track the implementation of static remapping.

#### Feature description
Static remapping is the ability to remap one name (topic, service, parameter) to another. See the [design doc](http://design.ros2.org/articles/static_remapping.html) for more info.

#### Implementation considerations

##### ROS 2 proposed implementation
TODO

##### ROS 1 Implementation
`roscpp` stores global remapping rules (from command line arguments or passed via `ros::init()`) in [two global variables](https://github.com/ros/ros_comm/blob/970372dfea631e6bafd936795222de3c79a6fca6/clients/roscpp/src/libros/names.cpp#L43-L44). These variables are written to only once; it happens during [library initialization](https://github.com/ros/ros_comm/blob/970372dfea631e6bafd936795222de3c79a6fca6/clients/roscpp/src/libros/init.cpp#L459-L464). One variable stores the remappings before expanding relative names to fully qualified names (unresolved), while the other stores the same remappings after expansion (resolved).

The unresolved remappings are only used to implement `searchParam`. Since it is not known what namespace the param should be found in, relative name remapping will change the relative parameter name being searched for (TODO if there is a rule from a relative name to a fully qualified name that matches the parameter being searched for, what will the master do when told to search with a fully qualified parameter)?

Remapping rules can also be passed into a `ros::NodeHandle` via [its constructor](https://github.com/ros/ros_comm/blob/970372dfea631e6bafd936795222de3c79a6fca6/clients/roscpp/include/ros/node_handle.h#L90-L104). Any name created via this node handle is [first checked against the remappings given to the constructor](https://github.com/ros/ros_comm/blob/970372dfea631e6bafd936795222de3c79a6fca6/clients/roscpp/src/libros/node_handle.cpp#L220-L225). If none match then the [global rules are checked](https://github.com/ros/ros_comm/blob/970372dfea631e6bafd936795222de3c79a6fca6/clients/roscpp/src/libros/node_handle.cpp#L227-L228).

The name remappings are evaluated when:
* [asked to wait for a service](https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/roscpp/src/libros/service.cpp#L89) (global remappings only, does not consider `NodeHandle` specific remappings, TODO why)
* [determining a namespace of a NodeHandle during construction](https://github.com/ros/ros_comm/blob/a46a0fe7f69cdd702ff15aaca3c2a1260a9be98f/clients/roscpp/src/libros/node_handle.cpp#L161)
* [advertising a topic](https://github.com/ros/ros_comm/blob/a46a0fe7f69cdd702ff15aaca3c2a1260a9be98f/clients/roscpp/src/libros/node_handle.cpp#L287)
* [subscribing to a topic](https://github.com/ros/ros_comm/blob/a46a0fe7f69cdd702ff15aaca3c2a1260a9be98f/clients/roscpp/src/libros/node_handle.cpp#L320)
* [advertising a service](https://github.com/ros/ros_comm/blob/a46a0fe7f69cdd702ff15aaca3c2a1260a9be98f/clients/roscpp/src/libros/node_handle.cpp#L350)
* [creating a service client](https://github.com/ros/ros_comm/blob/a46a0fe7f69cdd702ff15aaca3c2a1260a9be98f/clients/roscpp/src/libros/node_handle.cpp#L380)
* [Calling a service](https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/roscpp/include/ros/service.h#L69)
* getting, setting, or searching for parameters (too many places to link)

`rospy` also stores global remappings into [two global variables](https://github.com/ros/ros_comm/blob/970372dfea631e6bafd936795222de3c79a6fca6/clients/rospy/src/rospy/names.py#L86-L87) with the same meaning as in `roscpp`. All remappings in `rospy` are global. It does not have an equivalent to `NodeHandle`. Remappings either must come from the command line, or they [must be given as if they were command line arguments](https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/clients/rospy/src/rospy/client.py#L259) to the `argv` argument of `rospy.init_node`.

wjwwood, 2018-01-30:
You'll probably want the `rcl` functions you proposed to return a "ret type" and take an "output" parameter as an argument instead, but otherwise that list seems reasonable.

Thanks for such a thorough plan, it all sounds like you're on the right track to me.

dirk-thomas, 2018-02-01:
> `rcl_init()` parses command line arguments and stores the results globally

I would suggest to avoid global store. Instead the function should return "something" (and optionally manipulate the passed `argc` and `argv`). That information than needs to be passed around wherever it is required.

> rcl_client_init(...), ... : Implementation uses rcl_remap_name(...) on the supplied service/topic name

(same for other similar cases where a function now additionally considers remapping logic)

I would suggest a "layered" approach. If `rcl_client_init` performs the remapping there should be another API "under" it (e.g. `rcl_client_init_without_remapping`) which contains some part of the logic and can be tested independently without the need for any remapping logic. The API doesn't have to be something a user might use but should allow testing of a "smaller layer without the remapping complexity".

wjwwood, 2018-02-01:
> > rcl_init() parses command line arguments and stores the results globally

> I would suggest to avoid global store. Instead the function should return "something" (and optionally manipulate the passed argc and argv). That information than needs to be passed around wherever it is required.

-1, `rcl_init()` is already global. If you would like to make a non-global version of `rcl_init()` that's fine, but more than the remapping arguments needs to be considered.

sloretz, 2018-05-02:
All remapping features targeted at bouncy have been implemented. I think ros2/system_tests#268 will be the last remapping PR before bouncy is released.

I'm closing this issue because I think it has reached the end of it's useful life. Future work has been broken up into these tickets:
* ros2/design#173
* ros2/design#174
* ros2/rcl#232
* ros2/rcl#233
* ros2/rcl#234

closed 2018-05-02 by sloretz