ros2/ros2 #451: ros2 python demo program runs error

lxbeyond, 2018-01-26:
I build the ros/examples reps , and run c++ program well , but can not run python program , the error as follow:

/ros2/ros2workspace/src/examples/rclpy/topics/minimal_publisher$ python3 publisher_member_function.py 
Traceback (most recent call last):
  File "publisher_member_function.py", line 53, in <module>
    main()
  File "publisher_member_function.py", line 41, in main
    minimal_publisher = MinimalPublisher()
  File "publisher_member_function.py", line 25, in __init__
    self.publisher_ = self.create_publisher(String, 'topic')
  File "/opt/ros/ardent/lib/python3.5/site-packages/rclpy/node.py", line 111, in create_publisher
    check_for_type_support(msg_type)
  File "/opt/ros/ardent/lib/python3.5/site-packages/rclpy/node.py", line 38, in check_for_type_support
    ts = msg_type.__class__._TYPE_SUPPORT
AttributeError: type object 'type' has no attribute '_TYPE_SUPPORT' This might be a ROS 1 message type but it should be a ROS 2 message type. Make sure to source your ROS 2 workspace after your ROS 1 workspace.

   I did install both ros1 and ros2 in my ubuntu, and ros1 python program runs well, anyone can  give some advice  ?

lxbeyond, 2018-01-26:
I just figure it out, there is problem in PYTHONPATH , I clean it, and set the right path, then it works

closed 2018-01-26 by lxbeyond

tiagoamaralp, 2019-07-05:
> I just figure it out, there is problem in PYTHONPATH , I clean it, and set the right path, then it works

what's the PYTHONPATH you used? thanks in advance.

lxbeyond, 2019-07-08:
I nearly fotget this problem. If you encounter with this problem , you would be better to look up the "env" result in teminal.

timrobotson, 2019-08-22:
I just encountered this error because I converted a ROS1 application to ROS2 and didn't realize the message type and topic arguments are the other way around in ROS2 (Message type is the first argument, topic is second).

myboyhood, 2020-02-17:
> > I just figure it out, there is problem in PYTHONPATH , I clean it, and set the right path, then it works
> 
> what's the PYTHONPATH you used? thanks in advance.

I disable my ROS1 PYTHONPATH by change PYTHONPATH in ~/.bashrc
```
export PYTHONPATH=/opt/ros/crystal/lib/python3.6/site-packages
(disable)# PYTHONPATH=$PYTHONPATH:/opt/ros/melodic/lib/python2.7/dist-packages
``` 
and,if you want your ROS_DISTRO = crystal ,remenber to 
`source /opt/ros/crystal/setup.bash` after `source /opt/ros/melodic/setup.bash`