ros2/ros2 #454: Ardent patch release 1

dirk-thomas, 2018-01-30:
For an Ardent patch release 1 we will cherry-pick the minimal changes which are necessary to address important issues. The high level goals are:
  * provide `rviz` for Windows
  * fix problem(s) in FastRTPS
  * fix problem with include path order in `rosidl`
  * Provide `ros_environment` for the env vars specified in REP 149

We will do the following:

* Create PRs using a to be created source branch named `ardent-patch1` targeting the existing `ardent branch`:
  * `rviz` @wjwwood
  * `pluginlib` @wjwwood
  * `resource_retriever` @wjwwood
  * `fastrtps` @dirk-thomas
  * `rosidl` @mikaelarguedas
  * `ros_environment` @dirk-thomas
* Check is any packages should be updated on the macOS / Windows nodes (and if it affects the instructions): @wjwwood
* Roll some fat archives to check they "address" the problems.
* Merge the PRs.
* Make a patch release in the affected repos.
* Do the `bloom` releases.
  * @nuclearsandwich mentioned addressing a minor issue in the FastRTPS manifest which can be fixed along the way.
* Upload the archives under a new release in the `ros2` repo.
* Announce the patch release with a name line `ardent-2018-02-xx`.

wjwwood, 2018-01-31:
I created a single Windows CI job to see if `rviz` even compiles in this setup:

- Windows: [image]

nuclearsandwich, 2018-01-31:
> addressing a minor issue in the FastRTPS manifest which can be fixed along the way.

Addressed by https://github.com/ros2-gbp/fastrtps-release/pull/1

Noting for my own records that Fast-CDR hasn't changed since the ardent release so there's nothing there that needs bumping.

wjwwood, 2018-02-01:
We're tracking some issues with rviz's current state which we'd like to fix: https://github.com/ros2/rviz/pull/188#issuecomment-361993826

wjwwood, 2018-02-05:
I've finally finished updating all the CI machines in preparation for the release.

I'm going to update the rviz patch branch with some fixes that have been made since last week and then run CI again for this release. Then we're ready to make packages with the packaging job and do what ever level of testing we think is appropriate.

wjwwood, 2018-02-05:
I've also got a few staged changes to the wiki instructions, but we probably shouldn't change those until we have the patch release available (otherwise the existing binary would not work).

wjwwood, 2018-02-06:
I've updated the rviz pull request and so here's full CI on the updated CI machines:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

wjwwood, 2018-02-06:
On macOS there was a false positive cmake warning created by interleaved output and the new cppcheck test failures which were introduced by update cppcheck on the macOS build machines:

http://ci.ros2.org/job/ci_osx/3273/testReport/junit/(root)/projectroot/cppcheck/

They were fixed in https://github.com/ros2/rclcpp/pull/436. Should that be back ported too or should we just live with it (since we'll not be running CI and nightlies on it anyways)?

----

Here's packaging jobs:

- Linux:
  - Fast-RTPS: [image]
  - OpenSplice: [image]
- Linux-aarch64:
  - Fast-RTPS: [image]
  - OpenSplice: [image]
- macOS:
  - Fast-RTPS: [image]
  - OpenSplice: [image]
- Windows:
  - Fast-RTPS: [image]
  - OpenSplice: [image]

dirk-thomas, 2018-02-07:
The previous builds used a gist which seems to not have a newer version of FastRTPS and doesn't include `ros_environment`. The following builds use the `ros2.repos` file from ros2/ros2#455.

* Linux
  * FastRTPS [image]
  * OpenSplice [image]
* Linux-aarch64
  * FastRTPS [image]
  * OpenSplice [image]
* macOS
  * FastRTPS [image]
  * OpenSplice [image]
* Windows
  * FastRTPS [image]
  * OpenSplice [image]

dirk-thomas, 2018-02-07:
> Should that be back ported too or should we just live with it (since we'll not be running CI and nightlies on it anyways)?

I would say we keep it as is now.

dirk-thomas, 2018-02-07:
When testing the fat archives and later the Debian packages we should check the following:

* check that the new environment variables `ROS_VERSION` and `ROS_DISTRO` are there
* check that `rviz2` starts and try adding / removing camera displays
* build custom `rosidl_interfaces` in an overlay workspace (see ros2/rcl_interfaces#31)
* run a couple of demos nodes for sanity checking

nuclearsandwich, 2018-02-07:
## Windows 🎐 

### Fast-RTPS archive
* ✅ New environment variables
* ✅ rviz2 starts
* ✅ camera displays can be added and removed (no actual camera data to publish)
* ⛔️  dummy_robot_bringup runs and tf animates once but then launched process appears to crash.
* ✅ talker/listener, service,clients run
* ⛔️ Cannot run msbuild to test overlay workspace

### OpenSplice archive 
* ✅ New environment variables
* ✅ rviz2 starts
* ✅ camera displays can be added and removed (no actual camera data to publish)
* ✅ rviz animates tf data when dummy_robot_bringup is launched
* ✅ talker/listener, service,clients run
* ⛔️ Cannot run msbuild to test overlay workspace

Karsten1987, 2018-02-07:
Fast-RTPS on OSX looks good on a first glance. 
Similar to the Windows report above, the TF tree doesn't animate correctly and times out.

nuclearsandwich, 2018-02-07:
Added a short wiki page for the necessary setup steps for releasing packages with bloom. https://github.com/ros2/ros2/wiki/Releasing-a-ROS-2-package-with-bloom/_history

wjwwood, 2018-02-07:
## macOS 

### OpenSplice archive 
* ✅ New environment variables
* ✅ `rviz2` starts
* ✅ camera displays can be added and removed (no actual camera data to publish)
* ✅ `rviz` animates `tf` data when `dummy_robot_bringup` is launched
* ✅ talker/listener, service/clients run
* ✅ Reproduced in older binary, fixed in new

As a side note, macOS is doing a pretty good job these days letting you know what the issue is:

```
% rviz2
dyld: Library not loaded: /usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets
  Referenced from: /tmp/ros2-osx/bin/rviz2
  Reason: Incompatible library version: rviz2 requires version 5.10.0 or later, but QtWidgets provides version 5.9.0
[1]    58108 abort      rviz2
```

and

```
% rviz2
dyld: Library not loaded: /usr/local/opt/freetype/lib/libfreetype.6.dylib
  Referenced from: /tmp/ros2-osx/bin/rviz2
  Reason: Incompatible library version: rviz2 requires version 23.0.0 or later, but libfreetype.6.dylib provides version 22.0.0
[1]    59719 abort      rviz2
```

clalancette, 2018-02-07:
I did some testing on ARM64.  I specifically tested the issues listed in https://github.com/ros2/ros2/issues/454#issuecomment-363911929, and all of those were fine.  I also did spot checking of client->server, pub/sub, cpp->py on Fast-RTPS and OpenSplice, which worked fine.

However, I did run into a bug when trying to cross Fast-RTPS and OpenSplice.  If I run an opensplice `ros2 run demo_nodes_cpp add_two_ints_server`, and a Fast-RTPS `ros2 run demo_nodes_cpp add_two_ints_client`, the client waits around forever.

mikaelarguedas, 2018-02-08:
> I did run into a bug when trying to cross Fast-RTPS and OpenSplice

cross vendor services are not supported, in ardent or master, so I'll say it's expected

clalancette, 2018-02-08:
> cross vendor services are not supported, in ardent or master, so I'll say it's expected

Great, sounds fine then.  No other bugs to report.

nuclearsandwich, 2018-02-08:
Bloom release PRs are starting to roll in to ros2/rosdistro. My intention is to merge them all at once after https://github.com/ros2/ros2/pull/455 is reviewed and merged.

nuclearsandwich, 2018-02-09:
Debian packages are ready in the ros2 testing repository. Packaging jobs are queued for fat archives.

## MacOS
##### Fast-RTPS [image]
##### OpenSplice [image] (queued)

## Ubuntu Xenial

##### Fast-RTPS [image] (queued)

##### OpenSplice [image] (queued)

## Windows

##### Fast-RTPS [image]
##### OpenSplice [image] (queued)

clalancette, 2018-02-09:
I tested the ARM64 debs, using the basic recipe in https://github.com/ros2/ros2/issues/454#issuecomment-363911929 .  All seemed well there, including a pub/sub and a service test (I couldn't test rviz2, but I think that is OK).

dirk-thomas, 2018-02-09:
Same results for me when I tried them yesterday late afternoon. Can someone please test `rviz2`.

nuclearsandwich, 2018-02-09:
> Same results for me when I tried them yesterday late afternoon. Can someone please test rviz2.

rviz2 dummy robot bringup worked fine with both opensplice and fastrtps.

mikaelarguedas, 2018-02-09:
tested rviz2 from binaries, it launches fine and doesnt crash on camera plugin load/unload

the dummy_robot example didnt work first time with the tfs being received only once and then fading away. Worked fine the second time

dirk-thomas, 2018-02-09:
> I've also got a few staged changes to the wiki instructions, but we probably shouldn't change those until we have the patch release available (otherwise the existing binary would not work).

@wjwwood Can you push these changes and reference them here.

wjwwood, 2018-02-09:
I've pushed the changes:

- https://github.com/ros2/ros2/wiki/OSX-Install-Binary/_compare/6801c07e868dc6919eecb80dfa58d097135479ae...b67511b9323d016e0993e5eb2a32f52b50311208
- https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/a8369e7f20d5bb4e361be66de51e7107c8bdcbbe...1656b84e891c8bb39aa123cf825b2758be0a4e87
- https://github.com/ros2/ros2/wiki/Windows-Development-Setup/_compare/90f9e52734ef0e25d8a9319556c1412032d59926...8ac1d8bd2c887383606e2bbfad1055fc6532da8f

closed 2018-02-12 by wjwwood