ros2/ros2 #457: fix libraries with install_name_tool instead of disabling SIP

fferri, 2018-02-01:
It is not a wise idea to disable SIP.
You can make everything work without disabling SIP and even without setting DYLD_LIBRARY_PATH, just by fixing executables with the macOS command `install_name_tool`, adding every path contained in $DYLD_LIBRARY_PATH as RPATH.

Here's what I did (note: $PWD=/path/to/ros2-osx):

```
$ find . -name '*.so'    -exec install_name_tool -add_rpath $PWD/lib '{}' ';'
$ find . -name '*.dylib' -exec install_name_tool -add_rpath $PWD/lib '{}' ';'
$ find . -name '*.so'    -exec install_name_tool -add_rpath $PWD/opt/rviz_yaml_cpp_vendor/lib '{}' ';'
$ find . -name '*.dylib' -exec install_name_tool -add_rpath $PWD/opt/rviz_yaml_cpp_vendor/lib '{}' ';'
$ find . -name '*.so'    -exec install_name_tool -add_rpath $PWD/opt/rviz_ogre_vendor/lib '{}' ';'
$ find . -name '*.dylib' -exec install_name_tool -add_rpath $PWD/opt/rviz_ogre_vendor/lib '{}' ';'
```

After doing so, there's still a problem with the load of libPocoFoundation.47.dylib, because it is listed as `libPocoFoundation.47.dylib` and not as `@rpath/libPocoFoundation.47.dylib`, so I did:

```
$ install_name_tool -change libPocoFoundation.47.dylib @rpath/libPocoFoundation.47.dylib lib/librosidl_typesupport_c.dylib
$ install_name_tool -change libPocoFoundation.47.dylib @rpath/libPocoFoundation.47.dylib lib/librosidl_typesupport_cpp.dylib
```

After doing so, I can do:

```
$ csrutil status
System Integrity Protection status: enabled.
$ unset DYLD_LIBRARY_PATH
$ ros2
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: the following arguments are required: Call `ros2 <command> -h` for more detailed usage.
```

wjwwood, 2018-02-02:
> It is not a wise idea to disable SIP.

Agreed, but we ~~wasted~~ spent quite some time trying to work with it on, and we ran into a few issues -- more than just the library path getting dropped. For instance, I think @dhood noted a significant slow down on our CI jobs and perhaps some other things which I can't remember now.

If you can open pr's to fix the CMake logic so this is fixed (running `install_name_tool` manually or even in a post build script isn't really an attractive option in my opinion), then that would be appreciated. But for right now, we can't afford to spend a lot of more time on it. If disabling SIP is a problem for you, then I'd recommend avoiding ROS 2 on macOS for now, or work around it manually as you have.

P.S. running `ros2` is not really indicative of whether or not the system is fixed since that's a Python script that doesn't really load any of our libraries. Running `talker` in `demo_nodes_cpp` might be more compelling.

wjwwood, 2018-02-02:
See also: https://github.com/ros2/ros2/issues/409

dirk-thomas, 2019-10-11:
> These steps can be put in a post-install script to be run after ROS2 is unpacked to final location.

Just for the record in case it is not obvious: that will make the artifact non-relocatable.

wjwwood, 2019-10-11:
Actually, it should be possible to make almost all, if not all, of the RPATH's relative. Here's a reference that talks about a few of the options like `@executable` and `@load` paths:

https://wincent.com/wiki/@executable_path,_@load_path_and_@rpath

dirk-thomas, 2019-10-11:
If anyone using macOS could create a pull request with this that would be great.

rotu, 2019-11-30:
I think https://github.com/ament/ament_cmake/pull/212 may suffice to fix this.