ros2/ros2 #458: Refactor uses of uint64_t/int64_t as time with rcutils_time_point_t or similar

wjwwood, 2018-02-01:
As a follow up to https://github.com/ros2/rcutils/issues/79#issuecomment-362412816, we should try to use a typedef for types that are uses as time points (or durations) in all C/C++ api's, reusing, or using an alias to, one of the `rcutils` time typedefs.

This will convey more context than a bare primitive type and it will allow us to more easily change the storage of the time type going forward (even if we don't foresee needing to do so right now).

The slight downside of this is that people looking at a higher level function will need to trace the type down to it's primitive type in order to know the storage size, but perhaps we can address that with better documentation tools (e.g. if we could make doxygen show the "a.k.a." type next to typedefs or something like what compiler errors do).

jlblancoc, 2018-02-01:
Just curious (I'm sure this might have been discussed  already, but couldn't find it): why not directly using std::chrono duration and time points? Backwards compatibility? Pure C compatibility?

wjwwood, 2018-02-01:
Pure C, but since this is on Windows only, and since there probably isn't any use case for pure C on Windows, it might be fine to use C++11's time constructs instead.

For Unix systems (and more embedded use cases down the line), however, the pure C is nice, because it doesn't depend on the C++ runtime which can be pretty big.

wjwwood, 2018-02-01:
Sorry I was getting this confused with https://github.com/ros2/rcutils/issues/86.

The answer here is, yes pure C because `rcutils`, `rmw`, and `rcl` are pure C, so we can't use `std::chrono` in the interfaces there.

I'm not advocating for replacing usage of `std::chrono` with our time stuff, only that we use our time typedefs where we already pass `uint64_t` or `int64_t` through pure C API's already.

jlblancoc, 2018-02-01:
👍 then. 
Yes, using named typedef's will be really clarifying for API users.

clalancette, 2023-03-30:
I believe in the intervening years we've actually done a bunch of this (if not everywhere).  So I'm going to close this one out, but feel free to reopen if you think otherwise.

closed 2023-03-30 by clalancette