ros2/ros2 #460: ROS2 build issue

PRADEEPKJ, 2018-02-07:
Hi All,

I am trying to build the ROS2 following the steps those are mentioned.
Phasing the below issue. Please help me to resolve issue.

Thanks & Regards,
Pradeep

-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rcutils: 0.4.0 (/home/robot/ros2_ws/install/share/rcutils/cmake)
-- Found fastrtps_cmake_module: 0.4.0 (/home/robot/ros2_ws/install/share/fastrtps_cmake_module/cmake)
CMake Error at CMakeLists.txt:34 (find_package):
  Could not find a package configuration file provided by "fastrtps" with any
  of the following names:

    fastrtpsConfig.cmake
    fastrtps-config.cmake

  Add the installation prefix of "fastrtps" to CMAKE_PREFIX_PATH or set
  "fastrtps_DIR" to a directory containing one of the above files.  If
  "fastrtps" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/robot/ros2_ws/build/rmw_fastrtps_cpp/CMakeFiles/CMakeOutput.log".

dirk-thomas, 2018-02-07:
Please provide the exact steps you have done before seeing this output.

PRADEEPKJ, 2018-02-07:
Followed the steps from here https://github.com/ros2/ros2/wiki/Linux-Development-Setup

-Pradeep

dirk-thomas, 2018-02-07:
Please still post the exact steps and the complete output since I would expect that following the tutorial won't lead to the above error message.

PRADEEPKJ, 2018-02-07:
Exactly I followed the steps from that link. The below step/cli option failed to compile. src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install

mikaelarguedas, 2018-02-07:
Can you please provide the result of `history` in your terminal as well as the full build console output (in a gist) ?

By the look of it it looks like fastrtps was not built properly. But it is hard to investigate why without more information.

Do you have a folder called `Fast-RTPS` under `<MY_ROS2_WS>/src/eProsima` ?
When you run `ament build` does fastrtps shows up in the topological order displayed at the beginning ?
If you run `ament build --force-cmake --only fastrtps` what is the console output ? is the package and its libraries built properly?

PRADEEPKJ, 2018-02-07:
Thanks for the reply, FastRTPS is not cloned. 
Could not clone repository 'https://github.com/eProsima/Fast-RTPS.git': Cloning into '.'...
error: RPC failed; curl 56 GnuTLS recv error (-110): The TLS connection was non-properly terminated.

mikaelarguedas, 2018-02-07:
Fast-RTPS is a pretty big repository (~70MB) so it's the most likely to fail to clone if your internet connection is not stable. I'd retrying the cloning step until it succeeds for all repositories and then try to build your workspace

mikaelarguedas, 2018-02-08:
I'm going to consider this issue diagnosed and will thus close this ticket. Please feel free to comment here if you still face problems building after successfully cloning the repository and we can reopen this issue

closed 2018-02-08 by mikaelarguedas

wang20xiancai, 2024-09-16:
like this shell
git clone https://github.com/ros2/rmw_fastrtps ros2/rmw_fastrtps -b humble
git clone https://github.com/eProsima/Fast-DDS -b 2.6.2 eProsima/fastrtps

rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-6.0.1 urdfdom_headers" --rosdistro humble