ros2/ros2 #464: Needs decision: Latching

wjwwood, 2018-02-22:
This have come up several times in the past, but in summary DDS's durability setting doesn't behave exactly like latching in ROS 1. So, the question that needs a decision is "do we just use DDS's notion of durability, or implement our own latching mechanism, or both"?

TODO fill in details

wjwwood, 2018-02-23:
I updated the original issue description with a lot more detail. @ros2/team please have a look as I'll bring it up for discussion at the next meeting.

sloretz, 2018-02-27:
`tf2_ros` has an API that creates its own publishers and subscribers for `/tf_static`. I think it would be fine for it to set Transient Local on both sides.

For a topic like `/map` that is used in a lot of different packages, making a publisher use latching is about the same amount of work as the publisher using Transient Local. Is the cost of making every user set Transient Local when subscribing to that kind of topic greater than the cost of implementing our own latching mechanism?

paulbovbel, 2018-05-05:
Stumbled on this, so I thought I could offer up a user perspective. While it was a little surprising that latching had to be set sub-side as well as pub-side in ROS2, I don't think it's particularly cumbersome given proper documentation.

DDS durability feels 'good enough' in terms of matching ROS1 latch functionality. In terms of history vs queue length, in a latched scenario, I can imagine that any messages not important enough to be durable, are unimportant enough to be dropped off the publisher queue.

The durability contract being a two-sided issue also makes sense, since while the publisher must offer up the data, the subscriber should opt-in to handle not just future data, but whatever past data the publisher has in history.

I do like the idea of setting all publishers to be 'Transient Local' by default and avoid impedance mismatch, but I imagine some publishers would still want to opt-into 'Volatile' avoid storing history, due to any number of application-specific concerns.

clalancette, 2021-02-18:
Since the time that this issue was opened, we have clearly stuck with the DDS concept of "latched" topics; the publisher and subscriber must agree on this setting in order for true latching to work.

That being said, it would still be nice to make this work more automatically (and thus make it easier for users).  @Karsten1987 said that there is some code in `rosbag2` that kind of does this negotiation of settings, and that that code could be made more generic.  https://github.com/ros2/ros2cli/issues/594 hints at using the same infrastructure to do that for the CLI tools.

So I'm going to leave this old issue open in the hopes of improving the latching story in ROS 2.  That being said, we have no immediate plans to work on it so I'm going to mark it as backlog for now.

emersonknapp, 2021-02-18:
> That being said, it would still be nice to make this work more automatically (and thus make it easier for users). @Karsten1987 said that there is some code in `rosbag2` that kind of does this negotiation of settings, and that that code could be made more generic. [ros2/ros2cli#594](https://github.com/ros2/ros2cli/issues/594) hints at using the same infrastructure to do that for the CLI tools.

Yes, the rosbag2 subscriptions detect all existing publishers and subscribe with a QoS that is guaranteed to match them all. Generalizing that functionality to the ROS 2 core is tracked as https://github.com/ros2/rosbag2/issues/601 - not sure if we want a ros2/ros2 ticket for it instead/additionally

clalancette, 2021-02-19:
> Generalizing that functionality to the ROS 2 core is tracked as [ros2/rosbag2#601](https://github.com/ros2/rosbag2/issues/601) - not sure if we want a ros2/ros2 ticket for it instead/additionally

Ha, I had forgotten about that issue, thanks for reminding me.

I still think it makes sense to leave this one open.  The rosbag2 issue is about generalizing the code, but there would still be follow-up work elsewhere to use that code as appropriate.  So I'm going to leave this one open for now.

dcconner, 2023-07-27:
Can I request that we add this discussion to https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html

Especially the results column in table above.  The current table of "compatible" is misleading, and we need a discussion of latching front and center.

clalancette, 2023-07-27:
Please feel free to open a PR against the documentation at https://github.com/ros2/ros2_documentation, and we can iterate there.

dcconner, 2023-07-27:
@clalancette fair enough.  https://github.com/ros2/ros2_documentation/pull/3804