ros2/ros2 #468: Support for static linking

dirk-thomas, 2018-03-01:
This is a follow up of #306 to have a clear overview of which PRs are currently being considered to address this.

mikaelarguedas, 2018-03-01:
Can you also provide here the way you're testing it for future reference? thanks!

dirk-thomas, 2018-03-01:
> Can you also provide here the way you're testing it for future reference? thanks!

I am still working on this and will shortly provide a `.repos` file as well as instructions to reproduce the build.

mikaelarguedas, 2018-03-01:
> I am still working on this and will shortly provide a `.repos` file as well as instructions to reproduce the build.

Awesome, thanks

dirk-thomas, 2018-03-02:
This [ros2.repos](https://gist.github.com/dirk-thomas/67cbdafdff895755badc2d611b90fc1d) file references the branches used by the referenced PRs and removes some repositories for now (see the history of the gist for the diff).

After cloning those (with submodules for FastRTPS: `vcs import --recursive src < ros2.repos`) a couple of changes are currently still necessary (mostly copied from ament/ament_cmake#114):
* Remove Python generator from workspace
  * `rm -fr src/ros2/rosidl/rosidl_generator_py/`
* Comment out Python generator in CMake
  * `sed -e '/py/ s/^#*/#/' -i src/ros2/rosidl_typesupport/rosidl_default_generators/CMakeLists.txt`
* Remove the `test_msgs` package since it uses a Python specific path for installation
  * `rm -fr src/ros2/rcl_interfaces/test_msgs/`

Optionally you can apply the following patch (but this is not strictly necessary):
* Modify FastRTPS to build shared libraries
  * `sed -i -r '/\$\{ANDROID\_BUILD\_OPTIONS\}/a "-DBUILD\_SHARED\_LIBS=off"' src/eProsima/Fast-RTPS/cmake/dev/eprosima_libraries.cmake`

With this setup I am able to build the workspace with static libraries (`--cmake-args -DBUILD_SHARED_LIBS=0`) up to `rcl` successfully. It currently fails with linker errors for `rclcpp`. I will look into those next but expect that similar patch than the already referenced should fix the problem.

dirk-thomas, 2018-03-02:
I just updated the patch for `rcl`, created ros2/rclcpp#446 and modified the `rclcpp` branch in the gist. Now `rclcpp` passes the build too.

dirk-thomas, 2018-03-02:
After updating the dependency order in the rcl and rclcpp lifecycle packages (ros2/rcl#216, ros2/rcl#446) the whole workspace builds fine for me.

dirk-thomas, 2018-03-02:
@lmayencourt Can you please try to reproduce my result with the above instructions.

esteve, 2018-03-02:
FWIW, I've been building ROS2 statically up to `rcl` for a while (see https://github.com/esteve/ros2_objc#ios, iOS needs binaries to be built statically). It's been some time since I last tested it, but I'm curious about what issues @dirk-thomas found in the order in which dependencies were exported in `rcl` that required  https://github.com/ros2/rclcpp/pull/446 Was it to avoid circular linking dependencies in the examples? Thanks.

dirk-thomas, 2018-03-02:
> FWIW, I've been building ROS2 statically up to rcl for a while (see https://github.com/esteve/ros2_objc#ios, iOS needs binaries to be built statically).

Can you provide up-to-date instructions to reproduce this on a Linux machine. Currently the referenced instructions fail to clone the repos (e.g. `rclutils` doesn't exist). I would be happy to try to figure out what the differences are and if they need to be considered for a to-be-merged set of patches.

> Was it to avoid circular linking dependencies later in the examples?

Afaik there are no circular linking dependencies in ROS 2. I would even expect that to result in problems with shared linking on some platforms (e.g. Windows which doesn't even allow unresolved symbols in a shared library).

> what issues @dirk-thomas found in the order in which dependencies were exported in rcl that required ros2/rcl#216.

I started from the instructions provided in ament_cmake#114 and worked my way through the linker errors. From what I notices the order of the linked libraries was not "correct". E.g. a library `A` was linked before `B` even though `B` requires symbols from `A`. By swapping the order and linking `B` before `A` it was able to resolve the symbol correctly.

dirk-thomas, 2018-03-02:
@ros2/team I have places this and the referenced PRs in review to get some feedback on the general idea.

Before considering to merge it I want to wait for external feedback / confirmation that it addresses the specific use cases.

lmayencourt, 2018-03-05:
Hi @dirk-thomas,
Thanks for the update on this "static linking" topics ! I will try this and come back to you !

lmayencourt, 2018-03-05:
Your changes looks good to me ! 
I'm able to build the whole workspace in static (except for `libament_index_cpp.so`) and the `exmaples_rclccp_minimal_publishier` only have dynamic link to the system libraries. This is great !
Thanks for iterating on this.

esteve, 2018-03-05:
> Can you provide up-to-date instructions to reproduce this on a Linux machine. Currently the referenced instructions fail to clone the repos (e.g. rclutils doesn't exist). I would be happy to try to [...]

Sure, I've updated the repos files in https://github.com/esteve/ros2_objc and cleaned up the README a little.

I use the following on Linux for building ROS2 statically up to `rcl`:

```src/ament/ament_tools/scripts/ament.py build --isolated --parallel  --cmake-args -DTHIRDPARTY=ON -DCOMPILE_EXAMPLES=OFF -DBUILD_SHARED_LIBS=OFF -- --end-with rcl```

Additionally, if the user wants to mix parts of the ROS2 that are statically built with parts that are shared (e.g. if building the Python bindings), `-DCMAKE_POSITION_INDEPENDENT_CODE=ON` needs to be passed to `--cmake-args` (`-fPIC` is not enabled when building libraries statically), so perhaps it'd make sense to add it to `ament_cmake_ros` so that it's enabled by default (much like `BUILD_SHARED_LIBS`).

dirk-thomas, 2018-03-08:
@ros2/team Waiting for review on the referenced PRs.

mikaelarguedas, 2018-03-08:
Just to confirm my understanding for testing, this should work up to rclcpp now right ?

dirk-thomas, 2018-03-08:
> this should work up to rclcpp now right ?

Correct.

mikaelarguedas, 2018-03-08:
> >  this should work up to rclcpp now right ?

> Correct.

Sounds good, I can give it a shot (likely tomorrow us no-one gets to it before then)

mikaelarguedas, 2018-03-09:
I got it to build, I'll review the changes tomorrow. One global remark: each file where the order of the libraries is not alphabetical could use a comment explaining why the order is what it is, otherwise we may accidentally change the order in the future and won't notice the failure as we don't build statically on CI.

mikaelarguedas, 2018-03-09:
Can you clarify how you picked which libraries would be built statically and which one we keep building as shared in this set of PRs?
A few outstanding examples: 
- We link statically against thirdparty shared objects (as opposed to @esteve's approach of building Fast-RTPS dependencies statically, e.g. tinyxml2)
- The libraries built for testing are linked dynamically (and fail to link in the case of rcl and rclcpp)
  - suggestion for rcutils [here](https://github.com/ros2/rcutils/pull/95)
- ament_index_cpp still created a shared library (suggestion [here](https://github.com/ament/ament_index/pull/28) that doesnt introduce a dependency on `ament_cmake_ros`)

I opened these PRs as example to get feedback before looking into the other linker failures and libraries still linked dynamically

dirk-thomas, 2018-03-09:
I was just focusing on the use case from ARM to get that moving forward and get patches which are actually mergable.

Applying this to additional cases (like the memory tools and ament index) makes total sense.

mikaelarguedas, 2018-03-09:
Sounds good, I'm not sure if I'll have time to dig into the rcl and rclcpp current linking errors when building the tests. We can do it in a follow-up as we keep moving up the stack.

> Additionally, if the user wants to mix parts of the ROS2 that are statically built with parts that are shared (e.g. if building the Python bindings), `-DCMAKE_POSITION_INDEPENDENT_CODE=ON` needs to [...]

I think it makes sense to have this in `ament_cmake_ros` :+1: . As I'm not familiar with the potential side effects, I'd like to hear what the rest of @ros2/team think about it.

mikaelarguedas, 2018-03-14:
I reviewed all the related PRs and I think this is good to go. Some follow-up issues and discussions should be ticketed though

dirk-thomas, 2018-03-22:
Let me try to summarize the exact changes playing into this. That way future readers can more easily follow the discussion...

The first set of changes uses `ament_cmake_ros` to select the build type of the libraries rather than hard coding one type:
* `rcutils`
  * ros2/rcutils#94 compared to https://github.com/esteve/rcutils/tree/honor-build-shared-libs also applies the `BUILD_SHARED_LIBRARIES` to the memory tools library.
* `rosidl`
  * The different between ros2/rosidl#266 and https://github.com/esteve/rosidl/tree/honor-build-shared-libs is that the former uses the `BUILD_SHARED_LIB` setting which was used for the typesupport package for all libraries which are later built for that typesupport within downstream message packages. This is the better approach since different typesupport packages could be built with a different setting and that should not "collide" in a message package but only affect the target for that typesupport. Basically the message package can still decide on its own targets.
* `rosidl_typesupport`
  * The different between ros2/rosidl_typesupport#19 and https://github.com/esteve/rosidl_typesupport/tree/disable-multiple-typesupport-static is the same as for the `rosidl` repo.
Other than the described advantages the two different approaches are identical. I am convinced these are necessary and correct changes. The PRs referenced from this ticket are preferred due to their better handling of mixed builds.

Beside these changes I use the default content provided by a current [ros2.repos](https://github.com/ros2/ros2/blob/master/ros2.repos) file. In order to build this workspace up to `rcl` I had to further:
* remove `rmw_connext` and `rmw_opensplice` from the workspace
* ignore `rosidl_generator_py`
* comment out `rosidl_generator_py` in `rosidl_default_generators/CMakeLists.txt`

When building this workspace with the following comand the result was the same in both cases - linking errors due to missing symbols for the `rcl` library target:

> colcon build --base-paths src --packages-up-to rcl --cmake-args \ -DBUILD_SHARED_LIBS=OFF

[While I am using `colcon` for this build (since `--packages-up-to` benefits a lot from ignoring packages not in the path towards `rcl`] this can also be done with `ament_tools` with more time and manual removal of packages which don't support static building like `class_loader`.]

These results don't support the assumption that it is somehow possible to build without further changes.

@esteve After trying this again and doing the comparison workspace I don't see how your changes can link successfully without the reordering described below. They don't when I try to reproduce your result with the above instructions. If you can demonstrate otherwise I am more than curious to understand why. Please avoid any further changes in the process form the above to keep it easy to compare the results. E.g. I didn't fetch submodules for FastRTPS or anything else not mentioned in this comment.

---

That being said let summarize the second set of PRs referenced in this PR which address the ordering problem we ran into:
* ros2/rmw_fastrtps#190, ros2/rcl#216, ros2/rclcpp#446 change the order how dependencies are being linked and/or exported
* ament/ament_cmake#123 is only in support of the ordering of the PRs from the previous bullets.

With those changes applied to the workspace(s) both link successfully up to `rcl`.

---

Based on this additional investigation I am convinced the proposed changes make sense and will go ahead and merge the already approved PRs. @esteve Please feel free to still comment with your finding and I am more than happy to revert the reordering patches is it turns out that we can somehow avoid them.

closed 2018-03-22 by dirk-thomas