ros2/ros2 #469: Building failure due to unexpected cmake configuration

yenWu, 2018-03-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04 
- Installation type:
  - From source 
- Version or commit hash:
  - ros2/ros2: d89dcc090a65db69c6a30c663f7ae7d5f68bee35
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
# wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
wget https://raw.githubusercontent.com/ros2/ros2/d89dcc090a65db69c6a30c663f7ae7d5f68bee35/ros2.repos
vcs-import src < ros2.repos
cd ~/ros2_ws/
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install
```

#### Expected behavior
Building pass
#### Actual behavior
```
Process package 'urdfdom_headers' with context:
--------------------------------------------------------------------------------
 source_space => /home/yenkuan/ros2_ws/src/ros/urdfdom_headers
  build_space => /home/yenkuan/ros2_ws/build/urdfdom_headers
install_space => /home/yenkuan/ros2_ws/install
   make_flags => -j8, -l8
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'urdfdom_headers'
==> '. /home/yenkuan/ros2_ws/build/urdfdom_headers/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/yenkuan/ros2_ws/build/urdfdom_headers'
CMake Error: The source directory "/home/yenkuan/ros2_ws/src/ros2/urdfdom_headers" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:180: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

<== Command '. /home/yenkuan/ros2_ws/build/urdfdom_headers/cmake__build.sh && /usr/bin/make cmake_check_build_system' failed in '/home/yenkuan/ros2_ws/build/urdfdom_headers' with exit code '2'
<== Command '. /home/yenkuan/ros2_ws/build/urdfdom_headers/cmake__build.sh && /usr/bin/make cmake_check_build_system' failed in '/home/yenkuan/ros2_ws/build/urdfdom_headers' with exit code '2'

```

#### Additional information
I already use `git bitsec` to find the commit leading to the failure. It's fine on f21563f. Failed on d89dcc0.

mikaelarguedas, 2018-03-05:
Have you tried to build this in a fresh workspace? it looks like your build is expecting `urdfdom_headers` in the `ros2` folder. But the repos file you are using is cloning it in the `ros` directory.

This can happen only if your cmake cache is not clean or if you're workspace has not been cleaned and CMake is not running. Based on your console output CMake is not being ran on that package.
Can you try to either remove your build folder or to run your build using the `--force-cmake-configure` flag in your ament invocation?

yenWu, 2018-03-06:
You're right. I try the fresh one, and it works like a charm.
I don't remove the `build/`, so it leave the cmake cache in that. Thks for ur help!

mikaelarguedas, 2018-03-06:
You're welcome, glad it works now. Closing this issue then.

closed 2018-03-06 by mikaelarguedas