ros2/ros2 #470: Installation from source is not working, fails when building curl-7.57.0

kutschkem, 2018-03-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04 VM on Windows 7 host
- Installation type:
  - source
- Version or commit hash:
  - d1e484dfab4bca84d62f29ebf98ed607f0f0fba7  (release-latest as of 2018-03-06)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

On a fresh vagrant box "ubuntu/xenial64" with git and cmake 3.10.2 installed, follow the installation instructions on https://github.com/ros2/ros2/wiki/Linux-Development-Setup

The box is using a cntlm proxy on the host.

Vagrantfile

```
Vagrant.configure("2") do |config|
  config.vm.box = "ubuntu/xenial64"
  
  config.vm.network "private_network", ip: "192.168.50.4"
  
  config.vm.provider "virtualbox" do |v|
      v.name = "Ubuntu_ecal5"
      v.memory = 2048
  end
  
  if Vagrant.has_plugin?("vagrant-proxyconf")
    config.proxy.http = "http://10.0.2.2:3128"  
    config.proxy.https = "http://10.0.2.2:3128"  
    config.proxy.no_proxy = "localhost,127.0.0.1"  
  end
  
  #set up keyboard layout (de)
  config.vm.provision "shell", inline: "sed -ie '/^XKBLAYOUT=/s/\".*\"/\"de\"/' /etc/default/keyboard && udevadm trigger --subsystem-match=input --action=change"
  
  config.vm.provision "shell", inline: "apt-get update && apt-get -y install unzip make git"
  config.vm.provision "shell", path: "scripts/install_gcc.sh" 	# install gcc-5 and g++-5
  config.vm.provision "shell", path: "scripts/install_cmake.sh" # build and install cmake 3.10.2
  
  config.vm.provision "shell", path: "scripts/install_ros.sh"
  
    
end
```

script/install_gcc.sh

```
set -e

sudo apt-get update
sudo apt-get -y install gcc-5 g++-5
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-5 60 --slave /usr/bin/g++ g++ /usr/bin/g++-5
```

scripts/install_cmake.sh
```
set -e
# download cmake
version=3.10
build=2
mkdir ~/temp
cd ~/temp
wget https://cmake.org/files/v$version/cmake-$version.$build.tar.gz
tar -xzvf cmake-$version.$build.tar.gz
cd cmake-$version.$build/

# build cmake
./bootstrap
make -j2
sudo make install
```

scripts/install_ros.sh
```
set -e

sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros-latest.list'
sudo apt-key adv --keyserver ha.pool.sks-keyservers.net --recv-keys 421C365BD9FF1F717815A3895523BAEEB01FA116

sudo apt-get update
sudo apt-get -y install git wget
sudo apt-get -y install build-essential cppcheck cmake libopencv-dev python-empy python3-dev python3-empy python3-nose python3-pip python3-pyparsing python3-setuptools python3-vcstool python3-yaml libtinyxml-dev libeigen3-dev
# dependencies for testing
sudo apt-get -y install clang-format pydocstyle pyflakes python3-coverage python3-mock python3-pep8 uncrustify
# Install argcomplete for command-line tab completion from the ROS2 tools.
# Install from pip rather than from apt-get because of a bug in the Ubuntu 16.04 version of argcomplete:
yes | sudo pip3 install argcomplete
# additional testing dependencies from pip (because not available on ubuntu 16.04)
yes |sudo pip3 install flake8 flake8-blind-except flake8-builtins flake8-class-newline flake8-comprehensions flake8-deprecated flake8-docstrings flake8-import-order flake8-quotes pytest pytest-cov pytest-runner
# dependencies for FastRTPS
sudo apt-get -y install libasio-dev libtinyxml2-dev

mkdir -p ~/ros2_ws/src
cd ~/ros2_ws
wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
vcs-import src < ros2.repos

cd ~/ros2_ws/
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install
```

#### Expected behavior

Successfull installation.

#### Actual behavior

Error when building target curl-7.57.0

```
    default:     error: downloading 'https://github.com/curl/curl/releases/downl
oad/curl-7_57_0/curl-7.57.0.tar.gz' failed
    default:          status_code: 1
    default:          status_string: "Unsupported protocol"
    default:          log:
    default:          --- LOG BEGIN ---
    default:          Protocol "https" not supported or disabled in libcurl
    default:
    default:   Closing connection -1
    default:
    default:
    default:
    default:          --- LOG END ---
``` 

#### Additional information

Using curl on the commandline to download https://github.com/curl/curl/releases/download/curl-7_57_0/curl-7.57.0.tar.gz works without problem, so I wouldn't expect the problem to be an issue with the installed version of curl.

kutschkem, 2018-03-06:
The problem is apparently with cmake: https://gitlab.kitware.com/cmake/cmake/issues/17240
I will try if the build works with cmake 3.7.2, and report back.
If it turns out that this is the issue, maybe changing the downloads from https to http will help.

wjwwood, 2018-03-06:
Interesting, we're using cmake 3.10.x (not sure of the patch version) on our Windows CI and maybe our macOS CI machines too (you mentioned you're using 3.10.2 above). Does this only affect linux on such a high version of cmake?

mikaelarguedas, 2018-03-06:
IIRC we use 3.10.2 on both MacOS and Windows

kutschkem, 2018-03-06:
This bug report looks like it affects Windows too, but I#m not 100% sure it's the same issue.: https://gitlab.kitware.com/cmake/cmake/issues/17592

kutschkem, 2018-03-06:
By the way, building with cmake 3.7.2 seemed to work.

kutschkem, 2018-03-07:
The mistake was that I needed to build CMake with SSL support

```
set -e

sudo apt-get -y install openssl libssl-dev

# download cmake
version=3.10
build=2
mkdir ~/temp
cd ~/temp
wget https://cmake.org/files/v$version/cmake-$version.$build.tar.gz
tar -xzvf cmake-$version.$build.tar.gz
cd cmake-$version.$build/

# build cmake
./bootstrap -- -DCMAKE_USE_OPENSSL=ON
make -j2
sudo make install
```

Now the downloads work. Sadly this was not well documented on the side of cmake :-(

closed 2018-03-07 by kutschkem

Jrokisky, 2019-04-08:
@kutschkem ran into this same issue. Thank you for documenting the solution.