ros2/ros2 #472: Ardent patch release 2

mikaelarguedas, 2018-03-07:
The main goal of this release is to provide bug fixes for sros2 access control. No other package will be rev'd.

- [ ] merge fix in ardent branch: 
- [ ] tag new version of the repo
- [ ] update repos file and tag
- [ ] Check if MacOS and Windows nodes need to be updated
  - MacOS 2 package used in our code our outdated:
    - Packages directly used in core code and demos:
      - numpy: 1.14.0 -> 1.14.1_1
      - opencv: 3.4.0_1 -> 3.4.1_2
      - python: 2.6.4_2 -> 3.6.4_3
    - Packages related to RViz? (@wjwwood do you think we should update those before packaging?)
      - qt: 5.10.0_1 -> 5.10.1
      - jpeg 9b => 9c
      - libtiff 4.0.9_1 -> 4.0.9_2
      - ffmpeg 3.4.1 -> 3.4.2
    - Other packages (I'm not planning on updating those for this patch release):
      - ccache (latest HEAD)
      - gawk: 4.2.0_1 -> 4.2.1
      - libunistring: 4.5 -> 4.5.2 
      - mercurial 4.5 -> 4.5.2
      - mpfr 4.0.0 -> 4.0.1
      - x264 r2795 -> r2854
  - Windows
    - Java: jre8 8.0.144 -> 8.0.161
    - Packages used for testing:
      - cppcheck 1.81.0 -> 1.82.0
    - Packages used for tutorials:
      - curl 7.58.0 -> 7.58.0.20180213
      - git 2.15.1.2 -> 2.16.2
      - peazip: 6.5.0 -> 6.5.1
      - wget 1.19.1 -> 1.19.4
   - other packages (I'm not planning on updating those):
      - hg 4.4.2 -> 4,5
      - putty
      - vcredist140
- [ ] Kick-off packaging jobs (this is just to provide archives with the bugfix, as the bugfix only impact Connext, no testing of the archive is required)
- [ ] Make a patch release of sros2
- [ ] Do the bloom release
- [ ] Upload the archives on the ros2 repo undder a new release
- [ ] Sync packages to main
- [ ] Announce patch release on Discourse

wjwwood, 2018-03-07:
I believe any change to the version of dependencies from Homebrew will break our binaries, so yeah I guess those should be updated before doing packaging again.

mikaelarguedas, 2018-03-07:
> so yeah I guess those should be updated before doing packaging again.

Sounds good, thanks :+1:

mikaelarguedas, 2018-03-07:
Windshield test job with new jre: https://ci.ros2.org/view/All/job/test_ci_win/26

mikaelarguedas, 2018-03-08:
Mini1: job for ardent after package updates: [image]. (test failures are linter errors that have been fixed since and not backported to ardent, the CMake warning is a false positive)

icecube: [image]. (test failures are linter errors that have been fixed since and not backported to ardent)
windshield: [image]. (test failures are linter errors that have been fixed since and not backported to ardent + the usual flaky composition test)

mikaelarguedas, 2018-03-08:
The new python formulae apparently doesnt give pip anymore. I'm taking mini2 offline for the night and will hopefully fix it tomorrow

mikaelarguedas, 2018-03-08:
needs some sanity testing on the artifacts, then we should be good to go and upload the new release

wjwwood, 2018-03-09:
I just tested this before lunch on macOS at @mikaelarguedas's request and I found that our python scripts are not working because the `#!` lines use `/usr/bin/env` which strips `DYLD_LIBRARY_PATH`.

Since we're rewriting the `#!` lines already, I believe @mikaelarguedas is going to change it for macOS to point to `/usr/local/bin/python3` rather than doing `/usr/bin/env python3` until we can fix the linking issues caused by SIP.

mikaelarguedas, 2018-03-09:
I ran another packaging job with [the suggested fix](https://github.com/ros2/ci/pull/127), @wjwwood tested it and it fixes the linking issue for him.

New packaging jobs: MacOS ([fastrtps](https://ci.ros2.org/view/packaging/job/packaging_osx/938/), [opensplice](https://ci.ros2.org/view/packaging/job/packaging_osx/939), [both](https://ci.ros2.org/view/packaging/job/packaging_osx/940/))

mikaelarguedas, 2018-03-10:
For windows testing (thanks @sloretz for giving it a shot):
- system updates
- in a root terminal:
  - choco outdated
  - choco upgrade -y
  - pip install --upgrade (whatever is listed on the install page)
  - download the latest [OpenSSL 1.0.2n](https://slproweb.com/download/Win64OpenSSL-1_0_2n.exe)
  - download the [fat archive](https://ci.ros2.org/view/packaging/job/packaging_windows/931/artifact/ws/ros2-package-windows-AMD64.zip)
  - extract it
  - source the workspace
  - run `ros2 run demo_nodes_py talker` and `ros2 run demo_nodes_cpp listener` and see if they're happy chatting

mikaelarguedas, 2018-03-10:
Test sros2 patch (file generation only):
- source your workspace
- md C:\dev\ros2\sros2_demo
- cd C:\dev\ros2\sros2_demo
- ros2 security create_keystore demo_keys
- ros2 security create_key demo_keys talker
- Open C:\dev\ros2\sros2_demo\demo_keys\talker\permissions.xml and make sure it contains the following lines
<details>

``` xml
        <publish>
          <partitions>
            <partition></partition><partition>rq/talker</partition>
          </partitions>
          <topics>
            <topic>get_parametersRequest</topic><topic>get_parameter_typesRequest</topic><topic>set_parametersRequest</topic><topic>list_parametersRequest</topic><topic>describe_parametersRequest</topic>
          </topics>
        </publish>
        <subscribe>
          <partitions>
            <partition></partition><partition>rr/talker</partition>
          </partitions>
          <topics>
            <topic>get_parametersReply</topic><topic>get_parameter_typesReply</topic><topic>set_parametersReply</topic><topic>list_parametersReply</topic><topic>describe_parametersReply</topic>
          </topics>
        </subscribe>
```

</details>

- curl -sk https://raw.githubusercontent.com/ros2/sros2/ardent/examples/sample_policy.yaml -o .\demo_keys\policies.yaml
- ros2 security create_permission demo_keys talker demo_keys/policies.yaml
- Open C:\dev\ros2\sros2_demo\demo_keys\talker\permissions.xml and make sure it contains the following lines
<details>

``` xml
        <publish>
          <partitions>
            <partition>rt</partition>
          </partitions>
          <topics>
            <topic>chatter</topic>
          </topics>
        </publish>
        <subscribe>
          <partitions>
            <partition>rt</partition>
          </partitions>
          <topics>
            <topic>clock</topic>
          </topics>
        </subscribe>
        <publish>
          <partitions>
            <partition></partition><partition>rq/talker</partition>
          </partitions>
          <topics>
            <topic>get_parametersRequest</topic><topic>get_parameter_typesRequest</topic><topic>set_parametersRequest</topic><topic>list_parametersRequest</topic><topic>describe_parametersRequest</topic>
          </topics>
        </publish>
        <subscribe>
          <partitions>
            <partition></partition><partition>rr/talker</partition>
          </partitions>
          <topics>
            <topic>get_parametersReply</topic><topic>get_parameter_typesReply</topic><topic>set_parametersReply</topic><topic>list_parametersReply</topic><topic>describe_parametersReply</topic>
          </topics>
        </subscribe>
```

</details>

sloretz, 2018-03-10:
@mikaelarguedas 

I followed your instructions, + checked for system updates. 

`demo_nodes_py talker` and `demo_nodes_py listener` happily chat

The first part of testing the sros2 patch worked fine. In the second part `permissions.xml` also includes rules for allowing publishing and subscribing `parameter_events` in the partition `rt`. Is that expected?

The `curl` command wouldn't write a file for me, or output anything to stdout when I removed `-o .\demo_key\policies.yaml`. I downloaded it using the browser, but before that I noticed that the output to stdout from `ros2 security create_permission ...` appears the same whether the file `policies.yaml` exists or not. The command should probably error if the given policy file doesn't exist. Currently it produces a permissions.xml file allowing everything.

mikaelarguedas, 2018-03-10:
> The first part of testing the sros2 patch worked fine. In the second part permissions.xml also includes rules for allowing publishing and subscribing parameter_events in the partition rt. Is that expected?

Yes this is expected now that all cpp nodes can both publish parameter events (from before ardent) and receive some (as of Ardent the node can get information about use_sim_time and use the clock accordingly).

> The curl command wouldn't write a file for me, or output anything to stdout when I removed -o .\demo_key\policies.yaml. 

Looking a bit more into this it seems that the version of curl on @sloretz VM doesn't support `https`. While that's unfortunate we won't delay the release for this as it works on our other Windows machines.
We can add a not on the README from master to provide an alternative for users experiencing the same.

> The command should probably error if the given policy file doesn't exist. Currently it produces a permissions.xml file allowing everything.

That's definitely a bug. Can you please open an issue on sros2 so that we can fix it on master?

minggangw, 2018-03-13:
Hi there, we are developing the [rclnodejs](https://github.com/RobotWebTools/rclnodejs), which is a Node.js client of ROS2. I have a question here, is there a certain period or plan when the ROS2 will get updated (like **Ardent patch release 1&2**) because we want to align with the stable release when publishing our package and it's better to know the plan in order to arrange our releases. Thanks!

mikaelarguedas, 2018-03-13:
@minggangw There is no schedule for patch releases for Ardent. We will do new releases for critical bug fixes but don't expect it to be on a periodic basis.
I'm going to close this issue as the patch release 2 has [been synced](https://discourse.ros.org/t/new-packages-for-ardent-2018-03-10/4147) but I don't mean to stop the discussion, feel free to keep commenting on the ticket after it's closed.

closed 2018-03-13 by mikaelarguedas

halton, 2018-03-14:
@mikaelarguedas so for the major release, still follow 2 per year cycles that described at https://github.com/ros2/ros2/wiki/Roadmap?

dirk-thomas, 2018-03-21:
> so for the major release, still follow 2 per year cycles that described at https://github.com/ros2/ros2/wiki/Roadmap?

@halton The roadmap page you referenced doesn't mention a 2 year release cycle anywhere. As it states there the next release is planned for June 2018 which is ~6 months after the previous ROS 2 Ardent release in December 2017. We are planning to keep that cycle time - a new ROS 2 release every 6 month.

halton, 2018-03-22:
@dirk-thomas thanks for the explanation, I was asking 2 releases per year, not 2-year release, which you already answered.