ros2/ros2 #473: Ament Tutorial "no module  module named 'em'"

ludusrusso, 2018-03-08:
Hi,
when I try to follow the ament tutorial, I get this error:

```bash
hbr@hbr:~/ros2_ws/build/rcutils$ . /home/hbr/ros2_ws/build/rcutils/cmake__build.sh && /usr/bin/make -j3 -l3
[  1%] Expanding logging_macros.h.em
[  4%] Built target gtest
[  7%] Built target rcutils_memory_tools
[ 10%] Built target gmock_main
[ 13%] Built target gtest_main
[ 16%] Built target test_memory_tools
[ 21%] Built target gmock
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named 'em'
CMakeFiles/rcutils.dir/build.make:62: recipe for target 'include/rcutils/logging_macros.h' failed
make[2]: *** [include/rcutils/logging_macros.h] Error 1
CMakeFiles/Makefile2:1424: recipe for target 'CMakeFiles/rcutils.dir/all' failed
make[1]: *** [CMakeFiles/rcutils.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

I'm trying to understand it, anyway if I install `em` with `pip3`I got this:

```bash
hbr@hbr:~/ros2_ws/build/rcutils$ . /home/hbr/ros2_ws/build/rcutils/cmake__build.sh && /usr/bin/make -j3 -l3
[  1%] Expanding logging_macros.h.em
[  4%] Built target rcutils_memory_tools
[  7%] Built target gtest
[ 10%] Built target gmock_main
[ 13%] Built target gtest_main
[ 16%] Built target test_memory_tools
[ 21%] Built target gmock
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: module 'em' has no attribute 'invoke'
CMakeFiles/rcutils.dir/build.make:62: recipe for target 'include/rcutils/logging_macros.h' failed
make[2]: *** [include/rcutils/logging_macros.h] Error 1
CMakeFiles/Makefile2:1424: recipe for target 'CMakeFiles/rcutils.dir/all' failed
make[1]: *** [CMakeFiles/rcutils.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

Any help?

mikaelarguedas, 2018-03-08:
@ludusrusso Did you install all the dependencies listed on the [installation page](https://github.com/ros2/ros2/wiki/Linux-Development-Setup#how-to-setup-the-development-environment) ? it looks like you are missing `python3-empy`

ludusrusso, 2018-03-08:
@mikaelarguedas 
you're right sorry! 
I installed `empTy` instead of `empy`!! Tnx

closed 2018-03-08 by ludusrusso

mikaelarguedas, 2018-03-08:
:+1:
In general we recommend to install the dependencies from the ubuntu apt repository rather than pip as these will update automatically when you upgrade your system (and that the dependencies listed on the install page are the ones we use for building and testing the code :) so it should be a smoother ride if you use the same for your ROS 2 development).

ludusrusso, 2018-03-08:
Ok tnx! I'll do it so!