ros2/ros2 #475: MISRA meta-ticket

mjcarroll, 2018-03-27:
## Feature request

Rework part of the ROS2 stack (likely including rcutils, rcl, rmw) to be able to be compliant with [MISRA C 2012](https://www.misra.org.uk/Publications/tabid/57/Default.aspx#label-c3)

### Minimum Requirements

To start with, we'll need to get the linters set up for individual contributors as well as the CI.  We will then bring a minimal scope of packages into MISRA compliance.  Beyond that, compliance will be pursued on a per-package basis.

- [ ] Start wiki page about MISRA compliance in ROS 2.
- [ ] Create `ament_lint` for `cppcheck` [MISRA extensions](https://github.com/danmar/cppcheck/blob/master/addons/misra.py)
- [ ] Determine how to integrate `cppcheck` MISRA into CI jobs
- [ ] Create 'ament_pclint`
- [ ] Determine how to integrate `ament_pclint` into CI jobs (with private results)
- [ ] Document how to enable the linters
- [ ] Starting packages
  - [ ] rcutils
  - [ ] rcl
  - [ ] rmw

mjcarroll, 2018-04-06:
Wiki page: https://github.com/ros2/ros2/wiki/MISRA-Compliance-Guide

wjwwood, 2018-04-06:
First glance at the wiki page looks good, though we should make an effort to be clear that we're not claiming compliance. I don't think the wiki page is saying that we are, but we could be extra explicit that we aren't. And maybe you're planning to add more to it, in which case maybe this review is too early 😄

clalancette, 2021-01-07:
As far as I know, we aren't currently pursuing MISRA compliance in ROS 2.  So I'm going to close this one out.  If we do decide to pursue MISRA further in the future, we can open a new ticket.

closed 2021-01-07 by clalancette