ros2/ros2 #478: Fixup contribution to ament tutorial

wjwwood, 2018-04-04:
@GregBurns, this is only way I can contact you about your recent change to the ament tutorial:

https://github.com/ros2/ros2/wiki/Ament-Tutorial/_compare/50f26297850069e7311cfd3414c8673584ec51d5...e3a394fb30d436818202f640c5d43dc914850002

First off, thanks for improving it. I did, however, revert your change and make a similar change which doesn't require a hard coded commit hash. Please @GregBurns and @ros2/team have a look at my change and let me know if it's acceptable:

https://github.com/ros2/ros2/wiki/Ament-Tutorial/_compare/8698a81d12411aedcd82adeaa75f9bf8de5e6070...d5c0de8a1e36ed0859343c919991a6b0f63c243f

I'll go ahead and close this, and I can reopen it if we need to further discuss it.

closed 2018-04-04 by wjwwood

mikaelarguedas, 2018-04-04:
It looks like some failures were reported because people build a master overlay of example on top of released packages. It may make sense to ask them to checkout `$ROS_DISTRO`. Though our dev branches are called `master` and not `bouncy` so it's not ideal either...

GregBurns, 2018-04-04:
Your change is better but the formatting is messed up. I just fixed it..

Closed #478<https://github.com/ros2/ros2/issues/478>.

GregBurns, 2018-04-04:
> @GregBurns<https://github.com/GregBurns> sorry, but that formatting was intentional, why do you think your latest change is better?

For consistency with the rest of the page. But if it was intentional then so be it.

Greg.

wjwwood, 2018-04-04:
Ok, the reason I wanted it to be a shell comment (starts with `#`) is that when people are blindly copy-pasting each block one after another they don't run into an issue. You'd be surprised how many times people complain to us something like:

> It broke when I did:
> `git clone https://github.com/ros2/examples.git`
> And then:
> `git clone ~/ros2_ws/src/ros2/examples`

So, If you don't mind, I'll redo it so the non-default option, i.e. `git clone https://github.com/ros2/examples.git`, is a comment which is not easily copy-paste-run'able.

wjwwood, 2018-04-04:
Re: consistency, I see what you mean, like this section:

https://github.com/ros2/ros2/wiki/Ament-Tutorial#run-the-tests

There is a choice and the choice is two code blocks with some text between them, but I'd make the slight difference in this case that running both options won't break, they just do something different. So in this case, it's really an "exclusive or".

GregBurns, 2018-04-04:
Ok I get it. But why not just always do the local git clone from the underlay, then things always work.

Greg.

wjwwood, 2018-04-04:
That's true, I suppose I could remove the note entirely. It's just that most of the time you wouldn't do that, you instead clone what you want to change from github directly so that forking and later making a pull request is easier.

@ros2/team should I just remove the note about the other option entirely.