ros2/ros2 #480: Node discovery fails with nodes on multiple systems

Ralino, 2018-04-15:
## Bug report

When running nodes distributed across multiple systems, the node discovery seems to fail occasionally: Seemingly arbitrary nodes are missing when calling `NodeGraph::get_node_names()`, nodes later subscribing to a topic those missing nodes are publishing to do not get their messages, and nodes which where already subscribed will stop receiving their messages approx. 500s after they went missing.  Once nodes go missing, they do not come back until they are restarted.

**Required Info:**

- Operating System:
   Ubuntu 16.04, same problem on Raspbian, Debian Stretch
- Installation type:
   From source
- Version or commit hash:
  Current master branch, problem persists in older versions as well (at least last month)
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
The problem seems to only occur when there are nodes on multiple systems. [I made a simple setup consisting of five systemd-nspawn containers](https://github.com/Ralino/ros2-network-test), where each container runs one node, and each one has one subscriber and one publisher on the same topic. The problem also occurs on a more complex system which runs on six Raspberry Pis.

Note that the problem does not always appear, simply stopping and restarting all ros nodes a few times should do the trick (approx 1 out of 2 times does the problem occur on the container-setup on my machine). If it still does not occur, try increasing the number of containers.

#### Sample run 1
*The behavior is not entirely consistent, so I added a description of two slightly different test runs. This is a sample test run with seven nodes, all publishing regulary and subscribed to the same topic, each one on a different system. This one was done three weeks ago on the Raspberry Pi network.*
- Nodes are started sequentially: 1, 2, 3, 4, 5, 6, 7
- After Node7 started, Node2 and 3 cannot be found anymore with `ros2 node list`
- Node1 still gets messages from all nodes, including Node2 and 3
- Node7 never gets any message from Node2 or 3
- Approx. 500s after Node7 started, Node1 stops receiving messages from Node2 and 3

#### Sample run 2
*This is a sample test run with five [nodes, all publishing every 10s and subscribed to the same topic,](https://github.com/Ralino/ros2-network-test/blob/master/network_test_ws/src/network_test/src/main_network_test.cpp) each one in its own container. Listing the nodes is done in a [seperate node](https://github.com/Ralino/ros2-network-test/blob/master/network_test_ws/src/list_nodes/src/list_nodes.cpp). This one was done today using the linked container setup.*
- Nodes were started sequentially: 1, 2, 3, 4, 5
- Listing the nodes on container 1 shows only two nodes: Node1 and 3
- Both Node1 and 3 receive only messages from Node1 and 3, all other nodes do not receive *any* messages, including their own
- Listing the nodes in container 2 shows three nodes: Node1, 2 and 3
- After listing the nodes in container 2, Node1 and 3 receive messages from Node1, 3 *and 2*; all other nodes still do not receive any messages
- Approx. 500s after listing the nodes in container 2, Node1 and 3 stop receiving messages from Node2

mjcarroll, 2018-05-07:
So I'm not sure if this is a valid test, but I attempted to reproduce this with `docker-compose` based on a recent OSRF docker image for [ROS2 from source](https://github.com/osrf/docker_images/blob/master/ros2/source/source/Dockerfile)

My modifications are available in this fork: https://github.com/mjcarroll/ros2-network-test

Based on my understanding of how docker-compose's networking mechanism works, this should a similar environment, but I have been unable to reproduce the connectivity issues, even after 10+ minutes of execution.

Do you have any additional thoughts to help reproduce?

clalancette, 2019-09-24:
This has been open for a long time, and no response for over a year.  Fast-RTPS has been through a lot of change since this has been opened.  So I'll close this for now; if you feel this is in error, please feel free to reopen and provide additional information.  Thanks for the report.

closed 2019-09-24 by clalancette