ros2/ros2 #481: Target bionic (meta)

mikaelarguedas, 2018-04-16:
- [ ] Update Docker CI to use bionic
  - [ ] Track down dependency changes
  - [ ] Adapt code to work with new dep versions
  - [ ] stop building from source whatever comes with a high enough version (e.g. assimp)
- [ ] Build opensplice for amd64 and arm64 on bionic
- [ ] Fix new warnings
- [ ] Install bionic deb for Connext if they get built / released
- [ ] Update osrf source docker images to use bionic as base image

nuclearsandwich, 2018-04-16:
Is the plan to drop support for xenial on HEAD the moment that bionic is working or will there be a period of time where we need to support both with the same branch? Of course, ardent will remain a release supported on xenial until bouncy is out. I'm just wondering if we need to test both configurations at once or if we expect all ROS 2 developers to jump to bionic instantly.

dirk-thomas, 2018-04-16:
`master` is targeting the same platforms as the upcoming Bouncy release. So yes, developers tracking `master` have to move to Bionic.

mikaelarguedas, 2018-04-17:
First test results with Fast-RTPS only came back:
https://ci.ros2.org/job/ci_linux/4231/testReport/

63 tests failed, all involving rclcpp, seem to fall into 3 categories:
- [ ] SEGFAULT on parameters
- [ ] SEGFAULT on unbounded arrays
- [ ] Change of behavior of Fast-RTPS for secure participants.

I reran the jobs with Fast-RTPS master to be sure this result is not impacted by the fact we changed Fast-RTPS branch earlier today. The results are identical. https://ci.ros2.org/job/ci_linux/4233/testReport/

dirk-thomas, 2018-05-01:
Just for the record: once this rolled out the following PR can be merged: ament/ament_lint#98.

clalancette, 2018-05-25:
I spent some time today and tested out the ROS2 Turtlebot2 demo on Bionic.  I'll write a more complete report next week, but the short of it is that I was able to get all of the low-level parts of it working (kobuki node, astra driver, joystick, static_transform_publisher, and depthimage_to_laserscan).  However, cartographer_ros fails to start with:

```
F0525 23:46:33.171504 25516 lua_parameter_dictionary.cc:83] Check failed: status == 0 (2 vs. 0) [string "-- Copyright 2016 The Cartographer Authors..."]:49: attempt to index global 'SPARSE_POSE_GRAPH' (a nil value)
F0525 23:46:33.000000 25516 lua_parameter_dictionary.cc:83] Check failed: status == 0 (2 vs. 0) [string "-- Copyright 2016 The Cartographer Authors..."]:49: attempt to index global 'SPARSE_POSE_GRAPH' (a nil value)
```

It may be a relatively simple fix, but I did not have time to look at it further today.

mikaelarguedas, 2018-06-08:
The remaining bits for the release are tracked in https://github.com/ros2/ros2/issues/493
The optional task "get ROS 2 version of ecl + kobuki drivers" is out of our control so I'm going to close this

closed 2018-06-08 by mikaelarguedas