ros2/ros2 #484: Compile fails with gcc 4.x

johnmarkwayve, 2018-04-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04 aarch64 (modified by nvidia for px2)
- Installation type:
  - source
- Version or commit hash:
  - current master (8b73b17)
- DDS implementation:
  - default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
Follow [wiki](https://github.com/ros2/ros2/wiki/Linux-Development-Setup) to install from source while using gcc `gcc (Ubuntu/Linaro 4.9.3-13ubuntu2) 4.9.3` (though I suspect any gcc 4.x will exhibit the same problems)

#### Expected behavior
Successful install

#### Actual behavior
Compiler errors during `rosidl` build form C source. Related to use of C99/C11-style for loops, while gcc 4 defaults to a more C98-based standard ([see here](https://gcc.gnu.org/onlinedocs/gcc-4.9.0/gcc/Standards.html#Standards)).

For example:
```
/home/username/code/ros2/build/lifecycle_msgs/rosidl_generator_py/lifecycle_msgs/srv/_get_available_states__response_s.c: In function ‘lifecycle_msgs_get_available_states__response__convert_from_py’:
/home/username/code/ros2/build/lifecycle_msgs/rosidl_generator_py/lifecycle_msgs/srv/_get_available_states__response_s.c:107:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
     for (Py_ssize_t i = 0; i < size; ++i) {
     ^
```

#### Additional information

Build is successful with gcc 5.4, which defaults to a C11-based standard ([see here](https://gcc.gnu.org/onlinedocs/gcc-5.4.0/gcc/Standards.html#Standards)). However, I have need to use a legacy gcc for abi compatibility.

----

## Feature request
Either modify the build scripts to explicitly use C99 standard (as per the [developer guide](https://github.com/ros2/ros2/wiki/Developer-Guide)) or document compiler version support (preferably the former).

#### Implementation considerations
Given that you're already targeting C99, it seems sensible to explicitly use this in the build scripts even despite this particular problem.

mikaelarguedas, 2018-04-18:
As ROS 2 targets only operating systems that ship with compilers that have C99 enabled by default, we didnt modify the tools or code to enforce it.
If you are willing to provide pull requests on the packages that currently fail to build with it we will consider merging them. You can see an example of how to default to C99 if no standard has been specified [here](https://github.com/ros2/rmw/blob/a82424cfc32312182a775d8607d15c7af71e170f/rmw/CMakeLists.txt#L5-L8)

dirk-thomas, 2018-04-18:
As an alternative you might want to just set the compiler flags for your case using an environment variable.

johnmarkwayve, 2018-04-19:
Thanks for the suggestions. However, the full install still won't work with the older gcc/g++ version due (at least) to lack of experimental/filesystem

For the moment I'm working around my problem in another way and using gcc 5.4 to build ros2, so feel free to close this issue.

closed 2018-04-19 by johnmarkwayve