ros2/ros2 #485: Issue with Fast-RTPS

ChristianRitter, 2018-04-18:
## Issue with Fast-RTPS

**Required Info:**

- Operating System:
  Ubuntu 16.04
- Installation type:
  Source
- Version or commit hash:
  Output of git rev-parse HEAD:
    - rmw_fastrtps_cpp: 9438cca2a6a21b2436684607fea8a78624363f80
    - demos: 044c1bf9fd86ab7b130079ff0286bb775ffeaeff
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  N/A 

#### Steps to reproduce issue
This issue is related to https://github.com/ros2/rmw_fastrtps/issues/81. We uncommented the section in ```rmw_publisher.cpp``` that sets the heartbeat rate to 10ms. Though, this does not solve the issue it is only a slight improvement.

After that we follow the instructions provided by the tutorial in https://github.com/ros2/ros2/wiki/Quality-Of-Service. We also applied a delay instead of a loss using:
```
sudo tc qdisc add dev lo root netem delay 5s
```
but the error also appears independent of any delay or loss.

#### Expected behavior
A smooth video stream can be seen.

#### Actual behavior
The actual behavior depends on the settings:
- When using reliable mode and no delay/loss: All images arrive at ```showimage``` but there is some "stutter", i.e. the stream stops for about half a second and then all images since that stop arrive at the same time.
- When using best-effort and no delay/loss: Some images (about half of them) do not reach ```showimage```, e.g.:
```
[INFO] [showimage]: Received image #43
[INFO] [showimage]: Received image #45
[INFO] [showimage]: Received image #47
[INFO] [showimage]: Received image #49
[INFO] [showimage]: Received image #51
[INFO] [showimage]: Received image #53
[INFO] [showimage]: Received image #56
[INFO] [showimage]: Received image #57
[INFO] [showimage]: Received image #59
[INFO] [showimage]: Received image #60
```
- When using best-effort and delay: The percentage of images lost is highly increased.
- When using reliable and delay: No images reach ```showimage```.

#### Additional information
We require emulated delay to simulate high-distance tele-operation. We also rely on Fast-RTPS due to its license.

clalancette, 2018-08-02:
I'm sorry for the long delay here.  I believe the issues you have encountered with Fast-RTPS should be fixed in Bouncy.  I just did a few tests:

* No loss specified, best-effort, running `ros2 run image_tools cam2image -- -b -r 0`/`ros2 run image_tools showimage -- -s 0 -r 0` --> Images arrive right away, as expected
* No loss specified, reliable, running `ros2 run image_tools cam2image -- -b -r 1`/`ros2 run image_tools showimage -- -s 0 -r 1` --> Images arrive right away, as expected
* Loss specified using qdisc with 5 seconds delay, best-effort, running `ros2 run image_tools cam2image -- -b -r 0`/`ros2 run image_tools showimage -- -s 0 -r 0` --> Images arrive delayed by about 5 seconds, as expected.
* Loss specified using qdisc with 5 seconds delay, reliable, running `ros2 run image_tools cam2image -- -b -r 1`/`ros2 run image_tools showimage -- -s 0 -r 1` --> Images arrive delayed by about 5 seconds, as expected.

One thing I will note is that I had to run it with `-s 0` (don't show images) because I am ssh'ed into a container, so trying to X-forward uses lo, which is delayed by qdisc :).  If you are on a local machine (not ssh'ed), it should probably work without that.

Please give this another try and let us know if the latest release fixes this problem for you.  Thanks.

ChristianRitter, 2019-01-25:
Hi,

I am really sorry for not responding earlier. Due to time limitations we had to switch back to ROS1 and I lost track of this issue.

Nonetheless, thank you very much, the fix will be very helpful in the future.

I am going to close this issue now.

Best regards,
Christian

closed 2019-01-25 by ChristianRitter