ros2/ros2 #489: Update get_node_names to never return empty values

dhood, 2018-05-02:
Each supported RMW implementation (e.g. `rmw_connext_cpp`, `rmw_fastrtps_cpp`, `rmw_opensplice_cpp`) has its own vendor-specific implementation of `rmw_get_node_names`.

`rmw_get_node_names` is called from [`rcl_get_node_names`](https://github.com/ros2/rcl/blob/2c390d3a67664e0ba0bec86589f994a10bfee847/rcl/include/rcl/graph.h#L210), which is what all client libraries (c++, python, etc) use.

"Pure" non-ROS DDS participants will be detected by some vendors in `rmw_get_node_names`, and the current approach for implementing `rmw_get_node_names` returns a list that _contains_ these empty names.

The objective of this ticket is instead to have `rmw_get_node_names` in the rmw implementations only return names of participants that are non-empty.

After that has happened, we can revert these PRs that work-around the potential for empty names in the list:
- [ ] https://github.com/ros2/rclcpp/pull/435
- [ ] https://github.com/ros2/rclpy/pull/177
- [ ] https://github.com/ros2/rcl/pull/214
- [ ] https://github.com/ros2/ros2cli/pull/76

The only behaviour that needs to be changed is:
- [ ] [`rmw_connext_cpp`](https://github.com/ros2/rmw_connext/pull/276/files#diff-bccfea1954134a50cf75b01ae95188e3R96)

- `rmw_fastrtps_cpp` already [ignores participants without a name](https://github.com/ros2/rmw_fastrtps/pull/182/files#diff-e0ae150821b275cbd461f0895600e01aR71)
- `rmw_opensplice_cpp` already [skips participants without a name in the userdata](https://github.com/ros2/rmw_opensplice/pull/191/files#diff-722d8069c393a517aee8369e15ead3e6R158) (opensplice only supports participants being named via the user data: https://github.com/ros2/ros2/issues/438).

The recommendation for implementing the change is to:
1. over-allocate the array for the potential number of names, 
1. populate the first elements of the array with valid names encountered,
1. down-size the string array.

`rmw_opensplice_cpp` currently does step 3 by [coping the data into a smaller-sized array](https://github.com/ros2/rmw_opensplice/pull/191/files#diff-722d8069c393a517aee8369e15ead3e6R184). To avoid this step the recommendation is to add a new function for [string arrays](https://github.com/ros2/rcutils/blob/c72d2cf3da3dd4f2c8286f647c0718898f44fef0/src/string_array.c) that allows the unused portion of the array to be deallocated. More details in [this thread](https://github.com/ros2/rclcpp/pull/435#issuecomment-385060611).

CaptainTrunky, 2018-08-05:
If this one is still relevant, I would like to take a look.

dhood, 2018-08-06:
Still relevant, thanks @CaptainTrunky!

My recommendation would be to submit pull requests in the following "stages":
1. focus on the change to `rmw_connext_cpp` to not return non-empty names, copying the approach in `rmw_opensplice_cpp` which copies into a second string array (effective but not the most efficient),
1. revert the PRs listed in the above post,
1. implement the function that down-sizes `rcutils` string arrays and use that in `rmw_connext_cpp` and `rmw_opensplice_cpp` for efficiency gains (down-sizing rather than copying into a smaller string array).

Let us know if there are additional pointers that would help out

dirk-thomas, 2018-08-06:
Even when `rmw_connext_cpp` is updated I wouldn't recommend to revert the PRs that work-around the potential for empty names in the list. There are other RMW implementations out there which might return empty names and there is no harm in keeping the logic to prevent problems if the returned array contains empty entries.

dhood, 2018-08-06:
Last time we had this discussion at https://github.com/ros2/rclcpp/pull/435#issuecomment-377646560 we agreed that RMW implementations shouldn't return empty names because names of ROS nodes can't be empty. If there are other RMW implementations that are returning empty names I would argue that they should also be updated to only return names of actual ROS nodes. In the past this was something that you and I agreed on -- if perhaps there is additional context around why you think otherwise now, could you share it please to help me understand the change in opinion?

dirk-thomas, 2018-08-06:
I do agree that the API *should* not return empty values.

I just don't think that we should revert some of the changes which added "defensive" code to prevent bad things like segfaults from happening.

dhood, 2018-08-08:
alright, that makes sense, it'd be more effort to remove it. In that case, I'd say the objective of step 2 then is to:

1. check for empty names in `rcl` and error in the event that (some other) rmw implementations return empty names,
1. revert https://github.com/ros2/rcl/pull/214 so that other client libraries know that the effort of working around empty names is not needed.

CaptainTrunky, 2018-08-09:
Thanks @dirk-thomas, @dhood. It's a bit clearer what to do now. I'm struggling with building ROS2 from scratch, but it looks like it's almost working. I'll prepare a pull request within next few days.

CaptainTrunky, 2018-08-11:
Could you advise a recommended way of testing my changes? Especially two things:

-  how to reproduce 'empty' names from ROS2 perspective

- how to add my own unit tests

dhood, 2018-08-13:
Good questions, @CaptainTrunky 

> how to reproduce 'empty' names from ROS2 perspective

Empty names can come from DDS participants that are not created by ROS 2. The most common reason in the ROS 2 world is from nodes using `rmw_opensplice_cpp`, because Opensplice spawns some "pure" DDS participants in the background.

---

> how to add my own unit tests

There have been some tests added for `get_node_names` from rclcpp in this PR: https://github.com/ros2/system_tests/pull/260. My suggestion is that you modify that test to ensure that no empty names are returned. When that test is run with opensplice support, I imagine it would fail without your changes, and pass with your changes. You will need to have [set up opensplice support](https://github.com/ros2/ros2/wiki/Linux-Development-Setup#prismtech-opensplice-debian-packages-built-by-osrf) before you build your workspace.

---

> is there a way to build only specified projects using colcon?

To build dependencies of `test_rclcpp`, for example, you can run:

```
colcon build --event-handler console_direct+ --packages-up-to test_rclcpp
```

And to run just the tests for get_node_names you can do:
```
colcon test --event-handler console_direct+ --packages-select test_rclcpp --ctest-args -R node_name
```

dhood, 2018-09-21:
@CaptainTrunky how did you go? Let us know if you need additional tips. If you ran out of time to work on this, feel free to publish where you got up to so that someone else can take over.

CaptainTrunky, 2018-09-23:
Hi @dhood I've been on a trip last few weeks. I plan to submit corresponding fixes and tests by the end of the next week.

CaptainTrunky, 2019-07-01:
I guess I could consider it to be done. Passing tests w/o any issues.

jacobperron, 2021-03-26:
ros2/rcl#570 adds a guard against empty names being returned by `rcl`.

I think twe can retire this issue.

closed 2021-03-26 by jacobperron