ros2/ros2 #490: Build problem : doesnt find dependencies for cam2image

LLB8, 2018-05-09:
Hi everybody, 

I need to use ROS 2 for real time project. 

I use a computer on which ros indigo was installed with ubuntu 14.04. First I upgrade ubuntu to 16.04 LTS version, then I removed ros-indigo (I think but I don't know how to check it .. ). 

To install ROS 2 I followed the source tutorial for linux (https://github.com/ros2/ros2/wiki/Linux-Development-Setup). My problem occurs when I want to build the code in the workspace for the function "cam2image". 

It seems that some openCV dependencies cannot be found (libswscale, libavutil, libavformat and libavcodec). I tried to solve it by using this kind of command for each of them : `sudo cp libavcodec-ffmpeg.so.56 libavcodec-ffmpeg.so.56.bak` and then `sudo mv libavcodec-ffmpeg.so.56.bak libavcodec.so.54`. In fact the lib **libavcodec-ffmpeg.so.56** was in the ubuntu library whereas the lib **libavcodec.so.54** was not. 

It solves something I think because some files (ex : **libavcodec.so.54**) that where not found are found now .. However I still have a problem .. I post it below. I don't know at all what to do ! 

`+++ Building 'image_tools'
==> '. /home/pichat/ros2_ws/build_isolated/image_tools/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/pichat/ros2_ws/build_isolated/image_tools'
==> '. /home/pichat/ros2_ws/build_isolated/image_tools/cmake__build.sh && /usr/bin/make -j4 -l4' in '/home/pichat/ros2_ws/build_isolated/image_tools'
[ 42%] Built target showimage
[ 57%] Linking CXX executable cam2image
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `sws_getContext@LIBSWSCALE_2'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avio_close@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_seek_frame@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_alloc_frame@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_write_header@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_encode_video2@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_freep@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avpicture_get_size@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_codec_get_tag@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avio_open@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_dict_free@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_init_packet@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_open_input@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `sws_scale@LIBSWSCALE_2'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_flush_buffers@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_decode_video2@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_reduce@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_log_set_level@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_write_trailer@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_align_dimensions2@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_open2@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_close@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_guess_codec@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_sub_q@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_alloc_context@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_opt_set@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_free@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_free_frame@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_codec_get_id@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_read_frame@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_free_context@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `sws_getCachedContext@LIBSWSCALE_2'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_guess_format@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_register_all@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_free_packet@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_write_frame@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_new_stream@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_rescale_q@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_find_encoder@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_get_context_defaults3@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_get_riff_video_tags@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avpicture_fill@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_lockmgr_register@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_network_init@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_mallocz@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_find_stream_info@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avcodec_find_decoder@LIBAVCODEC_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `avformat_close_input@LIBAVFORMAT_54'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_dict_set@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `av_malloc@LIBAVUTIL_52'
/usr/local/lib/libopencv_videoio.so.3.2.0: undefined reference to `sws_freeContext@LIBSWSCALE_2'
collect2: error: ld returned 1 exit status
CMakeFiles/cam2image.dir/build.make:230: recipe for target 'cam2image' failed
make[2]: *** [cam2image] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/cam2image.dir/all' failed
make[1]: *** [CMakeFiles/cam2image.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

<== Command '. /home/pichat/ros2_ws/build_isolated/image_tools/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/pichat/ros2_ws/build_isolated/image_tools' with exit code '2'
<== Command '. /home/pichat/ros2_ws/build_isolated/image_tools/cmake__build.sh && /usr/bin/make -j4 -l4' failed in '/home/pichat/ros2_ws/build_isolated/image_tools' with exit code '2'
`

Thanks by advance

mikaelarguedas, 2018-05-09:
There seem to be an unconsistency on your system.

You have an OpenCV libraries that exist only on Ubuntu 18.04 but the description says you upgraded to Ubuntu 16.04. And some OpenCV dependencies that are still from 14.04 instead of the 16.04 ones, for example libavcodec54 is a [14.04 library](https://packages.ubuntu.com/trusty-updates/libavcodec54) and on 16.04 you should have [libavcodev-ffmpeg56](https://packages.ubuntu.com/xenial/amd64/libavcodec-ffmpeg56/filelist).

I recommend purging your system from all 14.04 packages and to reinstall libopencv-dev.

LLB8, 2018-05-09:
Thanks for your answer.

How do you advise to purge the system ? I thaught I did it already with some command lines that I don't remember... 

Some forum on the Internet explain that an upgrade made thanks to commande lines is not perfect. Do you think that I have to make another boot of the computer to install properly the 16.04 version ? 

As you can see I am not really a "linux expert" !

mikaelarguedas, 2018-05-09:
I usually never "upgrade" but always format and install the new ubuntu version on a "clean disk".
If you can afford to do that I would recommend it.

Otherwise you can:
- Make sure you will download only 16.04 package:
  - all entries in /etc/apt/source.list should have `xenial` and not `trusty` or `bionic`
  - remove any files you don;t need from the /etc/apt/source.list.d directory

- Try to remove the packages that are conflicting as they come up. In your case it looks like:
  - `sudo apt autoremove && sudo apt remove libavcodec* libavformat* libavutil* libopencv*`
and then trying to install opencv again:
  - `sudo apt clean && sudo apt update && sudo apt install libopencv-dev`

LLB8, 2018-05-11:
I will try to remove the package and then if it doesn't work I will install on a clean disk. Do you advise me to install the 16.04 version or 18.04 ? I think maybe it is better to work on the last one, however I don't what choice is the best one for ROS ! 

Otherwise, I want to thank you for your fast and efficient answers.

closed 2018-05-17 by LLB8