ros2/ros2 #492: Use of OpenCV with ROS 2

LLB8, 2018-05-17:
Hi everybody, 

I really need to clarify something and I don't know if I am the only person who has this doubt. 

I never use ROS before now and I want to use PCL and OpenCV in real-time. That is why I installed ROS 2. However, I feel that those libraries are not included in ROS 2 whereas I thought that ROS 2 was built from ROS 1. 

I wonder if it is necessary to install ROS 1 kinetic and to do a Bridge to use those library. If it is, is it possible to have a list of what tools are present in ROS 2 ? Is the only aim of ROS 2 to change to distribution protocol ?   

Thanks in advance for the clarification.

mikaelarguedas, 2018-05-29:
>  I thought that ROS 2 was built from ROS 1.

ROS 2 is a separate project from ROS 1 and is not built on top of ROS 1.

> I want to use PCL and OpenCV in real-time. However, I feel that those libraries are not included in ROS 2 whereas I thought that ROS 2 was built from ROS 1.

Both PCL and OpenCV are libraries independant from ROS so they can be used with ROS 1 or ROS 2 or neither.
ROS 2 will allow you to use the ROS2 concepts and communication protocol but will not change the internal code of these libraries. So if you use functions/libraries that are not real-time safe, ROS 2 will not make them real-time safe.

>  Is the only aim of ROS 2 to change to distribution protocol ?

ROS 2 has several goals and one of them is changing the distribution protocol.
To find more information about the various goals of ROS 2  and changes from ROS1 I encourage you to look at the design documents available on design.ros2.org ([Why ROS 2](http://design.ros2.org/articles/why_ros2.html) and [Changes between ROS 1  and ROS 2](http://design.ros2.org/articles/changes.html) are good starting points).

HTH,

LLB8, 2018-05-30:
Thanks a lot. 

It really helps, I think I will use only ROS 1 with the nodelets functionnalities in order to be as close as possible of real time. 

Bye

closed 2018-05-30 by LLB8