ros2/ros2 #493: Bouncy Bolson Pre-Release CheckList

mikaelarguedas, 2018-05-23:
Meta ticket tracking things to be done for the Bouncy release

- Kick-off:
  - [ ] REP-2000 for bouncy @mikaelarguedas: https://github.com/ros-infrastructure/rep/pull/174

- Infrastructure:
  - [ ] upgrade linux buildfarm hosts @nuclearsandwich 
  - [ ] create Bouncy distro: @mikaelarguedas
    - [ ] ros2/rosdistro
    - [ ] ros2/ros_buildfarm_config
    - [ ] Deploy bouncy management jobs to build.ros2.org
    - [ ] update arm node label to pick up bouncy jobs: `buildagent_arm64 || bouncy_binarydeb_ubv8`

- [ ] Bionic support: details tracked at https://github.com/ros2/ros2/issues/481 @mikaelarguedas

- Packaging:
  - [ ] Implement solution for allowing to opt out from multi rmw install: @nuclearsandwich 
    - possibilities:
      - put only `build_depend` on the "optional" rmws
      - inject a `Recommends` for the "optional" rmws
  - [ ] Extract `catkin_prepare_release` from `catkin` to be used by ROS 2 users @dirk-thomas 
    - can be renamed and point to the catkin provided one if present
  - [ ] Audit patches on release repositories
  - [ ] Propagate ignored files

- colcon:
  - [ ] address issues identified for Bouncy
  - [ ] provide debian packages hosted on the ROS and ROS2 apt repos
  - [ ] merge design PR
  - [ ] update wiki to refer to colcon everywhere
  - [ ] deprecate `ament_tools` repo

- Dependency management:
  - [x] ASIO: [SRU submitted](https://bugs.launchpad.net/ubuntu/+source/asio/+bug/1771903), keep using system version for now
  - [ ] Assimp: waiting on upstream:
    - alternative: keep vendoring it for all platforms
  - [ ] console_bridge: create a vendor package and use upstream on MacOS and Bionic (optional)
  - [ ] curl: @wjwwood to confirm if vendoring it is still needed
  - [ ] poco: use upsream on MacOS: update CI machines 
  - [ ] tinyxml2: Update custom choco package to 6.0.0

- Documentation
  - Update tutorials
  - update installation instructions
  - remove mentions to beta1 or earlier from the wiki

- T-shirt campaign

nuclearsandwich, 2018-06-08:
> configure build.ros2.org smtp @nuclearsandwich

There's some outstanding DKIM and SPF setup to do for email and I'm still waiting to hear back from AWS about de-throttling smtp

dirk-thomas, 2018-06-11:
For the bullets under "Audit patches on release repositories" it would be great to have a list of what actual patches are being applied for each of these packages. That would ease reviewing them a lot.

mikaelarguedas, 2018-06-11:
> For the bullets under "Audit patches on release repositories" it would be great to have a list of what actual patches are being applied for each of these packages. That would ease reviewing them a lot.

For the ones that I audited I put link of the PRs that move the patch to the source repo making the patch in the release repo obsolete.
For the ones that dont have the box checked, that means I didnt evaluate how the patches should be modified for bouncy yet and so there is no description of the patches being applied yet.
The unticked ones fall into 3 categories:
- packages that don't have a package.xml upstream and the patch consists in adding a package.xml
- packages that have different dependencies in the debian case and that we should evaluate (typesupport, rmws etc)
- packages that need extra dependencies only in the debian case: ros1_bridge

mikaelarguedas, 2018-07-03:
Remaining work vendoring console bridge ticketed at https://github.com/ros2/ros2/issues/521

Closing this

closed 2018-07-03 by mikaelarguedas