ros2/ros2 #494: ROS2 Bug Report

ChangsongYu, 2018-05-28:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - binaries ROS 2 Ardent Apalone - 2018-03-07/ROS 2 Ardent Apalone - 2018-02-09: ros2-ardent-package-windows-fastrtps-AMD64.zip

- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
0. open cmd
1. dev/ros2-windows/local_setup.bat (activate ros2 environment)
2. create package ROS2 talker
3. mkdir build
4. cd build
5. cmake ..\ -G"Visual Studio 14 Win64"
6. msbuild /p:Configuration=Debug
7. cd Debug
8. talker.exe

#### Expected behavior
[INFO] [talker]: Publishing: 'Hello World to chatter: 1'
[INFO] [talker]: Publishing: 'Hello World to chatter: 2'
[INFO] [talker]: Publishing: 'Hello World to chatter: 3'
[INFO] [talker]: Publishing: 'Hello World to chatter: 4'

#### Actual behavior
9. abort (error)

#### Additional information

When we call auto node = rclcpp::Node::make_shared("elena"); in Debug Mode, it throws an exception: node name must not contain characters other than alphanumerics or '_'
When we call auto node = rclcpp::Node::make_shared("elena"); in Release mode, it is ok.
=> We cannot run our program with ros2 in Debug Mode.

mikaelarguedas, 2018-05-28:
thanks @ChangsongYu for reporting this issue.

1. I see that you don't pass the Debug build type to cmake, have you tried doing that ? 
`-DCMAKE_BUILD_TYPE=Debug` 

1. Do you face the same problem if you modify the [existing talker example](https://github.com/ros2/examples/tree/ardent/rclcpp/minimal_publisher) to use your custom node name?

1. Can you provide more details about what you did in Step 2 "create package ROS2 talker"? If 2. works for you, there may be something different in the package you created yourself.

ChangsongYu, 2018-05-28:
Hi,

**create package ROS2 talker** means that I copied the example from https://github.com/ros2/ros2/wiki/Migration-Guide and didn't change anything. 
File-structure: 
-ros2_talker
  -CMakeLists.txt
  -package.xml
  -talker.cpp
  -build
    - ..... (a lot of files produces by cmake ..\ -G"Visual Studio 14 Win64")

I downloaded minimal_publisher and minimal_subscriber from https://github.com/ros2/examples/tree/ardent/rclcpp and tried them with debug mode again. The results are as same as that of ros2_talker 2 

1. we don't need to append -DCMAKE_BUILD_TYPE=Debug to cmake. If I append it, then throws a warning:
**CMake Warning:
  Manually-specified variables were not used by the project:
    CMAKE_BUILD_TYPE**

2. 
    open cmd
    dev/ros2-windows/local_setup.bat (activate ros2 environment)
    create package ROS2 talker
    mkdir build
    cd build
    cmake ..\ -G"Visual Studio 14 Win64"
    msbuild /p:Configuration=Debug
    cd Debug
    talker.exe

ChangsongYu, 2018-05-28:
Note: CMD is VS 2015 x64 Native Tools Command Prompt
I have already tested them on another laptop with Win 10, Visual Studio 14, the results are as same as that of ros2_talker.

dirk-thomas, 2018-06-20:
I don't think this is a bug. Mixing debug and release code is bad practice. There is simply no guarantee that a release library used by a debug executable is going to work.

The problem in this case is passing the node name as a `std::string` which is a STL object. That is not supported since the STL class is implemented differently in the library (in release mode) compared to the executable (in debug mode). So the interface "what" is being passed is inconsistent.

mikaelarguedas, 2018-07-05:
@ChangsongYu As of ROS Bouncy, we provide a Debug packaging artifact with all ROS2 compiled in Debug mode. This should allow you to build and run executable in a Debug overlay workspace.
You can find the artifact on the [Bouncy release page](https://github.com/ros2/ros2/releases/tag/release-bouncy) under the name `ros2-bouncy-windows-Debug-AMD64.zip`.

As @dirk-thomas pointed out this is actually not a bug as mixing release and debug is bad practice.

I'm going to close this as an alternative has been provided in ROS Bouncy. Feel free to comment here if the alternative doesn't work for you and we can reopen this ticket

closed 2018-07-05 by mikaelarguedas