ros2/ros2 #495: ROS 2 + clipse ROS Answers questions

dhood, 2018-05-29:
@flobotics, we are getting questions on ROS answers about the [Eclipse ROS 2 tutorials](https://github.com/ros2/ros2/wiki/Building-ROS-2-on-Linux-with-Eclipse-Oxygen) tutorials. Could you please help those users and update the tutorials?

https://answers.ros.org/question/292011/ros2-with-eclipse/
https://answers.ros.org/question/292487/configuring-ros2-with-eclipse/

flobotics, 2018-06-03:
i wrote back and edited the tutorial

closed 2018-06-04 by dhood

aakashsehgal, 2018-06-05:
This is not working for me...I guess i am following the tutorial correctly until the point where i need to set the environment variables.

1. Open Eclipse from the console where all commands were executed. For me it always says "No such file or directory". That is why I always open the file by double clicking on the icon. Is this wrong ?

2. To set the environment variables, it is asked to source the local_setup.bash file. But this file is generated only in after building the project i.e. not present in the src directory. So how is this possible ?

jajosheni, 2020-07-19:
'python-empy is not found' while following the tutorial.
I guess as python 2 is deprecated this package is not available anymore in ubuntu focal.

nitin5, 2020-11-18:
I am trying to setup Eclipse for ROS Foxy. 
I am unable to set the ament.py. The webpage have text command for user "We enter the amen.py settings" and image shown is of Edit Configuration.
I do not know location of ament.py in my installation! It is not present in the relative location of my workspace.

 I have been using colcon as per tutorials. Shall build configuration use that?

nitin5, 2020-11-19:
I have found a work-around, which involve a lot of manual setup rather than automatic detection from environment varibales and CMakelist.txt. Not sure it will work for all cases.
 
Replace ament command with colcon. Create a new Builder for the project with colcon command. set location as /usr/bin/colcon. Set appropriate arguments which you will use to compile the package from the terminal.

For indexer: In C/C++ General -> Path and Symbols -> tab includes -> for GNU C++ language: Add path of ROS includes (/opt/ros/foxy/include), Add path of includes of specific libraries which pkg uses, Add install path of the other ros-packages from the underlay/current workspace environment.