ros2/ros2 #496: Allow to create topics with "exact" topic name from rcl

mikaelarguedas, 2018-05-29:
## Feature request

#### Feature description
Allow to pass an exact topic name to `rcl` and create a publisher/subscriber with that exact topic name at the DDS level. This will allow for better integration with native DDS systems 
Currently users can bypass the custom ROS prefix added to the topic names, using the [`avoid_ros_namespace_conventions` flag](https://github.com/ros2/rmw/blob/8ea66dbbe89e78318cd2f2b4e7d8da51211d67bb/rmw/include/rmw/types.h#L205-L215), but this doesnt allow to bypass name expansion into a fully qualified name.

Follow up of https://github.com/ros2/ros2/issues/476, particularly this discussion https://github.com/ros2/rmw_fastrtps/pull/192#discussion_r178970046

#### Implementation considerations

RCL could pass both the name provided by the users as well as the fully qualified name to rmw. And the each rmw implementation would create a topic using on or the other based on the value of the `avoid_ros_namespace_conventions` provided in the qos profile

Karsten1987, 2018-05-29:
does it make sense to extend the scope of `avoid_ros_namespace_conventions` to also bypass the fqdn? This would probably entail that an additional set namespace is ignored.
I feel like having two flags might be unhandy.

mikaelarguedas, 2018-05-29:
Sorry I'm not sure I understand.

To clarify:
My suggestion is to bypass everything using  single flag. e.g. if you pass `ns=foo topic=bar` 

`avoid_ros_namespace_conventions=true`: you get DDS topic: `bar`
`avoid_ros_namespace_conventions=false`: you get DDS topic: `rt/foo/bar`

Your suggestion is to provide a way to bypass fqdn and another one to bypass the ros prefixes ?
so:
00 => `rt/foo/bar`
01 => `/foo/bar`
10 => `rt/bar`
11 => `bar`

like this? or something else ?

Karsten1987, 2018-05-29:
I was basically just wondering whether there is a strong use case to support the namespace expansion but not the prefixing. My question is then just whether we should support all four cases you mentioned above or only the two extreme ones.

mikaelarguedas, 2018-05-29:
I was going only to the first suggestion (single variable, all or nothing)

@ros2/team what do you think about these approaches ?

ruffsl, 2018-05-30:
I'm not sure if the use of cases 01 and 10. Having either the fully qualified domain name, or topic string directly map to the DDS topic seems like it would cover most use cases. But then again, I suppose you could support 01, 10 with the flag you specified and examining if the name space argument passed by there user was either an empty string, just "/", or just "/foo".

mikaelarguedas, 2018-05-31:
To summarize, the proposal is to keep only the existing flag `avoid_ros_namespace_conventions` and to modify the implementation to have the following behavior:
if you pass `ns=foo topic=bar`:

`avoid_ros_namespace_conventions=true`: you get DDS topic: `bar`
`avoid_ros_namespace_conventions=false`: you get DDS topic: `rt/foo/bar`

If users want to do an in between, like creating a topic `/foo/bar`, they will need to perform the expansion before and pass the resulting string to the `create_publisher`/`create_subscription`/... functions

spiderkeys, 2019-07-22:
I am working on a ROS2 application in which I need to publish to topics which do not have leading forward slashes. The target of communication is a pure DDS application, not a ROS2 application. I was hoping that using avoid_ros_namespace_conventions=true would help me achieve this, but it looks like there is still a leading slash on any topics created using this method preventing the ability to interoperate.

That said, I would be in favor of your proposal above, @mikaelarguedas.

wjwwood, 2019-07-22:
I don't see the value in two separate options (the `01` and `10` from above seem to be not very useful to me), so I'd be in favor of disabling topic name expansion when `avoid_ros_namespace_conventions` is `true`.

ruffsl, 2019-07-22:
> If users want to do an in between, like creating a topic `/foo/bar`, they will need to perform the expansion before and pass the resulting string to the `create_publisher`/`create_subscription`/... functions

Yea, if it's possible to introspect the node name and namespace mapping from same user level code at runtime, then I think `01` and `10` could be replicated by the user. The behavior for enabling `avoid_ros_namespace_conventions` would then be quite straightforward, in that the string you pass becomes the exact DDS topic string used.

guillaumeautran, 2022-05-17:
I'm just curious, could a naive implementation such as [this one](https://github.com/ros2/rcl/compare/master...clearpathrobotics:issue_496) implement that is being asked in this request?

guillaumeautran, 2022-05-18:
All of these layers should be going through `rcl`. But I'm not sure if it is that simple :) 

Note that my change does work correctly for `rclcpp`, ie: I can publish and subscribe to DDS topics such as `topic/my_topic`. This type of topic name does not get prefixed with a `/` character. Note however that my change disables any kind of processing on the topic name. There is no substitution or otherwise....

clalancette, 2022-06-02:
What I'll suggest here is to open a pull request with that change.  We can discuss it more on that pull request, and run some additional tests with it to see what happens.

mrmara, 2025-04-04:
Sorry to jump in this very old conversation, but I am really interested in the matter.
In particular I am trying to call a ROS2 service by pusblishing a ROS2 topic, I am almost there but the leading "/" blocks me. I tied to follow the merged PR but could not really find a solution.
Is there a way to create a ROS2 publisher and subscirber that does not convert its topic to DDS with a leading slash?

[image]