ros2/ros2 #497: Problem in the dynamic link library `C:\dev\ros2\fastrtps-1.5.dll`

cosmicog, 2018-05-31:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Binaries
- Version or commit hash:
  - ROS 2 Ardent Apalone - 2018-03-07, `ros2-ardent-package-windows-fastrtps-AMD64.zip`
- DDS implementation:
  - Fast-RTPS (I think so, I have OpenSplice installed and `OSPL_HOME` is set)
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue
I had Visual Studio 2017 and latest Visual C++ Installed, so I skipped installing 2015 for seeing the issues and trying later. Other steps were same here [installing-prerequisites](https://github.com/ros2/ros2/wiki/Windows-Install-Binary#installing-prerequisites) except I had OpenSplice installed and `OSPL_HOME` is set. Tried by deleting this too.
Just called it from terminal for seeing the error.
```
>C:\dev\ros2\Lib\demo_nodes_cpp\talker.exe
```

#### Result

[image]

sloretz, 2018-05-31:
Text in image for people searching: `The ordinal 4689 could not be located in the dynamic link library C:\dev\ros2\bin\fastrtps-1.5.dll.`

sloretz, 2018-05-31:
@cosmicog thanks for reporting. Would you mind checking if the same issue happens using the latest nightly? https://ci.ros2.org/view/packaging/job/packaging_windows/1021/artifact/ws/ros2-package-windows-AMD64.zip

dirk-thomas, 2018-07-05:
Can you please try this again with the recent Bouncy release and comment if the problem still exists.

cosmicog, 2018-07-06:
With the latest build 1057
```
The ordinal 4601 could not be located in the dynamic link library C:\dev\ros2\bin\fastrtps-1.5.dll.
```
I don't have an experience with windows libraries. Do you think it might be related only to my pc? I was thinking about fresh install. My computer can't install updates since months.

cosmicog, 2018-07-06:
With the latest build on ci.ros.org and the release on github, it gave:
```
The ordinal 4615 could not be located in the dynamic link library C:\dev\ros2\bin\fastrtps-1.5.dll.
```
this time. Do you think the problem may related to my computer? I didn't get any updates since months. I will try on another desktop and try to give feedback today.

cosmicog, 2018-07-06:
Finally listener heard `Hello World`s. I needed to fix hard-coded python path to `C:\Python37` , and it still gives `failed to create process` if I run `ros2` command. But this issue is resolved for me.

dirk-thomas, 2018-07-06:
@cosmicog Since you opened the ticket you should be able to close it when it is resolved. For any other problem (`failed to create process`) please create a new ticket with enough information and context.

closed 2018-07-06 by cosmicog