ros2/ros2 #499: Calling setup.bat failes if there is a space in the directory hierachy (Windows)

anhosi, 2018-06-05:
Wenn putting the ros2 workspace within a directory that contains a space, e.g. `C:\Users\firstname lastname\ros2_ws`. The following error occurs when sourcing  the `setup.bat`:
```Die Datei "C:\Users\firstname" kann nicht gefunden werden.
Die Datei "C:\Users\firstname" kann nicht gefunden werden.
Die Datei "C:\Users\firstname" kann nicht gefunden werden.
```

Translated: `The file "C:\Users\firstname" cannot be found.`
It seems like the current directory is not properly quoted in the setup script.

Building ros2 from source within that directory is working fine.

Operating system: Windows 10
Installation type: from source

Expected behavior instead: The setup.bat script should run without any errors.

dirk-thomas, 2018-06-06:
Can you please add information what version of ROS 2 you are using / building and what exact steps you are following.

anhosi, 2018-06-06:
I built ros2 from source using the ros2 master branch ros2.repos. I exported all repo versions into a repos file at here https://gist.github.com/anhosi/6ebf183216b4773c74713f90dfc106ef.

I built according to https://github.com/ros2/ros2/wiki/Windows-Development-Setup the variant for the debug build using `C:\Users\firstname lastname\ros2_ws` as workspace directory. The build itself was successful. I used VS2017.
Sourcing setup.bat leads to above error.

As a workaround I moved the the workspace to `C:\dev\ros2_ws`.

dirk-thomas, 2018-06-07:
We are currently transitioning from the build tool `ament_tools` to `colcon`. With the new tool I just verified that the problem doesn't exist anymore. We will likely not address the bug in `ament_tools`. I would recommend to not use spaces in the first place or (as soon as we updated the wiki) use the new instructions to build the workspace. If anyone would like to provide a patch for `ament_tools` we would happily review and merge it though.

clalancette, 2019-09-24:
I just tested this locally using colcon with a space in the path, and it seems to work for me.  I'm going to close this out, but feel free to reopen if you are still having problems.

closed 2019-09-24 by clalancette