ros2/ros2 #501: FastRTPS and OpenSplice don't talk on Empty messages

Karsten1987, 2018-06-07:
This is a continuation of the issue described in here: https://github.com/ros2/ros2/issues/476#issuecomment-389006954

I exclude this into its own issue as the underlying PRs for removing partitions were merged.

dirk-thomas, 2018-06-19:
I tried this locally as well as ran a set of CI builds with FastRTPS and OpenSplice for this case:
* Linux [image]
* Linux-aarch64 [image] unrelated test failures
* macOS [image]
* Windows [image] (specific [test](https://ci.ros2.org/job/ci_windows/4797/testReport/(root)/projectroot/test_publisher_subscriber__Empty__rclcpp__rmw_opensplice_cpp__rmw_fastrtps_cpp/))

All of them pass the specific test described in this ticket. Therefore I would assume the problem doesn't exist anymore in the current code base (or is at worst flaky). So I suggest to close this ticket until a reproducible problem can be described.

closed 2018-06-19 by dirk-thomas

dirk-thomas, 2018-06-21:
> Do you mean that the reproducible example linked in this issue: #476 (comment) was working for you ?

Yes.