ros2/ros2 #503: Add catkin_pkg as a dependency when installing from binary packages

minggangw, 2018-06-11:
## Bug report

**Required Info:**

- Operating System: 
  - Linux/macOS/Windows
- Installation type:
  - binaries
- Version or commit hash:
  - 
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

Any C++ project, run
```
ament build .
```

#### Expected behavior
Build successfully

#### Actual behavior
lack of module catkin_pkg

#### Additional information
The Wiki for the [Development Setup](https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/5a8c57317647913892ad327112c1d64272b0236d) section has been updated. I think we'd better add `catkin_pkg` as a dependency for installing from binary packages on Wiki also, as `ament` is a common command in ROS 2.0.

dirk-thomas, 2018-06-11:
The packages which need `catkin_pkg` declare it as a dependency in their manifest already (see e.g. https://github.com/ament/ament_tools/blob/f367b3258faddc1777fe82c1feaef8b3b9bcde4a/package.xml#L16). Do you have a specific package in mind where this is not the case at the moment?

minggangw, 2018-06-12:
Hi @dirk-thomas , I met this problem when running the ci of [rclnodejs](https://github.com/RobotWebTools/rclnodejs) using binary package, steps to reproduce:

1.Download the nightly build package from ci.ros.org for [windows](http://ci.ros2.org/view/packaging/job/packaging_windows/lastSuccessfulBuild/artifact/ws/ros2-package-windows-AMD64.zip) or [macOS](https://ci.ros2.org/view/packaging/job/packaging_osx/lastSuccessfulBuild/artifact/ws/ros2-package-osx-x86_64.tar.bz2) and unzip it.
2.Install the dependencies for [windows](https://github.com/RobotWebTools/rclnodejs/blob/develop/appveyor.yml#L17) or [macOS](https://github.com/RobotWebTools/rclnodejs/blob/develop/circle.yml#L5)
3.Build a simple [project](https://github.com/RobotWebTools/rclnodejs/tree/develop/test/cpp) by `ament build .`(I also tried https://github.com/ros2/examples/tree/master/rclcpp)
4.Reported `ModuleNotFoundError: No module named 'catkin_pkg'` on [windows](https://ci.appveyor.com/project/minggangw/rclnodejs/build/develop-639#L231) or [macOS](https://circleci.com/gh/minggangw/rclnodejs-1/573?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link)

The problem can be solved after installing `catkin_pkg`(python3 -m pip install catkin_pkg). If something is wrong, please correct me, thanks!

mikaelarguedas, 2018-06-12:
I think the confusion comes from the fact that @minggangw is referring to the "Install from binaries" instructions and not the "Install from source" ones.

@minggangw In general the installation from binaries should allow you to install everything you need to **run** ROS2 executables (aka the runtime dependencies but not the build dependencies). So it should not be expected to be able to build packages by following only the instructions from the binary install page.

Though as we now have packages depending on catkin_pkg at runtime (https://github.com/ros2/ros2cli/pull/94), I think you're right catkin_pkg should be added to the instructions from binary.

minggangw, 2018-06-13:
I mentioned the "Install from binaries" locates on [Wiki](https://github.com/ros2/ros2/wiki/Installation#binary-packages), not from source code. @mikaelarguedas You understand exactly what I have reported.

Not only build packages, but also any `ament` related commands failed without `catkin_pkg`. Thanks!

dirk-thomas, 2018-06-13:
I added the dependency to the following pages:
* [Linux-Install-Binary](https://github.com/ros2/ros2/wiki/Linux-Install-Binary/_compare/088133be89d54f7c62d15e3e8c40bde6c491517b...fa3edf2216d6dd8e11f242c34951312b55cf7f2a)
* [OSX-Install-Binary](https://github.com/ros2/ros2/wiki/OSX-Install-Binary/_compare/64a01f257b4bc9fcf742e68a18b2a75613e7c99f...182ba2eb284bd613388f9a15e86c1f246f565dee)
* [Windows-Install-Binary](https://github.com/ros2/ros2/wiki/Windows-Install-Binary/_compare/5ade8e963780ff8f314d74e290b0e78dc76cf2fe...1e3612a9be3d42852bb4854fd6c2610709906182)

Please feel free to make changes to wiki pages in the future when you are confident about the modifications. It is already editable by anyone (with a GitHub account).

Assuming that these changes address the problem described in the ticket I will go ahead and close it. Please feel free to comment and if necessary the issue can be reopened if further changes are necessary.

closed 2018-06-13 by dirk-thomas

minggangw, 2018-06-13:
Thanks, that fixed my issue.

asimonov-hitachi, 2019-04-09:
yes, just found your comments on https://github.com/ros2/ros2/issues/660 and realised I was using miniconda python accidentally. going back to system python3