ros2/ros2 #504: Failed to build fastcdr with latest ros2 master branch

gaoethan, 2018-06-12:
## Bug report

**Required Info:**

- Operating System: Ubuntu 16.04
  - 
- Installation type: source
  - 
- Version or commit hash: https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
  - 
- DDS implementation: Fast-RTPS
  - 
- Client library (if applicable): N/A
  - 

#### Steps to reproduce issue

```

$ wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
$ vcs import src < ros2.repos
$ src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install
```

#### Error

```
Process package 'fastcdr' with context:
--------------------------------------------------------------------------------
 source_space => /home/ethan/Myspace/rosdev/master/src/eProsima/Fast-CDR
  build_space => /home/ethan/Myspace/rosdev/master/build/fastcdr
install_space => /home/ethan/Myspace/rosdev/master/install
   make_flags => -j8, -l8
  build_tests => True
--------------------------------------------------------------------------------
+++ Building 'fastcdr'
Running cmake because arguments have changed.
==> '. /home/ethan/Myspace/rosdev/master/build/fastcdr/cmake__build.sh && /usr/bin/cmake -DBUILD_TESTING=1 -DCMAKE_INSTALL_PREFIX=/home/ethan/Myspace/rosdev/master/install /home/ethan/Myspace/rosdev/master/src/eProsima/Fast-CDR' in '/home/ethan/Myspace/rosdev/master/build/fastcdr'
-- Setting build type to 'Release' as none was specified.
-- The CXX compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring Fast CDR
-- Version: 1.0.7
-- To change de version modify the file configure.ac
-- Performing Test SUPPORTS_CXX11
-- Performing Test SUPPORTS_CXX11 - Success
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
CMake Error at /usr/share/cmake-3.5/Modules/CheckIncludeFile.cmake:66 (try_compile):
  Unknown extension ".c" for file

    /home/ethan/Myspace/rosdev/master/build/fastcdr/CMakeFiles/CMakeTmp/CheckIncludeFile.c

  try_compile() works only for enabled languages.  Currently these are:

    CXX

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:233 (check_include_file)
  /usr/share/cmake-3.5/Modules/TestBigEndian.cmake:34 (CHECK_TYPE_SIZE)
  cmake/common/check_configuration.cmake:57 (test_big_endian)
  CMakeLists.txt:152 (check_endianness)

-- Looking for sys/types.h - not found
-- Looking for stdint.h
CMake Error at /usr/share/cmake-3.5/Modules/CheckIncludeFile.cmake:66 (try_compile):
  Unknown extension ".c" for file

    /home/ethan/Myspace/rosdev/master/build/fastcdr/CMakeFiles/CMakeTmp/CheckIncludeFile.c

  try_compile() works only for enabled languages.  Currently these are:

    CXX

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:234 (check_include_file)
  /usr/share/cmake-3.5/Modules/TestBigEndian.cmake:34 (CHECK_TYPE_SIZE)
  cmake/common/check_configuration.cmake:57 (test_big_endian)
  CMakeLists.txt:152 (check_endianness)

-- Looking for stdint.h - not found
-- Looking for stddef.h
CMake Error at /usr/share/cmake-3.5/Modules/CheckIncludeFile.cmake:66 (try_compile):
  Unknown extension ".c" for file

    /home/ethan/Myspace/rosdev/master/build/fastcdr/CMakeFiles/CMakeTmp/CheckIncludeFile.c

  try_compile() works only for enabled languages.  Currently these are:

    CXX

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:235 (check_include_file)
  /usr/share/cmake-3.5/Modules/TestBigEndian.cmake:34 (CHECK_TYPE_SIZE)
  cmake/common/check_configuration.cmake:57 (test_big_endian)
  CMakeLists.txt:152 (check_endianness)

-- Looking for stddef.h - not found
-- Check size of unsigned short
CMake Error at /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:124 (try_compile):
  Unknown extension ".c" for file

    /home/ethan/Myspace/rosdev/master/build/fastcdr/CMakeFiles/CheckTypeSize/CMAKE_SIZEOF_UNSIGNED_SHORT.c

  try_compile() works only for enabled languages.  Currently these are:

    CXX

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:249 (__check_type_size_impl)
  /usr/share/cmake-3.5/Modules/TestBigEndian.cmake:34 (CHECK_TYPE_SIZE)
  cmake/common/check_configuration.cmake:57 (test_big_endian)
  CMakeLists.txt:152 (check_endianness)

-- Check size of unsigned int
CMake Error at /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:124 (try_compile):
  Unknown extension ".c" for file

    /home/ethan/Myspace/rosdev/master/build/fastcdr/CMakeFiles/CheckTypeSize/CMAKE_SIZEOF_UNSIGNED_INT.c

  try_compile() works only for enabled languages.  Currently these are:

    CXX

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:249 (__check_type_size_impl)
  /usr/share/cmake-3.5/Modules/TestBigEndian.cmake:39 (CHECK_TYPE_SIZE)
  cmake/common/check_configuration.cmake:57 (test_big_endian)
  CMakeLists.txt:152 (check_endianness)

-- Check size of unsigned long
CMake Error at /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:124 (try_compile):
  Unknown extension ".c" for file

    /home/ethan/Myspace/rosdev/master/build/fastcdr/CMakeFiles/CheckTypeSize/CMAKE_SIZEOF_UNSIGNED_LONG.c

  try_compile() works only for enabled languages.  Currently these are:

    CXX

  See project() command to enable other languages.
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/CheckTypeSize.cmake:249 (__check_type_size_impl)
  /usr/share/cmake-3.5/Modules/TestBigEndian.cmake:46 (CHECK_TYPE_SIZE)
  cmake/common/check_configuration.cmake:57 (test_big_endian)
  CMakeLists.txt:152 (check_endianness)

CMake Error at /usr/share/cmake-3.5/Modules/TestBigEndian.cmake:51 (message):
  no suitable type found
Call Stack (most recent call first):
  cmake/common/check_configuration.cmake:57 (test_big_endian)
  CMakeLists.txt:152 (check_endianness)

-- Configuring incomplete, errors occurred!
See also "/home/ethan/Myspace/rosdev/master/build/fastcdr/CMakeFiles/CMakeOutput.log".
See also "/home/ethan/Myspace/rosdev/master/build/fastcdr/CMakeFiles/CMakeError.log".
<== Command '. /home/ethan/Myspace/rosdev/master/build/fastcdr/cmake__build.sh && /usr/bin/cmake -DBUILD_TESTING=1 -DCMAKE_INSTALL_PREFIX=/home/ethan/Myspace/rosdev/master/install /home/ethan/Myspace/rosdev/master/src/eProsima/Fast-CDR' failed in '/home/ethan/Myspace/rosdev/master/build/fastcdr' with exit code '1'
<== Command '. /home/ethan/Myspace/rosdev/master/build/fastcdr/cmake__build.sh && /usr/bin/cmake -DBUILD_TESTING=1 -DCMAKE_INSTALL_PREFIX=/home/ethan/Myspace/rosdev/master/install /home/ethan/Myspace/rosdev/master/src/eProsima/Fast-CDR' failed in '/home/ethan/Myspace/rosdev/master/build/fastcdr' with exit code '1'

```

mikaelarguedas, 2018-06-12:
Thanks @gaoethan for reporting this issue.
This seems to be due to recent changes in Fast-CDR that fail with CMake 3.5.
I ticketed the issue upstream at https://github.com/eProsima/Fast-CDR/issues/20

mikaelarguedas, 2018-06-12:
@gaoethan this has now be fixed upstream https://github.com/eProsima/Fast-CDR/commit/e1c127684fee72ebff88398acbcf096f710a7061

Thanks for reporting :+1:

closed 2018-06-12 by mikaelarguedas