ros2/ros2 #506: Use Fast-RTPS 1.6.0 for Bouncy

mikaelarguedas, 2018-06-13:
A release candidate to Fast-RTPS 1.6.0 has been pushed upstream.

Connects to #493 

Running ci on it to check if it plays well with the latest state of the ROS2 codebase:

### Release:
* Linux [image] (test failure unrelated)
* Linux-aarch64 [image]
* macOS [image]
* Windows [image] (New warnings from Fast-RTPS code)

### Debug:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

### Xenial jobs
Xenial Release: Linux [image] (test failure unrelated)

Xenial Debug: Linux [image]

@richiware FYI: I'll post a summary of the things that come up as a follow-up comment. The only regression CI pointed out so far are the new warnings on Windows

mikaelarguedas, 2018-06-14:
@richiware Most failures are not related to the Fast-RTPS changes.
The only outstanding difference is the [new warnings on Windows](https://ci.ros2.org/job/ci_windows/4702/warnings43Result/new/).
Can you please let us know when this is addressed and we will retest it on CI.

Thanks!

mikaelarguedas, 2018-06-15:
New round of CI with new Fast-RTPS 1.6.0 and Fast-CDR 1.07 (both now pushed to the master banch upstream):

## Fast-RTPS + Opensplice

### Release:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Debug:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

richiware, 2018-06-18:
@mikaelarguedas What is the status of this issue? Did you detect something in new FastRTPS release?

mikaelarguedas, 2018-06-18:
@richiware everything is good, no issue detected in the new Fast-RTPS and Fast-CDR releases.

Thanks !

closed 2018-06-18 by mikaelarguedas