ros2/ros2 #508: ROS 2 source build fails with ament_tools - update using colcon?

Martin-Idel-SI, 2018-06-15:
The current `ros2.repos` file fails when building according to the instructions https://github.com/ros2/ros2/wiki/Linux-Development-Setup (also on Windows and OSX) due to a missing build-tool export in `map_msgs` (see https://github.com/ros-planning/navigation_msgs/issues/6). Colcon seems to handle this correctly (tested locally on Windows, Mac and Ubuntu). 

I could update the build from source guide to use colcon instead of `ament_tools` but I wanted to ask what your plan was first.

johnmarkwayve, 2018-06-15:
Please add some build instructions that are likely to work with recent `master` pulls. It seems like every few days a couple of things break with the build system I have at the moment.

Martin-Idel-SI, 2018-06-15:
@johnmarkwayve Here is a working setup: 

In addition to the dependencies provided in the wiki, use
`pip install colcon-common-extensions` 

Then, in order to build ros2, instead of 
`src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install`
use the approximately equivalent
`colcon build --merge-install --symlink-install --cmake-args " -DBUILD_TESTING=1"`

Since I don't know whether colcon is alread the official build tool in the sense everybody should use it, I'm waiting for confirmation.

mikaelarguedas, 2018-06-15:
> The current ros2.repos file fails when building according to the instructions https://github.com/ros2/ros2/wiki/Linux-Development-Setup (also on Windows and OSX) due to a missing build-tool export in map_msgs (see ros-planning/navigation_msgs#6). Colcon seems to handle this correctly (tested locally on Windows, Mac and Ubuntu).

It would be great to patch the faulty package.xml to make sure it keeps building with `ament_tools`

> I could update the build from source guide to use colcon instead of ament_tools but I wanted to ask what your plan regarding moving to colcon was first.

We are about to do this all at once (updating all tutorials), once the last colcon release is out (which should be very shortly).

> Please add some build instructions that are likely to work with recent master pulls. It seems like every few days a couple of things break with the build system I have at the moment.

Apologies, this happened in the last few days as we migrated our infrastructure to the new build tool for testing the upcoming Bouncy release.
Once the tutorials align with our Continuous Integration setup this should not happen again.

johnmarkwayve, 2018-06-15:
@Martin-Idel-SI Thanks, I'll see if I can massage it into a working state with colcon

@mikaelarguedas Thank you, I look forward to it :) In the mean time, is there anywhere I can see the scripts your CI is running?

mikaelarguedas, 2018-06-15:
> Thank you, I look forward to it :) In the mean time, is there anywhere I can see the scripts your CI is running?

All our CI scripts are hosted on https://github.com/ros2/ci/
The changes to migrate from ament to colcon is here: https://github.com/ros2/ci/pull/132
Though there were a few follow-up fixes so just looking at the current state of the code on the master branch may be better

dirk-thomas, 2018-06-15:
I would recommend to use `ament_tools` until the instructions have been updated. ROS 2 for Bouncy aims to support building with `ament_tools` so if that is not the case at the moment it needs to be fixed.

The specific problem has been fixed in ros-planning/navigation_msgs#7.

closed 2018-06-15 by dirk-thomas