ros2/ros2 #509: image callback in python very slow

rancheng, 2018-06-17:
we are publishing an image in ROS2 with a resolution of 600 by 600. Using rclpy in Python, the callback is very slow. We ran the ros2/ros1 bridge and made a very similar ROS1 subscriber and the performance is perfect.

We tried using opensplice and fastRTPS, but since using the ROS2/ROS1 bridge callback works perfectly, we are confident it isn't the transport layer and the problem is probably with the python subscriber cash.

The cam2image example uses such a low resolution that the problem doesn't appear.

dirk-thomas, 2018-06-20:
Please provide a reproducible example.

dirk-thomas, 2018-07-05:
Closing due to no response. Please feel free to comment with the requested information and this ticket can be reopened.

closed 2018-07-05 by dirk-thomas

Farrel7786729, 2018-07-15:
I'm experiencing exactly the same issue. My setup is VREP->ROS1->ROS2->rclpy. I receive decent fps with max resolution of 64*64. Anything more than that results in huge frame skips. 

It seems that that VREP->ROS1 works ok. So does ROS1->ROS2 bridge. 

With default setup my python code receives only each 10th image (my test resolution is 1024*512). When I set up QoS with KEEP_ALL policy, frameskip disappears, but my test module starts to "leak" memory, due to growing message queue, I assume. Although all frames are received, frame rate remains 1-2 fps. Here's my question on ros forum for reference https://answers.ros.org/question/296715/ros-image-message-rate-is-low/. 

Since environment is a bit tricky to reproduce, I'll try to come up with simple MWE.

Farrel7786729, 2018-07-25:
Looks like I've found an issue. I still need to run a couple of tests, though, just in order to be sure that nothing else causes delay. 
_image.py, which seems to be generated file, contains quite complex assert. For my distro it starts on line 199, and lines which are actually responsible for lag are 205, 206: 

```
            assert \
                ((isinstance(value, Sequence) or
                  isinstance(value, Set) or
                  isinstance(value, UserList)) and
                 not isinstance(value, str) and
                 not isinstance(value, UserString) and
                 all(isinstance(v, int) for v in value) and             ######## bottleneck
                 all(val >= 0 and val < 256 for val in value)), \   ########
                "The 'data' field must be a set or sequence and each value of type 'int' and each unsigned integer in [0, 255]"
```
Since it needs to be fixed in generator config, I cannot provide actual patch yet.

rancheng, 2018-07-26:
We have tried multiple things, QoS, change the cache size, etc. the thing we are doing is publishing a 600*600 image from ros1, and bridge it to ros2 in windows.
`class ImgSub(Node):

    def __init__(self):
        super().__init__('img_listener')

        reliability_policy = QoSReliabilityPolicy.RMW_QOS_POLICY_RELIABILITY_RELIABLE
        #reliability_policy = QoSReliabilityPolicy.RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT
        # history_policy = QoSHistoryPolicy.RMW_QOS_POLICY_HISTORY_KEEP_ALL
        history_policy = QoSHistoryPolicy.RMW_QOS_POLICY_HISTORY_KEEP_LAST
        custom_qos_profile = qos.QoSProfile()
        # custom_qos_profile.history = history_policy
        # custom_qos_profile.reliability = reliability_policy
        # custom_qos_profile.depth = 1

        # self.image_sub = self.create_subscription(Image, 'vehicle_leftImage_alternate', self.image_callback,
        #                                     qos_profile=custom_qos_profile)
        #
        # self.pose_sub = self.create_subscription(PoseStamped, 'vehicle_pose_alternate', self.pose_callback,
        #                                     qos_profile=custom_qos_profile)
        #
        self.image_sub = self.create_subscription(Image, 'vehicle_leftImage', self.image_callback)

    def image_callback(self, msg):
        try:
            #print('got image: ')
            key = float(msg.header.stamp.sec + msg.header.stamp.nanosec / 1000000000)
            image_np = np.array(msg.data, np.uint8)
            #image_np = cv2.imdecode(image_np, cv2.IMREAD_COLOR)
            image_np = image_np.reshape((msg.height, msg.width, 3)) #this might be  wrong order
            #cv2.imwrite("D:\\test.png", image_np)
            cv2.imshow('image', image_np)
            #cv2.waitKey(0)
            cv2.waitKey(1)
            #self.image_dict[key] = image_np
            time = msg.header.stamp.sec + msg.header.stamp.nanosec / 1000000000.0
            #print('got image: ' + str(time))
            #self.get_last_synced_state_key()
        except Exception as e:
            print(e)
`

Try publish a higher resolution, say 600*600 picture, to see that drop on fps.

Farrel7786729, 2018-07-26:
@rancheng, I won't be able to test your example untill monday. Just as an idea, did you try to run python interpreter with -O option? I did not notice that the code I mentioned was under `if __debug__:` last time. There's way more asserts which slow down execution.

mikaelarguedas, 2018-07-30:
>  did you try to run python interpreter with -O option? I did not notice that the code I mentioned was under `if __debug__:` last time. 

:+1: We put all the setter checks in `if __debug__` blocks to work around this type of performance issue. Note that this is only available as of ROS Bouncy (and not in Ardent).

kervel, 2018-11-30:
btw, if others are bitten by this issue, a temporary workaround is creating a small binding using pybind11 using buffer protocol to access array members as numpy arrays. we do this for the data-intensive python nodes now.

https://gist.github.com/kervel/75d81b8c34e45a19706c661b90d02548

JacobJBeck, 2019-02-26:
Why is this issue closed? I'm still experiencing this. I thought it might be related to the DDS implementation (based on [this issue](https://github.com/ros2/rmw_fastrtps/issues/81)), so I tested FastRTPS and OpenSplice using `cam2image` and my own python script which just shows a preview. I used `ros2 topic hz /image` to measure the rate, for which the results are shown below.

| DDS        | Resolution | FPS  |
|------------|------------|------|
| FastRTPS   |  320x240   | 30.0 |
| FastRTPS   | 800x600    | 4.7  |
| FastRTPS   | 1920x1080  | 1.0  |
| OpenSplice | 320x240    | 28.4 |
| OpenSplice | 800x600    | 4.3  |
| OpenSplice | 1920x1080  | 1.0  |

Versions used:
rclpy: 9efd331ec0a596e5f42d163047c4fdc6baa8c117
rmw_fastrtps: ea4fb7cfaa671a2118a7d54203c786d890d8ed38
rmw_opensplice: 25287a690e8e91e6f56da51944829957c629708f

dirk-thomas, 2019-03-27:
The performance bottle neck to convert the incoming message into a Python object has been resolved in ros2/rosidl_python#35.