ros2/ros2 #511: Update ROS 2 wiki to use colcon instead of ament_tools

dirk-thomas, 2018-06-18:
I will update the description with changes committed to the wiki repository while making progress on this topic:

dirk-thomas, 2018-06-18:
I think the wiki changes are complete. Putting this in review.

dhood, 2018-06-18:
I made these changes: https://github.com/ros2/ros2/wiki/Colcon-Tutorial/_compare/07ad48162089e7bca341661f2ad51be8ef3d1f46...3db804288f812b7031b7cfedc21fc7f7cca46918
https://github.com/ros2/ros2/wiki/Linux-Development-Setup/_compare/740ef4be8574d2cd9e070f7356cb36a4313dea81...338dde366ebfa1e2603b08f317b06a953c986e6a

This page has been left out: https://github.com/ros2/ros1_bridge/blob/master/README.md#build-the-bridge-from-source

dirk-thomas, 2018-06-18:
> Colcon-Tutorial

I would suggest to revert the change in [this line](https://github.com/ros2/ros2/wiki/Colcon-Tutorial/_compare/07ad48162089e7bca341661f2ad51be8ef3d1f46...3db804288f812b7031b7cfedc21fc7f7cca46918#diff-1ceb02b89e5af957409f6965f59b5118R98) since this is not a "decision" of `colcon` but a decision of `ament_cmake` to not support the concept of a devel space.

Just for the record: [this change](https://github.com/ros2/ros2/wiki/Colcon-Tutorial/_compare/07ad48162089e7bca341661f2ad51be8ef3d1f46...3db804288f812b7031b7cfedc21fc7f7cca46918#diff-1ceb02b89e5af957409f6965f59b5118R200) is not necessary. An `AMENT_IGNORE` files continues to work just fine. The same applies to the second change.

> This page has been left out:

This ticket only captures the "ROS 2 wiki" as per title. External documentation needs to be updated to - there is a separate bullet for that in #510.

dhood, 2018-06-18:
> I would suggest to revert the change in this line since this is not a "decision" of colcon but a decision of ament_cmake to not support the concept of a devel space.

https://github.com/ros2/ros2/wiki/Colcon-Tutorial/_compare/3db804288f812b7031b7cfedc21fc7f7cca46918...d106a2c74a32c42e45afa4e482f5dd6076471b9e

> Just for the record: this change is not necessary. An AMENT_IGNORE files continues to work just fine. The same applies to the second change.

OK, I didn't realise, thanks!

dhood, 2018-06-18:
I am installing colcon on my macbook. Is there an additional step I should do to get a `colcon` executable on the path?

I followed https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/190974de0ce22841cab88cb3160d1ed54a19565d...16726fd537e73b5b46700c1ab6d6c339eb99714a but it doesn't seem to be sufficient:

```
$ colcon
bash: colcon: command not found

$ python3 -m pip freeze | grep colcon
colcon-argcomplete==0.2.1
colcon-bash==0.3.0
colcon-cmake==0.2.1
colcon-common-extensions==0.2.0
colcon-core==0.3.1
colcon-defaults==0.2.0
colcon-devtools==0.2.0
colcon-library-path==0.2.0
colcon-metadata==0.2.0
colcon-notification==0.2.2
colcon-output==0.2.0
colcon-package-information==0.2.0
colcon-package-selection==0.2.0
colcon-parallel-executor==0.2.1
colcon-powershell==0.3.0
colcon-python-setup-py==0.2.0
colcon-recursive-crawl==0.2.0
colcon-ros==0.2.6
colcon-test-result==0.3.0
colcon-zsh==0.3.0
```

(I'm posting here because maybe the wiki is just missing a step; if it's more of a packaging issue we can move the discussion).

<details><summary>Installation log looks fine</summary>
<p>

```
$ python3 -m pip install colcon-common-extensions
Collecting colcon-common-extensions
  Downloading https://files.pythonhosted.org/packages/04/f7/4ab9b80b14ad3dd49b85e8a7ab3d8e65fe514889a500d0ae611583bc4fdb/colcon_common_extensions-0.2.0-py3-none-any.whl
Collecting colcon-devtools (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/e7/af/6d0d0833499cc72c72389af7e6a95b169299cd569c07440674617097c5e0/colcon_devtools-0.2.0-py3-none-any.whl
Collecting colcon-python-setup-py (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/ff/dd/844739af48165a7d5b432eaeeb6618e53987f7b54daa4db0f29e4290a4b6/colcon_python_setup_py-0.2.0-py3-none-any.whl
Collecting colcon-defaults (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/46/fb/3db7472ab8f72a61af66ccd4f27dd7057d431be5dc4857327f5b3becc690/colcon_defaults-0.2.0-py3-none-any.whl
Collecting colcon-metadata (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/c9/9d/5f3c9980e14cadb5ae48ef071062df820677931cd42e38baf4ec2700cc40/colcon_metadata-0.2.0-py3-none-any.whl
Collecting colcon-bash; sys_platform != "win32" (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/5f/7f/237a16f904859679bd93153b8193720e3292d4a4da21a413726615e377f3/colcon_bash-0.3.0-py3-none-any.whl
Collecting colcon-test-result (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/3d/7e/2c96d79cd6f4114140bc22d9535aa64611f634ee01a4b41b4cb2c162f6bb/colcon_test_result-0.3.0-py3-none-any.whl
Collecting colcon-zsh; sys_platform != "win32" (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/9e/b4/6af7131cc6ab38f39863783f8413bcd0c271ba628f39beb766537123d24a/colcon_zsh-0.3.0-py3-none-any.whl
Collecting colcon-package-information (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/e3/95/889b0906d6063df35cb21ad959307ff56f5e8da698ea73b6eb3426fe3b9c/colcon_package_information-0.2.0-py3-none-any.whl
Collecting colcon-output (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/f5/40/b1089f9e166e5d2b39ec731916cd30672a480bb41b547e33c9f645172270/colcon_output-0.2.0-py3-none-any.whl
Collecting colcon-library-path (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/8c/06/1e323a37096f89bb66d53abc79b3d9f1e385c761d54ef28f43d2f2a91977/colcon_library_path-0.2.0-py3-none-any.whl
Collecting colcon-notification (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/f3/eb/3ca1ebd4af812426ca4d3284ca11c55f2422f6d575c0bcfd422161504e9c/colcon_notification-0.2.2-py3-none-any.whl
Collecting colcon-package-selection (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/54/29/58188c40330d4a29f825eb2d8fdc7da48154b4d5b960a2c099528617ece9/colcon_package_selection-0.2.0-py3-none-any.whl
Collecting colcon-parallel-executor (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/12/f0/a909cbb8d96552c2e4c7b8f531fcfef504cd4d62d930df068cf363f5bdb6/colcon_parallel_executor-0.2.1-py3-none-any.whl
Collecting colcon-core (from colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/df/b5/6fd03c236d6576a41a880b43465777d38e58a53cbcd0ab5fa8039e2936c5/colcon_core-0.3.1-py3-none-any.whl (89kB)
    100% |████████████████████████████████| 92kB 3.5MB/s
Collecting colcon-powershell (from colcon-common-extensions)
[... 12 lines cut ...]
  Downloading https://files.pythonhosted.org/packages/c5/81/2dc013556d53c9f9a1955e66f8620a05690eb6de9efc9fdb6cd748c66d24/pytest_rerunfailures-4.1-py2.py3-none-any.whl
Requirement already satisfied: EmPy in /usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from colcon-core->colcon-common-extensions)
Requirement already satisfied: pytest in /usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages (from colcon-core->colcon-common-extensions)
Collecting pytest-repeat (from colcon-core->colcon-common-extensions)
  Downloading https://files.pythonhosted.org/packages/e8/cd/e86554eb58d067c3b2fe0be395ad324a7ec46be4278a29da0b848954aa4a/pytest_repeat-0.4.1-py2.py3-none-any.whl
[... cut ...]
Successfully installed colcon-argcomplete-0.2.1 colcon-bash-0.3.0 colcon-cmake-0.2.1 colcon-common-extensions-0.2.0 colcon-core-0.3.1 colcon-defaults-0.2.0 colcon-devtools-0.2.0 colcon-library-path-0.2.0 colcon-metadata-0.2.0 colcon-notification-0.2.2 colcon-output-0.2.0 colcon-package-information-0.2.0 colcon-package-selection-0.2.0 colcon-parallel-executor-0.2.1 colcon-powershell-0.3.0 colcon-python-setup-py-0.2.0 colcon-recursive-crawl-0.2.0 colcon-ros-0.2.6 colcon-test-result-0.3.0 colcon-zsh-0.3.0 pytest-repeat-0.4.1 pytest-rerunfailures-4.1
```

</p>
</details>

dirk-thomas, 2018-06-18:
> I am installing colcon on my macbook. Is there an additional step I should do to get a colcon executable on the path?

I am not familiar how macOS handles the pip locations. On Linux when installed globally into `/usr` it works out-of-the-box and when installed into the user-specific location you need to add `~/.local/bin/` to your `PATH`.

dhood, 2018-06-18:
I can work on the osx instructions if you can test it on windows and update those instructions? (unless the colcon executable is known to work on windows)

dirk-thomas, 2018-06-18:
Afaik it works on Windows since the packages are getting installed "globally" into `c:\python36\lib\site-packages`.

dhood, 2018-06-18:
Have you tested it already? If not, could you please check that the pip installation works on windows, because I don't see any console script being installed anywhere on osx, it's not just that it's not on the path. Windows might have the same issue.

dirk-thomas, 2018-06-18:
This is exactly what happens on CI. Without installing a Windows machine from scratch I don't know what else I should check.

dhood, 2018-06-18:
My issue was that my python (installed from brew) installs packages to `/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages` and, as suspected, the installation dir of `colcon-core` doesn't get on the path. This doesn't seem to be the typical installation (not sure what caused the difference), so for most people the installation instructions are good as they are. Installation in a virtual environment works fine for me.

I will try to cleanup my python installation but if that doesn't work I'll add a note to the troubleshooting section to run `python3 -m pip install -U colcon-core` so that then the following warning gets printed and suggests the corrective action:

```
Installing collected packages: colcon-core
  The script colcon is installed in '/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed colcon-core-0.3.1
```
, which _doesn't_ get printed when it's installed as a dependency of `colcon-common-extensions` (output in previous comment).

The wiki changes look good though 👍

dirk-thomas, 2018-06-18:
I will close this ticket assuming all ROS 2 wiki pages have been updated accordingly. If there are any cases remaining please feel free to update them and post a link to the change here.

closed 2018-06-18 by dirk-thomas

ALxander19, 2019-07-02:
I still have the error! Could someone give my a hand, I install colcon on Mac OSX and what I get is this:
-bash: colcon: command not found

jacobperron, 2019-07-03:
@ALxander19 As per our [support guidelines](http://wiki.ros.org/Support), I think a more appropriate place to ask for help is [ROS Answers](http://answers.ros.org). This ticket has been closed for quite some time and not the most visible place for troubleshooting.