ros2/ros2 #514: ros2 node list caching

dhood, 2018-06-20:
This is the first set of release testing we have done with the changes from https://github.com/ros2/ros2/issues/438 in how we detect other nodes for `ros2 node list`.

I have noticed that sometimes nodes that have shutdown are still detected. The behaviour depends on the RMW implementation being used in the daemon, and in the node.

These are my results from testing on Xenial from-source using `./install/lib/demo_nodes_cpp/talker` and then `ros2 daemon stop` + `ros2 node list`.

Expected behaviour is that after the talker is killed, `ros2 node list` does not show it in the list. This is not always the case.

| Node (row) | _Daemon (column)_ | FastRTPS | Connext | OpenSplice |
| --- | --- | --- | --- | --- |
| **FastRTPS** | | [image] node removed from list | [image] node not removed | [image] node not removed |
| **Connext** | | [image] node removed from list | [image] node removed from list | [image] node not removed |
| **OpenSplice** | | [image] node removed after ~10s | [image] node removed from list | [image] node not removed |

From memory, this has had some cross-vendor issues even before we swapped, but I don't think it was as widespread as this.

dirk-thomas, 2018-06-20:
> after the talker is killed

Can you clarify this statement? I would expect if you actual *kill* the talker that is has no way to notify other participants of it shutting down. So I would expect it to continue to be listed until it "ages out" due to never responding anymore (no what those timeouts are for each rmw impl.)

I also don't think this is related to where me transfer the node information.

dhood, 2018-06-20:
I used bad wording. I meant stopped, with ctrl-c, not with sigkill.

mm318, 2018-12-27:
I noticed that when running `examples_rclcpp_minimal_publisher/publisher_lamda` with FastRTPS and RTI Connext, there were HEARTBEATS and ACKNACKS being sent. With OpenSplice, there weren't.

mm318, 2019-01-04:
I have a workaround here: https://github.com/mm318/rmw_opensplice/pull/2/files

I think the underlying problem is in OpenSplice though (but I may be wrong and that OpenSplice is behaving as intended).

dirk-thomas, 2019-08-01:
ros2/rmw_opensplice#275 aims to address the issue with OpenSplice.