ros2/ros2 #516: FastRTPS fails to find asio and tinyxml2 on Windows

wjwwood, 2018-06-23:
I am testing the Windows from source installation tutorial:

asdf

And I started in a fresh Windows VM.

When building FastRTPS I get this:

```
-- Setting build type to 'Release' as none was specified.
-- Selecting Windows SDK version 10.0.17134.0 to target Windows 10.0.10240.
-- The C compiler identification is MSVC 19.14.26431.0
-- The CXX compiler identification is MSVC 19.14.26431.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.14.26428/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.14.26428/bin/Hostx86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.14.26428/bin/Hostx86/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.14.26428/bin/Hostx86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring Fast RTPS
-- Version: 1.6.0
-- To change de version modify the file configure.ac
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- fastcdr library found...
-- Could NOT find asio (missing: ASIO_INCLUDE_DIR) 
-- Could NOT find tinyxml2 (missing: TINYXML2_INCLUDE_DIR) 
-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE  
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
ASIO_INCLUDE_DIR (ADVANCED)
   used as include directory in directory C:/dev/ros2/src/eProsima/Fast-RTPS/src/cpp
   used as include directory in directory C:/dev/ros2/src/eProsima/Fast-RTPS/src/cpp
   used as include directory in directory C:/dev/ros2/src/eProsima/Fast-RTPS/src/cpp
   used as include directory in directory C:/dev/ros2/src/eProsima/Fast-RTPS/src/cpp
TINYXML2_INCLUDE_DIR (ADVANCED)
   used as include directory in directory C:/dev/ros2/src/eProsima/Fast-RTPS/src/cpp
   used as include directory in directory C:/dev/ros2/src/eProsima/Fast-RTPS/src/cpp
   used as include directory in directory C:/dev/ros2/src/eProsima/Fast-RTPS/src/cpp
   used as include directory in directory C:/dev/ros2/src/eProsima/Fast-RTPS/src/cpp

-- Configuring incomplete, errors occurred!
See also "C:/dev/ros2/build/fastrtps/CMakeFiles/CMakeOutput.log".
See also "C:/dev/ros2/build/fastrtps/CMakeFiles/CMakeError.log".
```

I did install `asio` and `tinyxml2` (along with the others) from our chocolately packages (I downloaded and installed them yesterday).

```
C:\dev\ros2>choco list --local-only
Chocolatey v0.10.11
asio 1.12.1
chocolatey 0.10.11
chocolatey-core.extension 1.3.3
chocolatey-windowsupdate.extension 1.0.2
cmake 3.11.4
cmake.install 3.11.4
cppcheck 1.82.0
curl 7.60.0
eigen 3.3.4
git 2.17.1.2
git.install 2.17.1.2
KB2919355 1.0.20160915
KB2919442 1.0.20160915
KB2999226 1.0.20170509
KB3033929 1.0.3
KB3035131 1.0.1
patch 2.5.9
python 3.6.5
python3 3.6.5
tinyxml-usestl 2.6.2
tinyxml2 6.0.0
vcredist140 14.12.25810
vcredist2015 14.0.24215.20170201
23 packages installed.
```

I'm building with `colcon build`.

Does anyone have any idea why this wouldn't work out of the box?

Can someone try to reproduce on their Windows machine?

nuclearsandwich, 2018-06-23:
When I first saw this I thought that maybe https://github.com/eProsima/Fast-RTPS/pull/233 had merged and we were accidentally relying on the third-party versions on Windows but vcs import doesn't pull in submodules (at least for me) so that's unlikely.

Those two packages are in the global cmake package registry, which on Windows uses the Windows registry. The last update cycle I think I saw registry variables fail to update properly but I had it chalked up to operator error of some kind. You should find entries at `Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Kitware\CMake\Packages\TinyXML2` and `Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Kitware\CMake\Packages\Asio` with an empty (Default) key and a PackageDir key pointing to the `share/cmake` path of each installed package.

If those values are present then I did see some recent colcon PRs working on CMAKE_MODULE_PATH (https://github.com/colcon/colcon-cmake/pull/19 and https://github.com/colcon/colcon-cmake/pull/20) but from a very cursory glance I don't think those would affect package discovery during build.

wjwwood, 2018-06-26:
Looks like the registry keys were missing. I uninstall `asio` and reinstalled it and the key was there. I had to use `-n` on a second uninstall attempt for `tinyxml2`.

I wonder if a Windows update could remove them?

wjwwood, 2018-06-26:
I had to do the same for `eigen` (including the `-n` to uninstall after uninstall failed), I imagine I'll have to do the same (at least reinstall) for `tinyxml`.

wjwwood, 2018-06-26:
So I solved this by reinstalling the chocolatey packages, and I added a note in the troubleshooting section:

https://github.com/ros2/ros2/wiki/Windows-Development-Setup/_compare/d4de94d08ff5094ea1ef5a27b6a83fb9b6d6e47f...e82818163c6c99b8bb53b972e8de841909ec20ac

closed 2018-06-26 by wjwwood

nuclearsandwich, 2018-06-26:
I also ran into this on my test setup and arrived at the same solution. I think the registry key may be silently failing to insert if the inserted path doesn't exist. I am going to proceed with the tutorial for now and then come back to this since @wjwwood has posted a workaround.

wjwwood, 2018-07-10:
This is from @serge-nikulin:

> William,
>
> I have found the reason for this one: https://github.com/ros2/ros2/wiki/Windows-Development-Setup#cmake-packages-unable-to-find-asio-tinyxml2-tinyxml-or-eigen.
>
> Basically, CMake package has to be installed before `asio` and  etc packages https://github.com/ros2/ros2/wiki/Windows-Install-Binary#install-dependencies.
>
> Only in this case `asio` will register itself in `cmake`
>
> The procedure:
>
> - install cmake
> - set path to cmake
> - exit admin window
> - start new admin window (test that cmake works)
> - proceed to installation of `asio` and etc. (note that `asio` will spend some time in cmake registration)
>
> Note that the above will break installation of binary ROS2.

@nuclearsandwich is there anything we could change to help with this (such that asio always creates the keys, even if cmake was installed second)?

nuclearsandwich, 2018-07-11:
> @nuclearsandwich is there anything we could change to help with this (such that asio always creates the keys, even if cmake was installed second)?

In theory it is always forcing the installation of keys. Although it's possible that logic is insufficient. I think the right answer is for our chocolatey packages to depend on cmake via chocolatey which should ensure that cmake is installed (and the package registry configured) before our packages are installed.

mikaelarguedas, 2018-07-11:
> In theory it is always forcing the installation of keys. Although it's possible that logic is insufficient. I think the right answer is for our chocolatey packages to depend on cmake via chocolatey [...]

Is this something that we envision by Bouncy first patch release?

Otherwise we should probably update the instructions to preemptively install cmake in the default installation flow rather than expecting most users to refer to the troubleshooting section.
As we already have "Install Visual Studio" in the binary installation instructions I think the overhead of adding CMake to that page before installing asio/tinyxml is acceptable.

nuclearsandwich, 2018-07-11:
> Is this something that we envision by Bouncy first patch release?

I think so. I would like to get to it before this week is over. It should be a matter of updating the manifest file for each package and then re-packaging them, the real work is in testing that the fix is sufficient.

mikaelarguedas, 2018-07-12:
> I think so. I would like to get to it before this week is over.

Sounds good :+1:

As I was modifying the installation instructions for other reasons I took the opportunity to move the CMake install to the Install-Binary page: https://github.com/ros2/ros2/wiki/Windows-Install-Binary/_compare/3a3b739093097278cff434451021aca9d51ab8a8...6e37b0bb7873e0a9dc8bc861c6560a8d76e653b6