ros2/ros2 #517: Build failure during rmw_fastrtps_cpp step: ld: library not found for -ltinyxml2

J-Rojas, 2018-06-26:
## Bug report

**Required Info:**

- Operating System:
  - MacOSX High Sierra
- Installation type:
  - Source - Master branch
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/9d4725115f86c4248f2d71bef42a2fa740a56ba3
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
I've followed the Wiki instructions to install packages via Brew and set up my environment as described there. 
I then execute colcon.
The build fails because it cannot locate the tinyxml2 library.
```

colcon build --symlink-install
Starting >>> ament_package
Starting >>> ament_flake8
Starting >>> fastcdr
Starting >>> poco_vendor
Starting >>> ament_cppcheck
Starting >>> gtest_vendor
Starting >>> osrf_pycommon
Starting >>> console_bridge
Finished <<< ament_package [1.72s]                                                                                                                                                             
Finished <<< ament_flake8 [1.73s]
Starting >>> ament_cmake_core
Starting >>> ament_pep257
Finished <<< ament_cppcheck [1.88s]                                                                                                                                      
Starting >>> tinyxml_vendor
Finished <<< poco_vendor [2.82s]                                                                                                                                                          
Finished <<< gtest_vendor [2.85s]
Finished <<< fastcdr [2.85s]
Finished <<< osrf_pycommon [2.85s]
Starting >>> fastrtps                                                                                                                                     
Starting >>> gmock_vendor
Starting >>> urdfdom_headers
Starting >>> orocos_kdl
Finished <<< tinyxml_vendor [1.93s]                                                                                                                                                         
Finished <<< ament_cmake_core [2.03s]
Starting >>> ament_cmake_test
Starting >>> ament_cmake_libraries
Finished <<< console_bridge [3.95s]                                                                                                                                                    
Starting >>> ament_cmake_export_libraries
Finished <<< ament_pep257 [2.63s]                                                                                                                                                   
Finished <<< gmock_vendor [1.59s]
Starting >>> ament_cmake_include_directories
Starting >>> ament_cmake_export_definitions                                                                                                                                                          
Finished <<< urdfdom_headers [1.81s]
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_test [1.07s]                                                                                                                                   
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_libraries [1.13s]                                                                                                                                               
Finished <<< ament_cmake_export_libraries [0.99s]
Starting >>> ament_cmake_export_dependencies
Starting >>> ament_cmake_python
Finished <<< ament_cmake_include_directories [0.50s]
Starting >>> ament_cmake_export_interfaces                                                                                                          
Finished <<< orocos_kdl [2.21s]
Finished <<< ament_cmake_export_definitions [0.40s]
Starting >>> ament_cmake_target_dependencies
Starting >>> ament_copyright
Finished <<< ament_cmake_export_include_directories [0.41s]
Starting >>> uncrustify
Finished <<< ament_cmake_export_link_flags [0.83s]                                                                                                                                                    
Starting >>> ament_index_python
[... 171 lines cut ...]
The file /usr/local/share/colcon-notification/colcon-terminal-notifier.app does not exist.
```

#### Expected behavior

The build correctly locates tinyxml2 and configures the CMake files.

#### Actual behavior

Build fails. There are other submodules that correctly locate tinyxml2 within this project. For example the fast-rtps submodule builds correctly:

```
-- Found asio: /usr/local/include  
-- Found TinyXML2: /usr/local/lib/cmake/tinyxml2
```

However the failure occurs in the rmw_fastrtps_cpp submodule, which appears to not configure the cmake files to search for tinyxml2 in the proper location.

#### Additional information

Link.txt from the failed build directory. The -ltinyxml2 is added however there's no reference to the system path for this library.

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -dynamiclib -Wl,-headerpad_max_install_names  -o librmw_fastrtps_cpp.dylib -install_name @rpath/librmw_fastrtps_cpp.dylib CMakeFiles/rmw_fastrtps_cpp.dir/src/client_service_common.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/demangle.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_client.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_participant.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_publisher.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_service.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/get_subscriber.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/identifier.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/namespace_prefix.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/qos.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_logging.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_client.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_compare_gids_equal.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_count.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_get_gid_for_publisher.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_get_implementation_identifier.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_guard_condition.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_init.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_node.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_node_names.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_publish.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_publisher.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_request.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_response.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_serialize.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service_names_and_types.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_service_server_is_available.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_subscription.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_take.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_topic_names_and_types.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_trigger_guard_condition.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_wait.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/rmw_wait_set.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/ros_message_serialization.cpp.o CMakeFiles/rmw_fastrtps_cpp.dir/src/type_support_common.cpp.o -Wl,-rpath,/Users/jrojas/work/ros2_ws/install/fastrtps/lib -Wl,-rpath,/Users/jrojas/work/ros2_ws/install/rosidl_typesupport_introspection_c/lib -Wl,-rpath,/Users/jrojas/work/ros2_ws/install/rosidl_typesupport_introspection_cpp/lib -Wl,-rpath,/Users/jrojas/work/ros2_ws/install/rcutils/lib -Wl,-rpath,/Users/jrojas/work/ros2_ws/install/rmw/lib -Wl,-rpath,/Users/jrojas/work/ros2_ws/install/rosidl_generator_c/lib -Wl,-rpath,/Users/jrojas/work/ros2_ws/install/fastcdr/lib /Users/jrojas/work/ros2_ws/install/fastrtps/lib/libfastrtps.1.6.0.dylib /Users/jrojas/work/ros2_ws/install/rosidl_typesupport_introspection_c/lib/librosidl_typesupport_introspection_c.dylib /Users/jrojas/work/ros2_ws/install/rosidl_typesupport_introspection_cpp/lib/librosidl_typesupport_introspection_cpp.dylib /Users/jrojas/work/ros2_ws/install/rcutils/lib/librcutils.dylib /Users/jrojas/work/ros2_ws/install/rmw/lib/librmw.dylib /Users/jrojas/work/ros2_ws/install/rosidl_generator_c/lib/librosidl_generator_c.dylib /Users/jrojas/work/ros2_ws/install/fastcdr/lib/libfastcdr.1.0.7.dylib -ltinyxml2 
```

mikaelarguedas, 2018-07-03:
@J-Rojas from the console output it looks like fastrtps is using it's own version of asio and TinyXML instead of the system one.
This should not happen except if you cloned Fast-RTPS recursively including it's third party submodules.
Can you try again by cloning only Fast-RTPS but not the submodules.

PS: If you update Fast-RTPS to the latest master, it should also work as it should now be using the system version of the dependencies and not the vendored ones as of https://github.com/eProsima/Fast-RTPS/commit/ea57f15aaf689144cb8e75aa8d23e7b6ed832b3a

J-Rojas, 2018-07-03:
@mikaelarguedas I'm not following you exactly on why Fast-RTPS is the problem. 

The build error is in the rmw_fastrtps_cpp package. I've searched this package for references to tinyxml2, but didn't find any. I'm not certain where it's picking up the tinyxml2 link dependency.

```
[100%] Linking CXX shared library librmw_fastrtps_cpp.dylib
ld: library not found for -ltinyxml2
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [librmw_fastrtps_cpp.dylib] Error 1
make[1]: *** [CMakeFiles/rmw_fastrtps_cpp.dir/all] Error 2
make: *** [all] Error 2
```

Fast-RTPS is using my locally installed files. Re: submodule fetch, I followed the instructions on the [OSX build page](https://github.com/ros2/ros2/wiki/OSX-Development-Setup)  which make use of the `vcs import` to pull all of the repos (and their submodules). I deleted the submodules and also double checked that the Fast-RTPS git tree was checked out out as master (it was git hash you pointed to already). I then rebuilt (after cleaning the build folder) and still encountered the same error above.
Below is the successful build of the Fast-RTPS package.
 
```
-- Setting build type to 'Release' as none was specified.
-- The C compiler identification is AppleClang 9.1.0.9020039
-- The CXX compiler identification is AppleClang 9.1.0.9020039
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring Fast RTPS
-- Version: 1.6.0
-- To change de version modify the file configure.ac
-- Performing Test SUPPORTS_CXX14
-- Performing Test SUPPORTS_CXX14 - Success
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- fastcdr library found...
-- Found asio: /usr/local/include  
-- Found TinyXML2: /usr/local/lib/cmake/tinyxml2
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/jrojas/work/ros2_ws/build/fastrtps
Scanning dependencies of target fastrtps
[  1%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/log/Log.cpp.o
[  2%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/TimedEvent.cpp.o
[  3%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/TimedEventImpl.cpp.o
[  4%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/AsyncWriterThread.cpp.o
[  5%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/resources/AsyncInterestTree.cpp.o
[  6%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/Endpoint.cpp.o
[  7%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/RTPSWriter.cpp.o
[  8%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/StatefulWriter.cpp.o
[  9%] Building CXX object src/cpp/CMakeFiles/fastrtps.dir/rtps/writer/ReaderProxy.cpp.o
[... 518 lines cut ...]
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/xmlparser
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/xmlparser/XMLParser.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/xmlparser/XMLProfileManager.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/xmlparser/XMLEndpointParser.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/xmlparser/XMLTree.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/xmlparser/XMLParserCommon.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/fastrtps_dll.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/log
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/log/Colors.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/log/Log.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/log/StdoutConsumer.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/fastrtps_fwd.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/subscriber
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/subscriber/Subscriber.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/subscriber/SubscriberHistory.h
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/subscriber/SubscriberListener.h
[... cut ...]
-- Installing: /Users/jrojas/work/ros2_ws/install/fastrtps/include/fastrtps/config.h
-- Installing: /Users/jrojas/work/ros2_ws/install/fastrtps/lib/libfastrtps.1.6.0.dylib
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/lib/libfastrtps.1.dylib
-- Up-to-date: /Users/jrojas/work/ros2_ws/install/fastrtps/lib/libfastrtps.dylib
-- Installing: /Users/jrojas/work/ros2_ws/install/fastrtps/lib/fastrtps/cmake/fastrtpsTargets.cmake
-- Installing: /Users/jrojas/work/ros2_ws/install/fastrtps/lib/fastrtps/cmake/fastrtpsTargets-release.cmake
```

mikaelarguedas, 2018-07-03:
My point is that fastrtps is the package exporting the `tinyxml2` library (or exported target in this case).
If you look in the generated `fastrtpsTargets.cmake` file `/Users/jrojas/work/ros2_ws/install/fastrtps/lib/fastrtps/cmake/fastrtpsTargets.cmake` you will find `tinyxml2` being exported to downstream packages.

The reason the package failing is `rmw_fastrtps_cpp` is because this it the first package of the stack depending on `fastrtps`  (aka the first one calling `find_package(fastrtps)` and linking against it's libraries).

> Fast-RTPS is using my locally installed files.

Oh right this is where homebrew is installing those. It's good then :+1: it's not using the submodules

>  I followed the instructions on the OSX build page which make use of the vcs import to pull all of the repos (and their submodules).

vcs import should NOT clone the submodules except if asked explicitely with the `--recursive` flag.
But this doesn't seem related to this issue as FastRTPS is using tinyxml2 from your local space so submodules are not involved.

So my guess is that you are facing some Mac system protection issues. 
Did you follow the [instructions to disable SIP on your system](https://github.com/ros2/ros2/wiki/OSX-Development-Setup#disable-system-integrity-protection-sip)?

We faced this issue in the past, this ticket may give you some ideas to work around it: https://github.com/ros2/build_cop/issues/58

klintan, 2018-11-09:
Not sure if this is relevant for this specific issue, however it seems the newest version of tinyxml2 on homebrew is now 7.0 which means you need to manually specify the version of tinyxml2 to use for the latest Ros2 release (ROS 2 Bouncy Bolson - 2018-08-24) to work.

This is the installation cmd for 2.6.2 which seems to work: 

`brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/8dcb60ac8cb735a3b36c0160dc4e5d38f9d7f3e1/Formula/tinyxml2.rb`
 
Similar to the issue found here.
https://github.com/ros2/ros2/issues/440

dirk-thomas, 2018-11-29:
I will close this for now since there was no further response. Please feel free to comment and the ticket can be reopened if necessary.

closed 2018-11-29 by dirk-thomas