ros2/ros2 #522: [macOS Binary]pyyaml installing failed on python3.7 installed by brew

minggangw, 2018-07-04:
Recently homebrew has upgraded the python3 from 3.6.5 to [3.7](https://formulae.brew.sh/formula/python). And when I installed the pyyaml, which is a dependency of ROS 2.0, through `pip`, there was an [error](https://circleci.com/gh/RobotWebTools/rclnodejs/333?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link) (looks like it's incompatible with 3.7 version)

```
 building '_yaml' extension
  creating build/temp.macosx-10.12-x86_64-3.7/ext
  clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -I/usr/local/include -I/usr/local/opt/openssl/include -I/usr/local/opt/sqlite/include -I/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c ext/_yaml.c -o build/temp.macosx-10.12-x86_64-3.7/ext/_yaml.o
  In file included from ext/_yaml.c:271:
  ext/_yaml.h:10:9: warning: 'PyString_CheckExact' macro redefined [-Wmacro-redefined]
  #define PyString_CheckExact PyBytes_CheckExact
          ^
  ext/_yaml.c:139:11: note: previous definition is here
    #define PyString_CheckExact          PyUnicode_CheckExact
            ^
  ext/_yaml.c:1410:17: warning: assigning to 'char *' from 'const char *' discards qualifiers [-Wincompatible-pointer-types-discards-qualifiers]
    __pyx_v_value = yaml_get_version_string();
                  ^ ~~~~~~~~~~~~~~~~~~~~~~~~~
  ext/_yaml.c:2577:52: warning: incompatible pointer types passing 'int (void *, char *, size_t, size_t *)' (aka 'int (void *, char *, unsigned long, unsigned long *)') to parameter of type 'yaml_read_handler_t *' (aka 'int (*)(void *, unsigned char *, unsigned long, unsigned long *)') [-Wincompatible-pointer-types]
      yaml_parser_set_input((&__pyx_v_self->parser), __pyx_f_5_yaml_input_handler, ((void *)__pyx_v_self));
                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
  /usr/local/include/yaml.h:1368:30: note: passing argument to parameter 'handler' here
          yaml_read_handler_t *handler, void *data);
                               ^
```

I eliminate this error by installing the old version, which is 3.6.5

```
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/f2a764ef944b1080be64bd88dca9a1d80130c558/Formula/python.rb
```

Is it necessary to update the WiKi for the macOS platform to point out?

mikaelarguedas, 2018-07-05:
Thanks @minggangw for reporting this. Yeah I think pinning down the version on the wiki for now is the right approach until all our dependencies can be installed successfully with Python 3.7.

The Wiki is editable by anyone, would you mind submitting this change to the MacOS installation instructions (from source and from binaries) ? thanks!

minggangw, 2018-07-06:
No problem, I am going to update the WiKi both `from source` section and `from binaries` section.

minggangw, 2018-07-06:
I have updated the macOS installing/building section of WiKi, please check it out:

- [OSX Install Binary](https://github.com/ros2/ros2/wiki/OSX-Install-Binary/_compare/e7b03ccbc714c1c3b2397680fc20a3d6c5e47741...87080531da4f3479bcb9d05023ea874e91077bc0)
- [OSX Development Setup](https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/8e20672da057b441a634428e2fccecdb15e540f2...6924358120595331a513766645184878a5883f10)

I am going to close this issue, please reopen it if anything is wrong. Thanks!

closed 2018-07-06 by minggangw

mikaelarguedas, 2018-07-06:
looks good, thanks @minggangw for the fix !