ros2/ros2 #523: Issue building from source on ARM

qtdavidson55, 2018-07-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - source
- Version or commit hash:
  - 85a0b67bff0abc53af8371ea11345a1fd495ef74
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - Trying to build all libraries, but rviz_ogre_vendor generating error.

#### Steps to reproduce issue
I followed the installation instructions for building from source on the wiki, skipping over the section titled "Install more DDS implementations (Optional)"

#### Expected behavior
Correct build after use of `colcon build --symlink-install`

#### Actual behavior
Build crashes after trying to build rviz_ogre_vendor. 

Error message:

```
In file included from /home/nvidia/qdavidson/meca500/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/Ogre.h:71:0,
                 from /home/nvidia/qdavidson/meca500/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-build/Components/Python/OgrePYTHON_wrap.cxx:4838:
/home/nvidia/qdavidson/meca500/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreInstanceManager.h: In function ‘PyObject* _wrap_InstanceManager_getInstanceBatchMapIterator(PyObject*, PyObject*)’:
/home/nvidia/qdavidson/meca500/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreInstanceManager.h:98:53: error: ‘typedef Ogre::vector<Ogre::InstanceBatch*>::type Ogre::InstanceManager::InstanceBatchVec’ is private
         typedef vector<InstanceBatch*>::type        InstanceBatchVec;   //vec[batchN] = Batch
                                                     ^
/home/nvidia/qdavidson/meca500/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-build/Components/Python/OgrePYTHON_wrap.cxx:323953:3: error: within this context
   SwigValueWrapper< Ogre::ConstMapIterator< Ogre::map< Ogre::String,Ogre::InstanceManager::InstanceBatchVec,std::less< Ogre::String > >::type > > result;
   ^
/home/nvidia/qdavidson/meca500/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-build/Components/Python/OgrePYTHON_wrap.cxx: At global scope:
/home/nvidia/qdavidson/meca500/ros2_ws/build/rviz_ogre_vendor/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-build/Components/Python/OgrePYTHON_wrap.cxx:437818:16: error: ‘_wrap_new_AnyNumeric’ was not declared in this scope
     (initproc) _wrap_new_AnyNumeric,          /* tp_init */
                ^
make[5]: *** [Components/Python/CMakeFiles/_Ogre.dir/OgrePYTHON_wrap.cxx.o] Error 1
make[4]: *** [Components/Python/CMakeFiles/_Ogre.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [ogre-master-ca665a6-prefix/src/ogre-master-ca665a6-stamp/ogre-master-ca665a6-build] Error 2
make[1]: *** [CMakeFiles/ogre-master-ca665a6.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rviz_ogre_vendor	[ Exited with code 2 ]
```

#### Additional information
arm64 on Nvidia Jetson TX2

AlexisTM, 2018-07-06:
Did you try the deb version?

clalancette, 2019-09-24:
I'm sorry we never responded to this.

`rviz_ogre_vendor` has been building on arm64 for quite a while now; the latest successful compilation is here: https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_debug/950/ .

I'm going to close this out for now, but please feel free to reopen if you still run into problems.  Thanks for the report.

closed 2019-09-24 by clalancette