ros2/ros2 #525: "failed to create process" when trying to run ros2 command.

cosmicog, 2018-07-06:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Binaries
- Version or commit hash:
  - release-bouncy bd2e019
- DDS implementation:
  - Adlink OpenSplice
- Client library (if applicable):
  - N/A 

After sourcing `C:\dev\ros2\local_setup.bat`, calling even single `ros2` command gives this error:
```
> ros2
failed to create process
```
I could be able to run talker and listener by their full paths.

cosmicog, 2018-07-06:
I think the setup uses `PythonPath` becuase it's being updated with ros libraries after the call:
```
...
PUBLIC=C:\Users\Public
PythonPath=C:\dev\ros2\Lib\site-packages;C:\Python37\
PYTHON_PATH=C:\Python37\
ROS_DISTRO=bouncy
...
```
And sorry, I forgot to mention that calling `ros2.exe` directly, gives same error too.

dirk-thomas, 2018-07-06:
The environment variable is certainly named `PYTHONPATH` (see https://github.com/ament/ament_package/blob/a6c73aeff77c188597211fee9f9e34508487b531/ament_package/template/environment_hook/pythonpath.bat.in#L4).

Please set the following environment variable `set COLCON_TRACE=1` before calling `C:\dev\ros2\local_setup.bat` and post the output which enumerates all scripts being used to setup the environment.

Please also post the difference between the environment variables of a "clean" command prompt compared to after sourcing the setup file (without manually setting any additional environment variables).

cosmicog, 2018-07-06:
The variable `PythonPath` is updated after the call. And I tried setting `PYTHONPATH` to `C:\dev\ros2\Lib\site-packages;C:\Python37\` manually. It gave same error after calling `ros2`. I think this error might be related to something else.

**Clean version before calling**:
```
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\Gazi\AppData\Roaming
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=Fri Jul  6 17:22:01 2018
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=DESKTOP-GAZi-W10
ComSpec=C:\WINDOWS\system32\cmd.exe
DriverData=C:\Windows\System32\Drivers\DriverData
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
FPS_BROWSER_USER_PROFILE_STRING=Default
HOMEDRIVE=C:
HOMEPATH=\Users\Gazi
LOCALAPPDATA=C:\Users\Gazi\AppData\Local
LOGONSERVER=\\DESKTOP-Gazi-W10
NUMBER_OF_PROCESSORS=4
OneDrive=C:\Users\Gazi\OneDrive
OpenCV_DIR=C:\opencv
OPENSSL_CONF=C:\OpenSSL-Win64\bin\openssl.cfg
OS=Windows_NT
OSPL_HOME=C:\dev\HDE\x86_64.win64
Path=C:\Python37\Scripts\;C:\Python37\;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Git\cmd;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\ProgramData\chocolatey\bin;C:\dev\OpenSSL-Win64\bin\openssl.cfg;C:\ProgramData\chocolatey\lib\tinyxml2\lib;C:\dev\opencv-3.4.1-vc15.VS2017\opencv\x64\vc15\bin;;C:\WINDOWS\System32\OpenSSH\;C:\Users\Gazi\AppData\Local\Microsoft\WindowsApps;
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY;.PYW
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 158 Stepping 9, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=9e09
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=$P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules;C:\Program Files\Intel\
PUBLIC=C:\Users\Public
PythonPath=C:\Python37\
PYTHON_PATH=C:\Python37\
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\WINDOWS
TEMP=C:\Users\Gazi\AppData\Local\Temp
TMP=C:\Users\Gazi\AppData\Local\Temp
USERDOMAIN=DESKTOP-GAZi-W10
USERDOMAIN_ROAMINGPROFILE=DESKTOP-GAZi-W10
USERNAME=Gazi
USERPROFILE=C:\Users\Gazi
windir=C:\WINDOWS
```
**After calling** `local_setup.bat`:
```
ALLUSERSPROFILE=C:\ProgramData
AMENT_PREFIX_PATH=C:\dev\ros2
APPDATA=C:\Users\Gazi\AppData\Roaming
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=Fri Jul  6 17:22:01 2018
CMAKE_PREFIX_PATH=C:\dev\ros2
COLCON_PREFIX_PATH=C:\dev\ros2\
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=DESKTOP-GAZi-W10
ComSpec=C:\WINDOWS\system32\cmd.exe
DriverData=C:\Windows\System32\Drivers\DriverData
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
FPS_BROWSER_USER_PROFILE_STRING=Default
HOMEDRIVE=C:
[... 1267 lines cut ...]
call "C:\dev\ros2\share\visualization_msgs\hook\pythonpath.bat"
call "C:\dev\ros2\share/visualization_msgs/local_setup.bat"
call "C:\dev\ros2\share\common_interfaces\package.bat"
call "C:\dev\ros2\share\common_interfaces\hook\cmake_prefix_path.bat"
call "C:\dev\ros2\share\common_interfaces\hook\path_dll.bat"
call "C:\dev\ros2\share\common_interfaces\hook\path.bat"
call "C:\dev\ros2\share\common_interfaces\hook\path-scripts.bat"
call "C:\dev\ros2\share\common_interfaces\hook\pythonpath.bat"
call "C:\dev\ros2\share/common_interfaces/local_setup.bat"
call "C:\dev\ros2\share\rviz_default_plugins\package.bat"
call "C:\dev\ros2\share\rviz_default_plugins\hook\cmake_prefix_path.bat"
call "C:\dev\ros2\share\rviz_default_plugins\hook\path_dll.bat"
call "C:\dev\ros2\share\rviz_default_plugins\hook\path.bat"
call "C:\dev\ros2\share\rviz_default_plugins\hook\path-scripts.bat"
call "C:\dev\ros2\share\rviz_default_plugins\hook\pythonpath.bat"
call "C:\dev\ros2\share/rviz_default_plugins/local_setup.bat"
call "C:\dev\ros2\share\rviz2\package.bat"
call "C:\dev\ros2\share\rviz2\hook\cmake_prefix_path.bat"
call "C:\dev\ros2\share\rviz2\hook\path_dll.bat"
call "C:\dev\ros2\share\rviz2\hook\path.bat"
call "C:\dev\ros2\share\rviz2\hook\path-scripts.bat"
call "C:\dev\ros2\share\rviz2\hook\pythonpath.bat"
call "C:\dev\ros2\share/rviz2/local_setup.bat"

```

mcevoyandy, 2018-07-08:
I'm also struggling with this... Once I noticed 3.7 was installed I downgraded to 3.6.5 but then started getting errors related to rclpy and rmw_fastrtps_cpp.dll

python talker
```
$ ros2 run demo_nodes_py talker
Failed to load entry point 'launch': DLL load failed: %1 is not a valid Win32 application.
The C extension 'C:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'echo': DLL load failed: %1 is not a valid Win32 application.
The C extension 'C:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Traceback (most recent call last):
  File "C:\dev\ros2\lib\demo_nodes_py\talker-script.py", line 11, in <module>
    load_entry_point('demo-nodes-py==0.5.1', 'console_scripts', 'talker')()
  File "c:\python36\lib\site-packages\pkg_resources\__init__.py", line 476, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "c:\python36\lib\site-packages\pkg_resources\__init__.py", line 2700, in load_entry_point
    return ep.load()
  File "c:\python36\lib\site-packages\pkg_resources\__init__.py", line 2318, in load
    return self.resolve()
  File "c:\python36\lib\site-packages\pkg_resources\__init__.py", line 2324, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "C:\dev\ros2\Lib\site-packages\demo_nodes_py\topics\talker.py", line 18, in <module>
    from rclpy.node import Node
  File "C:\dev\ros2\Lib\site-packages\rclpy\node.py", line 16, in <module>
    from rclpy.client import Client
  File "C:\dev\ros2\Lib\site-packages\rclpy\client.py", line 18, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\dev\ros2\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 31, in <module>
    rclpy_implementation = _import('._rclpy')
  File "C:\dev\ros2\Lib\site-packages\rclpy\impl\__init__.py", line 21, in _import
    return importlib.import_module(name, package='rclpy')
  File "c:\python36\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: DLL load failed: %1 is not a valid Win32 application.
The C extension 'C:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
```

cpp talker:
```
$ ros2 run demo_nodes_cpp talker
Failed to load entry point 'launch': DLL load failed: %1 is not a valid Win32 application.
The C extension 'C:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions
Failed to load entry point 'echo': DLL load failed: %1 is not a valid Win32 application.
The C extension 'C:\dev\ros2\Lib\site-packages\rclpy\_rclpy.pyd' failed to be imported while being present on the system. Please refer to 'https://github.com/ros2/ros2/wiki/Rclpy-Import-error-hint' for possible solutions

>>> [rcutils|error_handling.c:155] rcutils_set_error_state()
This error state is being overwritten:

  'failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:135'

with this new error message:

  'failed to load shared library of rmw implementation. Exception: Cannot load library: C:\dev\ros2\bin/rmw_fastrtps_cpp.dll, at C:\J\workspace\ci_packaging_windows\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:135'

rcutils_reset_error() should be called after error handling to avoid this.
<<<

>>> [rcutils|error_handling.c:155] rcutils_set_error_state()
This error state is being overwritten:
```
This continues forever with basically the same message over and over... directly calling talker.exe produces the same error.

mikaelarguedas, 2018-08-11:
@cosmicog @mcevoyandy Can you confirm that this is fixed with the [new bouncy release](https://github.com/ros2/ros2/releases/tag/release-bouncy-20180720) that target Python3.7 ?

cosmicog, 2018-08-11:
I [tried](https://github.com/ros2/ros2/releases/tag/release-bouncy-20180720) and I am confirming that the issue is solved.

closed 2018-08-11 by cosmicog

sea-bass, 2018-12-19:
I have tried Python 3.6 and 3.7 with Bouncy and Crystal, and no combination resolves this for me either... not sure what is missing. Any info I can look for/provide that could pinpoint the issue?

klahaag, 2018-12-27:
I had the same issue with anaconda's python 3.7.0. The ros2.exe only outputted:
failed to create process
Interestingly python ros2-script.py ran with no error.

I then installed python 3.7.2 via chocolatey as described here (with the required python packages):
https://index.ros.org/doc/ros2/Windows-Install-Binary/
It installs python to C:\Python37
I added C:\Python37 and C:\Python37\Scripts to my path.
Then ros2.exe ran normal.

Random "bug" guessing:
* maybe easy install requires a dll from an exact python version? maybe abi incompatible?
* maybe the path does matter?

fldvrees, 2019-05-07:
I had the same issue.
I read that `ros2.exe` uses the `ros2-script.py` [here](https://stackoverflow.com/a/47081531/3506115).
In `ros2-script.py` the first line (shebang line) demands that the script is run with `C:\Python37\Python_d.exe`. 

> #!c:\python37\python_d.exe

My problem was that I didn't have `C:\Python37\Python_d.exe`. I only had the `C:\Python37\Python.exe`. 

I solved this by reinstalling Python, but now with all the debug options enabled.
After the new installation I had `C:\Python37\Python_d.exe`, and running ros2 worked!

This stupid issue costed me 8 hours of my life!

good0613, 2019-05-07:
Thanks @fldvrees !
I had the same issue and replaced ROS2 debug version to release version.
In the release version, ros2-script.py was using;
#!c:\python37\python.exe

shanpenghui, 2020-07-06:
> I had the same issue.
> `ros2.exe` uses the `<ROS installation path>/Scripts/ros2-script.py`.
> In `ros2-script.py` the first line (shebang line) demands that the script is run with `C:\Python37\Python_d.exe`.
> 
> > #!c:\python37\python_d.exe
> 
> My problem was that I didn't have `C:\Python37\Python_d.exe`. I only had the `C:\Python37\Python.exe`.
> 
> I solved this by reinstalling Python, but now with all the debug options enabled.
> After the new installation I had `C:\Python37\Python_d.exe`, and running ros2 worked!
> 
> This stupid issue costed me 8 hours of my life!

It wasted two days of mine!
Just reinstalling doesn't work for me.
It was very annoying. The solution to this problem is troublesome. I will show how I solve it below.

- Operating System:

Windows 10

- Installation type:

Binaries
Version or 

- ros2 version:

ros2-foxy-20200605-windows-debug-amd64.zip

- DDS implementation:

rti_connext_dds-6.0.1

- Client library (if applicable):

N/A

1.The real reason is exactly as what you said. @FlorisDevreese . Tks for reminding.
2.After reason was found.When I finish reinstalling python 3.8.3,there still no python_d.exe.
   I doubt that if the method of installing is wrong. At first I install py 3.8.3 using choco.
   `choco install -y python --version 3.8.3`
   So I try another way: python-3.8.3-amd64.exe in https://www.python.org/downloads/release/python-383/
   [https://www.python.org/ftp/python/3.8.3/python-3.8.3-amd64.exe](url)
   (if I could be told, I won't do that again and successed once time)
   In the process of installing, I made a mistake. I didn't choose the options:(Exactly i dont know which, just select all)

-    Associate files with Python
-    Download debugging symbols
-    Download debug binaries
   
3.But some error come out when installing.
`Error Code 0x80072F7D`
The reason is internet setting.
Just follow the steps as below.
[https://www.drivethelife.com/windows-10/fix-microsoft-store-error-code-0x80072f7d.html](url)
[image]

michael1309, 2020-07-15:
Just for clarifying: You find the debug options on the second option page of the python installer. These options must be activated. After reparing your installation you can find "python_d.exe" in your python path.
[image]

Timo9Madrid7, 2020-11-24:
> I had the same issue.
> `ros2.exe` uses the `<ROS installation path>/Scripts/ros2-script.py`.
> In `ros2-script.py` the first line (shebang line) demands that the script is run with [...]

excellent!!!

ManMan88, 2020-12-14:
Had similar issue and found out that my version of ros2-script.py contained:
`#!c:\python38\python.exe`
while the installation through choco installs python 3.9
changing to 
`#!c:\python39\python.exe`
solved it

Rutherfordum, 2021-03-04:
> > Я была такая же проблема. 
> > `ros2.exe`использует расширение `<ROS installation path>/Scripts/ros2-script.py`. 
> > В `ros2-script.py`первой строке (строка shebang) требует, чтобы сценарий запускался с `C:\Python37\Python_d.exe`.
> > > #! c: \ python37 \ python_d.exe
> > 
> > 
> > Моя проблема заключалась в том, что у меня этого не было `C:\Python37\Python_d.exe`. У меня был только файл `C:\Python37\Python.exe`.
> > Я решил это, переустановив Python, но теперь со всеми включенными параметрами отладки. 
> > После новой установки у меня был `C:\Python37\Python_d.exe`и запущенный ros2 заработал!
> > Эта глупая проблема стоила мне 8 часов моей жизни!
> 
> Два дня потрачены зря!
> Просто переустановка у меня не работает.
> Это очень раздражало. Решение этой проблемы хлопотное. Ниже я покажу, как я это решаю.
> 
> * Операционная система:
> 
> Windows 10
> 
> * Тип установки:
> 
> Двоичные файлы
> 
> * версия ros2:
> 
> ros2-foxy-20200605-windows-debug-amd64.zip
> 
> * Реализация DDS:
> 
> rti_connext_dds-6.0.1
> 
> * Клиентская библиотека (если есть):
> 
> N / A
> 
> 1. Настоящая причина именно в том, что вы сказали. @ ФлорисДевриз . Спасибо за напоминание.
> 2.После того, как причина была найдена. Когда я заканчиваю переустановку python 3.8.3, python_d.exe все еще отсутствует.
> Сомневаюсь, что если метод установки неправильный. Сначала я устанавливаю py 3.8.3 с помощью choco.
> `choco install -y python --version 3.8.3`
> Поэтому я пробую другой способ: python-3.8.3-amd64.exe в [https://www.python.org/downloads/release/python-383/ ](https://www.python.org/downloads/release/python-383/)
> [https://www.python.org/ftp/python/3.8. 3 / python-3.8.3-amd64.exe](url)
> (если бы мне сказали, я больше не буду этого делать и один раз [добьюсь](url) успеха)
> В процессе установки я допустил ошибку. Я не выбирал варианты: (Точно не знаю какие, просто выберите все)
> 
> * Свяжите файлы с Python
> * Скачать символы отладки
> * Скачать двоичные файлы отладки
> 
> 3. Но при установке вылезает какая-то ошибка.
> `Error Code 0x80072F7D`
> Причина в настройках интернета.
> Просто следуйте инструкциям ниже.
> [https://www.drivethelife.com/windows-10/fix-microsoft-store-error-code-0x80072f7d.html](url)
> [image]
> 
> После этого все заработало !!

Спасибо большое, мне это очень сильно помогло )

ItsMeTheBee, 2021-03-12:
Hey all!

I am trying to install the foxy debug version. 
Calling ros2 works as expected but when I try to use `ros2 topic list` the output it this

```
Failed to load entry point 'list': Module use of python38.dll conflicts with this version of Python.
Traceback (most recent call last):
  File "ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.9.8', 'console_scripts', 'ros2')())
  File "C:\dev\ros2_foxy\Lib\site-packages\ros2cli\cli.py", line 39, in main
    add_subparsers_on_demand(
  File "C:\dev\ros2_foxy\Lib\site-packages\ros2cli\command\__init__.py", line 249, in add_subparsers_on_demand
    extension.add_arguments(
  File "C:\dev\ros2_foxy\Lib\site-packages\ros2topic\command\topic.py", line 29, in add_arguments
    add_subparsers_on_demand(
  File "C:\dev\ros2_foxy\Lib\site-packages\ros2cli\command\__init__.py", line 236, in add_subparsers_on_demand
    extension = command_extensions[name]
KeyError: 'list'
```

Same thing for `python_d ros2-script.py topic list `

I tried installing all packages again with p`ython_d -m pip install [packages] --force-reinstall --no-cache-dir` but that didnt change anything.
Any ideas what might cause this issue?
As far as I can see python38 and python_d share the same libraries so it shouldn't make a difference which one is used.

clalancette, 2021-03-12:
> As far as I can see python38 and python_d share the same libraries so it shouldn't make a difference which one is used.

No, Windows cares hugely about this.  You can see this in your error message:

```
Failed to load entry point 'list': Module use of python38.dll conflicts with this version of Python.
```

You should follow the instructions at https://docs.ros.org/en/foxy/Installation/Windows-Development-Setup.html#extra-stuff-for-debug-mode .

In the future, please ask questions like this on https://answers.ros.org.  You are much more likely to get a response there then commenting on a closed ticket.

DanialLiu, 2021-04-12:
Maybe the version of python you installed is not same as ROS2 required.
check /ROS2_PATH/scripts/ros2-script.py 
change python path in first line to your correct python path

28945513, 2021-04-15:
> 也许您安装的python版本与所需的ROS2不同。
> 检查/ROS2_PATH/scripts/ros2-script.py将
> 第一行中的python路径更改为正确的python路径

But when I execute'ros2 run demo_nodes_py listener', it still reports error'failed to create process'

LLeeSG, 2021-08-20:
noted that u can also modify the python.exe path in ros2-script.py and local_setup.bat, which may help those  who don't wanna change their orig interpreter path

kwea123, 2021-08-20:
If you are using ROS2-xxxx-**debug**, you need to manually modify the header of `ros2-script.py` (inside `C:\dev\ros2_xxx\Scripts`) from `#!c:\python38\python_d.exe` to `#!c:\python38\python.exe` (remove the `_d`).
So I recommend to download directly the **release** version, which doesn't require this change. The tutorial should've mentioned this though...