ros2/ros2 #526: Python 3.7 Release Issues

wjwwood, 2018-07-09:
With the release of Python 3.7, most all of our binary packages, and some of our instructions need updating. Here's a list of things related to this problem that I've see so far:

- https://github.com/ros2/ros2/wiki/OSX-Install-Binary/_compare/e7b03ccbc714c1c3b2397680fc20a3d6c5e47741...87080531da4f3479bcb9d05023ea874e91077bc0
- https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/8e20672da057b441a634428e2fccecdb15e540f2...6924358120595331a513766645184878a5883f10
- https://github.com/ros2/ros2/issues/525

@minggangw FYI since you made the wiki changes.

minggangw, 2018-07-09:
We found this issue due to the error happening on our CIs. Now only the `brew` deploys the Python3.7, the [Windows](https://ci.appveyor.com/project/minggangw/rclnodejs/branch/develop) (choco)/[Ubuntu](https://travis-ci.org/RobotWebTools/rclnodejs/branches)(apt) platforms do not have this problem.

Considering the Python3.7 will be deployed in the future, we can update the instructions of Wiki to install 3.6 version explicitly, until the dependencies can work well on Python3.7.

mikaelarguedas, 2018-07-11:
On Windows, pinning the python version when installing in a fresh environment worked for me. I updated the wiki accordingly: https://github.com/ros2/ros2/wiki/Windows-Install-Binary/_compare/d93631ab6d9a7330e5f8dc8378ec532acd837ffd...3a3b739093097278cff434451021aca9d51ab8a8

I'm currently testing Python3.7 on Windows and should be able to ship the first Bouncy patch release with 3.7 removing the need to pin the version.

wjwwood, 2018-07-19:
I think this has been resolved and will be fixed in the bouncy patch1 release.

wjwwood, 2018-07-19:
Well, I'll leave this open until I can update the wiki pages (after bouncy patch 1 is finished).

mikaelarguedas, 2018-07-20:
Wiki pages updated for Python 3.7:
https://github.com/ros2/ros2/wiki/OSX-Install-Binary/_compare/87080531da4f3479bcb9d05023ea874e91077bc0...585e0be094a5fde8bb0717093634716740fa5be0
https://github.com/ros2/ros2/wiki/OSX-Development-Setup/_compare/c3e15a0ea715a5e7d705f6ba8dd1a60f6938ba0d...a0b843ced9da1fe3255efa9e9e27b5bd0f8b6fc2
https://github.com/ros2/ros2/wiki/Windows-Development-Setup/_compare/e26c7a10847632677e0258b06e0ebe2b1f96a7c6...eaee144944f59ba2a8c6761715ead836ca01d4cb
https://github.com/ros2/ros2/wiki/Windows-Install-Binary/_compare/b2114db57a7f2e1ca8962526a18f1216d7418e40...585e0be094a5fde8bb0717093634716740fa5be0

And the patch release is tagged: https://github.com/ros2/ros2/releases/tag/release-bouncy-20180720

Closing this

closed 2018-07-20 by mikaelarguedas