ros2/ros2 #529: ROS 2 Crystal Clemmys

dirk-thomas, 2018-07-09:
This meta ticket enumerate the tasks which are being worked on for the ROS 2 release scheduled for December 2019:

* Parameters in `rclpy` ros2/rclpy#
* Time ros2/ros2#
* Image transport ros2/ros2#
* Launch ros2/launch#
* Rosbag
* Gazebo ROS pkgs ros2/ros2#
* Doc v1 - ROS 2 package index ros2/ros2#

minggangw, 2018-07-11:
Combined with the bullets listed above and the [roadmap](https://github.com/ros2/ros2/wiki/Roadmap) from Wiki, I have some comments below:

>Port robot_web_tools

We have already created two projects under RobotWebTools org which target for ROS 2.0
- [rclnodejs](https://github.com/RobotWebTools/rclnodejs) - a Node.js client, maybe you have known.
- [ros2-web-bridge](https://github.com/RobotWebTools/ros2-web-bridge) - a bridge to connect to browser, complying with [rosbridge v2.0 Protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md). But it's backed by `rclnodejs` and written in JavaScript, very simple and acts as a proxy.

By saying **port**, do you mean to port the current [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) and use [rclpy](https://github.com/ros2/rclpy) to replace the one designed for ROS 1 as the backend,and still use the Python language or a complete new design? 

The two module we implemented have the equal functionality (but JavaScript specific) compared with the current `rosbridge_suite`, so I think we could take advantage of them to finish this task and make some contributions. An aggressive idea that we can migrate to Node.js ecosystem by leveraging JavaScript.

>Actions: code generation, server / client API

We have just finished this [experimental feature](https://github.com/RobotWebTools/rclnodejs/tree/actionlib) based on rclnodejs client. Thanks for @chris-smith's support from RethinkRobotics, he extracted the actionlib module and we re-use this module to achieve it in ROS 2.0. I am wondering that are there any changes, like functions, interfaces and toolchain, compared with the one for ROS 1.0 package?

>Performance testing

We opened an [issue](https://github.com/ros2/system_tests/issues/261) to discuss the benchmark design for the current ROS 2.0 clients(CPP/Python/Node.js, maybe more) about 3 months ago, actually we have had some initial result about it, please see [here](https://github.com/RobotWebTools/rclnodejs/tree/actionlib/benchmark). Now we are focusing on the automated testing and generating result graphs of the benchmark.

> Web visualization

By using the `ros2-web-bridge`, we did some simple test based on [roslibjs](https://github.com/RobotWebTools/roslibjs) and [ros2djs](https://github.com/RobotWebTools/ros2djs) to verify the compatibility. My team is concentrating on the web technologies, so I think we have more experience in this field and we can do something for it definitely. We are looking forward to the detailed plans. 

Thanks!

dirk-thomas, 2018-07-11:
> Port robot_web_tools

The different approaches (port `rosbridge_suite` to use `rclpy` as well as your new package) will be considered. It is unlikely that we will come up with a completely new design.

> Actions: code generation, server / client API

This item is referring to the task of parsing `.action` files, generate code based on these datatypes, and implementing server / client API for the languages support by the "core" of ROS 2 (C++, Python).

> Performance testing

There have been posts about different approaches in the past (e.g. on Discourse) and before starting work on this item it will be evaluated what the exact requirements and goals are and if/how we leverage any of the existing approaches.

> Web visualization

This item is a synonym for the "Port robot_web_tools" item on the roadmap.

minggangw, 2018-07-16:
@dirk-thomas thanks for your explanation and clarification, I will catch up with the latest changes and be ready to participate.

minggangw, 2018-07-30:
I opened an issue https://github.com/RobotWebTools/rosbridge_suite/issues/345 on [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite) to track the **Port robot_web_tools** task. Please feel free to input your suggestions, thanks!

yechun1, 2018-10-18:
Check with ROS2 Roadmap, there listed image_pipeline as future plan. 
We have started porting image_pipeline and plan to finish most of the functions before Crystal release, currently the depth_image_proc is ported and be ready for code review. Would you please consider to include image_pipeline on Crystal release?

We need help to create ros2 branch by project owner on ros-perception/image_pipeline so that we could submit PR and upstream the code. @dirk-thomas would you please help on this. If there is any comments about image_pipeline porting, please feel free to let us known, thanks.

image_pipeline port task:
- Image Pipeline ros-perception/image_pipeline#360

dirk-thomas, 2018-10-29:
> Would you please consider to include image_pipeline on Crystal release?

This solely depends on the state of the ported packages by the time. Whenever you are ready (and we have created a `crystal` distribution in the `rosdistro` and your dependencies are available) please feel free to release the repository using `bloom`.

asdunc, 2018-11-09:
@dirk-thomas @tfoote Happy Friday. According to https://discourse.ros.org/t/timeline-for-crystal/6676 we have only 1.5 weeks until the drop dead date for core features. 

Is it possible to segment/organize the list above into features that: 
must have pull requests merged by 11/21 
features that must have pull requests merged by 12/4
features that can come after (not dependent on crystal)

Finally, is it possible to do a call to action on the feature owners to get them to update their ETA for pull request? My team being one of the ones that need to provide that update as well.

dejanpan, 2018-11-15:
@dirk-thomas from Apex:
1. We wrote documentation and consolidated scripts for integration testing in ROS 2 into a central place. We also proposed improvements and asked for a feedback: https://discourse.ros.org/t/integration-testing-in-ros-2/6859. This task will definitely spill beyond Crystal as it is quite open-ended.
2. Regarding https://github.com/ros2/rclcpp/issues/572, we will have an update tmr morning. A prototype for deadline and liveliness is possible for Crystal but not an entirely thought through implementation.

dirk-thomas, 2018-11-15:
> Is it possible to segment/organize the list above into features that:
> must have pull requests merged by 11/21
> features that must have pull requests merged by 12/4
> features that can come after (not dependent on crystal)

Several of the items are not strictly in one category therefore I don't think it can be reordered as you described it. E.g. take actions: the message generator part of it should be in by milestone 2 since they require maintainers of third-party message generators to update their code. The rcl API should be purely additional - nobody currently relying on it since its new - hence it can get merged until the milestone 3. For stuff Open Robotics is working on there will be no new features post milestone 3. For leaf packages or even new packages there is no constraint like that - everybody decides on how own if they want to release a new feature or package in e.g. even January.

mkhansen-intel, 2018-11-15:
@tfoote @dirk-thomas - these are the PRs we're gated by for Navigation debian release:
message_filters PR [13] (https://github.com/ros2/message_filters/pull/13)
geometry2 PR [81] (https://github.com/ros2/geometry2/pull/81) @gaoethan has to address the review feedback in the next few days

And for Realsense & Object Analytics - we'd like this one merged and image_pipeline included in the Crystal release
image_pipeline PR [362] (https://github.com/ros-perception/image_pipeline/pull/362)

yechun1, 2018-11-20:
We'd like to release [object analytics](https://github.com/intel/ros2_object_analytics/tree/crystal) in Crystal, and the package was gated by below packages, would you please help to added below dependence packages in crystal/distribution.yaml?
cv_bridge – https://github.com/ros-perception/vision_opencv.git ros2 branch
message_filters – https://github.com/ros2/message_filters
pcl_perceptions – https://github.com/ros2/pcl_conversions.git
tf2_ros – https://github.com/ros2/geometry2.git

we'd also like to release image_depth_proc PR [362] (https://github.com/ros-perception/image_pipeline/pull/362). this is blocked by transport #531

yechun1, 2018-11-20:
@dirk-thomas would you please help to create https://github.com/ros2-gbp/image_pipeline-release.git? I would like to be maintainer to release for image_depth_proc package. thanks.

ross-desmond, 2018-11-20:
**Node graph API to fastrtps:** pending pull requests
rmw - https://github.com/ros2/rmw/pull/158
rcl - https://github.com/ros2/rcl/pull/333
rclpy - https://github.com/ros2/rclpy/pull/247
rmw_fastrtps - https://github.com/ros2/rmw_fastrtps/pull/236
rmw_implementation - https://github.com/ros2/rmw_implementation/pull/49
ros2cli - https://github.com/ros2/ros2cli/pull/159

**cmake macro for generating keys and keystore for nodes [5]**: pending repo/pull request
**ros2 cli generate policies [6]**: pending push to remote branch and pull request (sros2)
**ros2 secure services [1, 2, 3]**: pending push to remote branch and pull request (sros2)

Secure ROS services https://discourse.ros.org/t/ros2-security-working-group-online-meeting/6393/17
    [1] Design: Secure ROS services
    [2] Design: Secure parameters service (hardcoded currently in sros2)
    [3] Changes to security yaml configuration (tbd)
    [4] RMW Updates (FastRTPS, Connext, OpenSplice) (tbd)
    [5] Secure key storage https://discourse.ros.org/t/ros2-security-tools-for-development-and-production/6487
    [6]Nag script to check key storage and recommend setting

Priority: 

1. Complete issues related to rmw, rcl API for Crystal
2. Complete security related tasks
3. Add node graph implementation to rmw_connext, rmw_opensplice_cpp

nuclearsandwich, 2018-11-20:
@yechun1 I've created the image_pipeline-release repository and you should have received an invite to join it via GitHub.

yechun1, 2018-11-22:
@nuclearsandwich thanks for help.
We would also want to build librealsense as ros2 package. Would you please help to create https://github.com/ros2-gbp/librealsense-release.git for us? And also added @doronhi as maintainer. Many thanks.

dirk-thomas, 2018-11-22:
@yechun1 I created the [repo](https://github.com/ros2-gbp/librealsense-release) and sent both of you an invite to be an admin collaborator.

yechun1, 2018-11-23:
@dirk-thomas thanks for help.

May I known when could  below four packages be available in crystal rosdistro? These packages are required by ros2_intel_realsense, ros2_object_analytics and image_pipeline release. 
1. cv_bridge
2. image_transport
3. pcl_conversions
4. image_geometry

mjcarroll, 2018-11-28:
@yechun1 `cv_bridge` and `image_geometry` should be released via: https://github.com/ros2/rosdistro/pull/361

nuclearsandwich, 2018-11-28:
And now `pcl_conversions` is as well.

dirk-thomas, 2019-01-17:
Closing since Crystal has :ship: last month.

See #607 for the next release.

closed 2019-01-17 by dirk-thomas