ros2/ros2 #532: gazebo_ros_pkgs

dirk-thomas, 2018-07-09:
* Port the "main" plugins to ROS 2

Excluded:
* Exploit new features in ROS 2 (e.g., multiple nodes per process)

Please feel free to update the list with more information as needed.

ironmig, 2018-07-11:
I've posted some ideas for the porting plan and design on gazebo_ros_pkgs:
[https://github.com/ros-simulation/gazebo_ros_pkgs/issues/512#issuecomment-404245724](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/512#issuecomment-404245724)

I understand the intention to get the first version of this out and then iterate, but I think it would be reasonable to include support for multiple nodes in this first iteration.

Currently each plugin has its own NodeHandle, and I think Node is the closest equivalent to this in ROS2. Replacing each plugin's NodeHandle with a Node may be even easier than refactoring to have each plugin share the same Node.

chapulina, 2018-07-20:
The design and progress will be tracked on [this wiki page](https://github.com/ros-simulation/gazebo_ros_pkgs/wiki/ROS-2-integration).