ros2/ros2 #536: Update installation instructions to use rosdep to install dependencies

mikaelarguedas, 2018-07-10:
### Building from source instructions
 - [x] Linux [installation from source](https://github.com/ros2/ros2/wiki/Linux-Development-Setup)
 - [ ] MacOS [building from source](https://github.com/ros2/ros2/wiki/OSX-Development-Setup)

### Installing "fat archive" instructions 
- [ ] Linux [installation from fat archives](https://github.com/ros2/ros2/wiki/Linux-Install-Binary)
- [ ] MacOS [installation from fat archives](https://github.com/ros2/ros2/wiki/OSX-Install-Binary)

trade-off, we will install development packages instead of only runtime dependencies

mikaelarguedas, 2018-08-20:
Apt repo for stretch has been created and packages imported: http://build.ros2.org/job/import_upstream/64/

Building from source instructions now work for Debian Stretch (except for the extra DDS implementation packages that are not available on that platform)

clalancette, 2023-03-30:
We've actually done this a long time ago, though not for macOS (which is Tier 3 at this point).  There is an open issue on switching to rosdep for macOS at https://github.com/ros2/ros2_documentation/issues/1353 , so I'm going to close this one out.