ros2/ros2 #537: Bouncy Patch release 1

mikaelarguedas, 2018-07-12:
Ticket to aggregate the list of tickets that should be fixed and / or backported for the Bouncy first patch release:

## Tickets to address:
- [ ] Release Python 3.7 based archives: https://github.com/ros2/ros2/issues/526
- [ ] Memory leaks in Python generated code: https://github.com/ros2/rosidl_python/issues/5
- [ ] 

## PRs to backport

## Packages to version bump and release

mikaelarguedas, 2018-07-13:
Out of the 63 PRs merged since bouncy:
## Considered for backport:
https://github.com/ros2/rosidl_python/pull/7 (listed above)
https://github.com/ros2/rclpy/pull/203 (listed above)
https://github.com/ros2/launch/pull/96 (listed above)
https://github.com/ros2/launch/pull/99
https://github.com/ros2/ros2cli/pull/121

## Maybe?
https://github.com/ros2/launch/pull/82 @wjwwood PR description says "doesnt need to be in for bouncy" does that mean no backport?

mikaelarguedas, 2018-07-13:
> I was hoping to fix the ctrl-c issues with launch and then take all the changes merged so far in launch. Given that it is new and for the most part a leaf packages right now.

That sounds good to me

wjwwood, 2018-07-13:
After talking with @mikaelarguedas I think we'll just wait and not block this patch release on the launch issue because I don't know how long it will take to fix it.

As for the other patches in `launch` since the release, I think we should just take them all in a `0.5.2` patch release, as they are all small bug fixes in my opinion: https://github.com/ros2/launch/compare/0.5.1...master

dirk-thomas, 2018-07-17:
Maybe colcon/colcon-core#85 should be implemented, merged, and released before building the fat archives of the patch release?

dirk-thomas, 2018-07-18:
Maybe colcon/colcon-ros#26 should be reviewed, merged, and released before announcing the patch release.

mikaelarguedas, 2018-07-18:
Packaging jobs for bouncy patch 1:
 * Ubuntu bionic: [image]
 * Ubuntu xenial: [image]
 * Ubuntu aarch64 bionic: [image]
 * Ubuntu aarch64 xenial: [image]
 * MacOS: [image]
 * Windows: [image]
 * Windows Debug: [image]

mikaelarguedas, 2018-07-18:
Repos files update:
ros2.repos: https://github.com/ros2/ros2/pull/548
turtlebot2_demo.repos: https://github.com/ros2/turtlebot2_demo/pull/90

mikaelarguedas, 2018-07-19:
New round with ament_package 0.5.2

 * Ubuntu bionic: [image]
 * Ubuntu xenial: [image]
 * Ubuntu aarch64 bionic: [image]
 * Ubuntu aarch64 xenial: [image]
 * MacOS: [image]
 * Windows: [image]
 * Windows Debug: [image]

sloretz, 2018-07-20:
There are a couple issues with the xenial archive https://ci.ros2.org/job/ci_packaging_linux/137/
* Python publisher slowly leaks memory
* Could not `ctrl` + `c` `ros2 launch`

If the publisher leak is smaller and `ctrl` + `c` behavior is identical to the current bouncy release then I would say the patch looks ok. I'll check that next.

# Test Results

<details><summary>Python publisher slowly leaks, python subscriber does not leak.</summary>

In one terminal

```python
from time import sleep

import rclpy

from geometry_msgs.msg import Twist

def main(args=None):
    rclpy.init(args=args)

    node = rclpy.create_node('leak_test_publisher')

    publisher = node.create_publisher(Twist, 'topic')

    msg = Twist()

    while rclpy.ok():
        msg.linear.x = 3.14
        msg.angular.z = 2.718
        publisher.publish(msg)

    node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

In another terminal ran:

```
ros2 topic echo /topic > /dev/null
```

In a third terminal

```
top -o RES
```

The listener's memory usage stayed constant at 48040 Kib. The publisher's memory usage slowly increased from 36904 KiB to 42976 KiB.

</details>

<details><summary>Building rosidl_generator_py has a lot of cmake warnings, but passes</summary>

Roughly ran

```
mkdir -p ros2-xenial/src
cd ros2-xenial/src
git clone https://github.com/ros2/rosidl_python.git
cd ..
colcon build
```

Got stderr output

[... 241 lines cut ...]
  <license>Apache License 2.0</license>

  <buildtool_depend>ament_cmake</buildtool_depend>
  <build_depend>pluginlib</build_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>

```

Then built the package

```
subshell$ colcon build --event-handlers console_direct+ --packages-select test_pkg
Starting >>> test_pkg
-- Found ament_cmake: 0.5.1 (/home/developer/workspaces/ros2-linux/share/ament_cmake/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Found pluginlib: 2.1.1 (/home/developer/workspaces/ros2-linux/share/pluginlib/cmake)
-- Found poco_vendor:  (/home/developer/workspaces/ros2-linux/share/poco_vendor/cmake)
-- rcutils was found by plugin lib
```

</details>

mikaelarguedas, 2018-07-20:
> Python publisher slowly leaks, python subscriber does not leak

This should not leak at all like on master.
@dirk-thomas you tested that on bionic, do you see the same behavior?

> Building rosidl_generator_py has a lot of cmake warnings, but passes

expected (if you don't have connext installed) :+1:

> Able to launch process with an environment variable, but could not kill the process using `CTRL` + `c`

Expected? @wjwwood to confirm

sloretz, 2018-07-20:
> This should not leak at all like on master.

Roger. For what it's worth it leaks a lot less with the patch release than the initial release

initial release: 32956 KiB at CPU time 0.91 seconds, 66484 KiB at 1min 2.94 seconds CPU time
patch1: 28840 KiB at CPU time 1.63 seconds, 36496 KiB at 1 min 1.54 seconds CPU time

The launch `CTRL` + `c` behavior seems to be the same on both patched an unpatched. Repeating the test on both (minus the `env` keyword argument) I can't kill either using sigint, and sigterm only randomly works. I'm not sure what's going on but the patch release doesn't make it worse.

mikaelarguedas, 2018-07-20:
Thanks @sloretz for the testing!

Regarding the Python leak:

As this example doesn't follow hour guidelines (never spins or let the time to the underlying machinery to run) I am not sure this is a good testing setup.

I tested this on Xenial using a 1kHz publish loop and let it run for 10minutes (621k messages) without noticing a single byte increase on either side. So I'm going to say it's good enough (/much better) for the patch release.

<details>
<summary>Terminal1</summary>

```
$ ros2 topic pub /chatter geometry_msgs/Twist -r 1000 -p 1000
```

</details>

<details>
<summary>Terminal2</summary>

```
$ ros2 topic echo /chatter > /dev/null
```

</details>

Though it may be good to test it at higher frequencies and/or for bigger messages to see where the breaking point is.

wjwwood, 2018-07-20:
Ctrl c of launch is still expected to be broken.

dhood, 2018-07-20:
OS X binary spot check looks fine (though I found I had to reinstall my python dependencies after upgrading to python3.7). Building an overlay without e.g. opensplice using colcon works ok. I couldn't upgrade to qt11 on osx 10.10 (not supported) so wasn't able to test rviz, but `DYLD_LIBRARY_PATH` gets extended to include ogre libs correctly in zsh.

dirk-thomas, 2018-07-20:
> This should not leak at all like on master.
> you tested that on bionic, do you see the same behavior?

I tried (again) with the setup mentioned in the original ticket (`ros2 topic pub /topic <nested msg e.g. geometry_msgs/Vector2Stamped> -r 1000 -p 1000`, `ros2 topic echo /topic`) and it doesn't leak any memory.

> For what it's worth it leaks a lot less with the patch release than the initial release

With the custom Python code I do see an increasing memory usage. It might be due to the fact that the custom code runs a busy loop - adding a `sleep(0.001)` after each publish give the GC a "chance" to run. With that change I don't see any leak.

sloretz, 2018-07-20:
> With the custom Python code I do see an increasing memory usage. It might be due to the fact that the custom code runs a busy loop - adding a sleep(0.001) after each publish give the GC a "chance" to run. With that change I don't see any leak.

Adding a direct call to `gc.collect()` makes the memory usage constant too.

mikaelarguedas, 2018-07-20:
Path release is out https://discourse.ros.org/t/new-packages-for-ros-2-bouncy-2018-07-20/5421

closing

closed 2018-07-20 by mikaelarguedas